lowRISC/lowrisc-chip #3: Where to find the download for the Wishbone to AXI bridge ?

davidfongsj1108, 2016-02-23:
Hi,

For the lowrisc chip tapeout in 2016, I saw there is block for Wishbone to AXI bridge.

Where can I download this block ? 

I hope it is written in Chisel (.scala) so I can extract to verilog.

Thanks,

David

wsong83, 2016-02-23:
Hello David,

The figure in Alex's presentation is the plan for the test chip which
does not represent the current available components.
AXI/Tileine to Wishbone bridge is one in the plan but not currently
available.
I would expect this bridge to be written in Verilog (AXI-to-Wishbone) or
in Chisel (Tilelink-to-Wishbone).
The bridge can be a slave or a master depending on requirement but for
the first implementation perhaps slave first.

Best regards,
Wei

davidfongsj1108, 2016-02-23:
Hi Wei Song,

Thanks for the update.

The NASTI interface seems to be connected to either FPGA or a SIMULATION.

src/main/verilog/chip_top.sv

Will there be an option for a NASTI interface targeted to a generic DDR interface (more like an ASIC)?

Thanks,

David

Dr Wei Song ( ?? )
http://wsong83.github.io/

-

wsong83, 2016-02-24:
Yes, when the lowRISC test chip is ready to go.  -Wei

davidfongsj1108, 2016-02-24:
Hi Wei,

Is there any possibility of getting the Chisel/Systemverilog code for the Wishbone to AXI bridge and NASTI interface with DDR controller before the test chip is ready to go ?

I just need to use the lowrisc SOC as a proof of concept where we can add our special IP

so even alpha code is ok to start with.

Best regards,

David

http://wsong83.github.io/

-

wsong83, 2016-02-24:
Hello David,

We do not have any private repoes. What you can access is all what we 
can provide now.
Currently there is no Wishbone/AXI bridge available. I am not sure when 
it will be available.
Since both Wishbone and AXI are well standardized protocols, you might 
find a bridge from opencore?

As for the NASTI/DDR interface, NASTI is a sub-set of AXI.
Personally I do not quite get what is needed to support an ASIC DDR 
controller compared with an FPGA DDR controller.
Is there any special requirement for the AXI bus to support ASIC DDR 
controller?
Otherwise, I think the current NASTI interconnect might just do the job 
needed?

Best regards,
Wei

davidfongsj1108, 2016-02-25:
Hi Wei,

Thanks for your reply.

A FPGA DDR controller has the bits and pieces hidden in the block.

One only needs to interface to it from the CPU+System Logic.

Generally, it cannot be retargeted to an ASIC or another FPGA vendor.

An ASIC DDR controller (assuming no encryption) would have all the bits and pieces

fully visible in verilog and some generic IO structure.

It can be targeted to any FAB vendor and possibly an FPGA (but with lesser performance).

Best regards,

David

wsong83, 2016-02-25:
Hello David,

Oh, are you actually talking about an open DDR memory controller.
Although we very much like an open one and always try to get one from
collaborators, this is not in current plan.
For the first test chip, both the DDR PHY and controller will be
commercial IPs.

Best regards,
Wei

closed 2016-05-06 by wsong83