lowRISC/lowrisc-chip #12: Set MAM addresses and sizes from chip_top

wallento, 2016-07-09:
The MAM provides the information about the memory regions it can access. This should be configured in chip_top based on the definitions of dev_map.vh and consts.vh.

wsong83, 2016-07-09:
BRAM_BASE/BRAM_OFFSET and MEM_BASE/MEM_OFFSET already provide this information?
The only notice here is BRAM_BASE is not defined when BRAM is not there.

wallento, 2016-07-09:
We also need the size, that is currently missing.

Created this issue because I don't want to change this last minute before the release now.

wsong83, 2016-07-09:
Yes, change after release.

For the size, the size of the memory section or the size of the data bus?
For each memory section, OFFSET + 1 is the size in byte.

wsong83, 2016-08-03:
It seems I will need to differentiate mask and size due to the support for tagged memory.
With tagged memory, the main memory is reduced to an unaligned size like 224MB for a 256MB memory, where the higher 32MB reserved for tag.

As a result, the routing configuration for nasti_crossbar in socip will be changed accordingly.