lowRISC/lowrisc-chip #14: Stability issue when loading elf to memory

wallento, 2016-07-10:
We observed that the verification of the elf in memory hangs occasionally. The verification is temporarily de-activated. Suspects of this issue are:

 * The glip UART transport
 * The actual physical interface
 * The MAM timing

The problem seems only half-way undeterministic. Adding debug probes to get more details is probably a good start.

wallento, 2016-07-10:
@asb @wsong83

wallento, 2016-07-10:
After a first round of debug probes I can see that the low level interface is not hanging: It shows sufficient credit for flow control and all FIFO interfaces are ready. The same applies to the interfaces of the MAM. Next round is MAM state.

wallento, 2016-07-10:
It seems to be an issue with the logic. The MAM is in idle state and it seems to be about the same time. So probably flaw in handling counters or so. Will try to narrow it down soon.

wallento, 2016-07-11:
As everything looks good, I am back to suspecting the UART line. As we are running at only 25 MHz, with 3MBaud we have 4 percent shift. This is exactly the boundary usually allowed, so maybe the clock is not adjusted per symbol in the USB-UART chip and transmitting larger chunks of data then creates loss.

wallento, 2016-08-15:
The issue is with the last packet in a read reply, but it is not deterministic which one. The packet leaves the FPGA and enters glip, so that I now focus on the software.

wsong83, 2016-08-22:
I assume this issue is potentially fixed. Close it for now, feel free to reopen it again if stuck again.

closed 2016-08-22 by wsong83