lowRISC/lowrisc-chip #15: Generic bootloader

wallento, 2016-07-10:
We should build a generic bootloader, that later will probably be replaced by a secure bootloader in a minion.

But for now it should follow the following precedence:
 * Check a specific address on an memory-mapped debug module that overwrites the boot, currently only to "boot from RAM"
 * Mount SD and load program to RAM, jump to RAM
 * Mount FPGA Flash, load program and jump to RAM

wsong83, 2016-07-11:
The plan from RISC-V is to use Coreboot and I believe lowRISC will follow that once it is more clear about how to do it. My personal suggestion would be, if it is a marginal effort to make the current boot procedure easier to use, it is OK. Otherwise, do not spend much time on standarizing boot procedure as it is likely to be changed significantly.

Secure boot is another thing. It is in the plan and possibly a major release when it is done. So my suggestion would be research, propose but not implement for now.

wallento, 2016-07-11:
Oh, I think there is a misunderstanding. Its now a full blown bootloader, like coreboot or so, but just extending the current `boot.c`.