lowRISC/lowrisc-chip #23: Using open source graphics processor adds lowrisc

OJOIN, 2016-08-22:
Using open source graphics processor adds lowrisc

gplgpu

GPL v3 2D/3D graphics engine in verilog

https://github.com/asicguy/gplgpu

miaow

An open source GPU based off of the AMD Southern Islands ISA.

https://github.com/VerticalResearchGroup/miaow

wsong83, 2016-08-22:
Thanks for the suggestion.
We are aware of the need for a GPU and the opensourced implementations but still pondering when and how to do it.

OJOIN, 2016-08-25:
Many core GPGPUS unified architecture using software renderer.

LowRISC not require a separate GPU ip.Choose many core GPGPUS unified architecture using software renderer .It appears to be a good technical route.

SalviaRenderer

Introduction

SALVIA is a pure object-oriented software 3D renderer with a programmable pipeline only. Interfaces, operations and parameters of it take advantage of modern graphics APIs such as D3D 10 and OpenGL.

https://github.com/wuye9036/SalviaRenderer

https://code.google.com/archive/p/softart/

Muli3D

Description
Muli3D is a rendering library which implements features of current graphics hardware in software. The focus is on ease of use and clear code; therefore Muli3D is best suited for research and teaching.

https://sourceforge.net/projects/muli3d/

Allegro 

Allegro is a cross-platform library mainly aimed at video game and multimedia programming. It handles common, low-level tasks such as creating windows, accepting user input, loading data, drawing images, playing sounds, etc. and generally abstracting away the underlying platform. However, Allegro is not a game engine: you are free to design and structure your program as you like.

Allegro 5 has the following additional features:

Supported on Windows, Linux, Mac OSX, iPhone and Android
User-friendly, intuitive C API usable from C++ and many other languages
Hardware accelerated bitmap and graphical primitive drawing support (via OpenGL or Direct3D)
Audio recording support
Font loading and drawing
Video playback
Abstractions over shaders and low-level polygon drawing
And more!

http://liballeg.org/index.html