lowRISC/lowrisc-chip #28: Bootloader

furkanturan, 2017-02-09:
Hello,

I tested the v2 branch previously for my project. It is documented well on the lowrisc.org. I ported the design back to ZedBoard, making it use the half of ARM's DDR memory, and I provided the UART and SD card via PMODs. I followed the steps as described: built the boot.c, and ./bbl, provided the Linux kernel. Everything works.

I want to try the v3 branch as well. It interests me as it has a different boot mechanism than BBL, and on my project I need to work with a custom bootloader. (I am also trying to follow the Coreboot port progress) However, I didn't understand how to use the bootloader in this branch. It has an own sub-repository; however no description/readme.

Can you please, briefly explain how to use the bootloader of the v3 branch to boot Linux?

Kind Regards,

wsong83, 2017-02-09:
Hello Furkan Turan,

The boot procedure is indeed different with v2 because the upstream
Linux has changed.
It is still a BBL but in a separate submodule.
The BBL in v3 needs to be compiled into the same ELF with the kernel.
So BBL no long load Linux kernel from SD but from a location in physical
memory.

I think you can compare the bootloader submodule with the riscv-pk from
the v3 lowrisc-chip repo.
The bootloader submodule is revised from that version of riscv-pk.
The actual revision is roughly the same for the v2 except for the change
in loading Linux kernel.

There is a script (does mentioned in tutorial) in riscv-tools:
    riscv-tools/make_root.sh
It has all the commands needed to regenerate a BBL+Kernel image.

Hope this is helpful,
Wei

furkanturan, 2017-02-21:
Hello Again Wei,

Thank you very much for your reply, it was very helpful indeed. I created the boot.bin image and working on loading it. At this point I have another issue, but I will ask that as another question with a dedicated title so that the topic can help others which may observe the similar problem.

Kind Regards,

Furkan

closed 2017-02-21 by furkanturan