lowRISC/lowrisc-chip #30: ISA regression tests non-functional in spike

asb, 2017-02-21:
I haven't been able to pin down exactly when things were broken, but I can confirm it's independent of the PTE changes etc that were introduced since the last release. Building debug-v0.3 of lowrisc-chip and then trying to run the isa tests under spike (e.g. with `make -C ./riscv-tests/isa/ run`) results in failure:

    HTIF: attempt to read from illegal address 0x400005c0

wsong83, 2017-02-21:
The spike test for debug-v0.3 was broken. If I remember it right, spike still relies on htif so there was no attempt to make it work. Why do you need spike regression tests?

asb, 2017-02-21:
I was modifying spike to support tag rules, to aid development of further software support. Now I see the problem I can sort out a working riscv-test-env. Let's leave this open as a tracking issue though, because we should sort out a proper fix at some point.

@jrrk - what riscv-tests and riscv-tests-env are you running on the minion?

wsong83, 2017-04-06:
@asb Do you have a working version of the spike that runs the isa+tag regression and can boot linux? Would you like to push it to lowrisc repos? Then this issue can be closed.

asb, 2017-04-07:
The current solution to running tests with spike is to use the tagpipe-spike branch of riscv-test-env. Given that the mismatch between spike and the FPGA build should disappear once we update to a newer spike and a newer rocket, I think that's probably the best solution for now.

I implemented a basic version of tag pipeline rules in spike (though not yet pctag) to aid my work on the gcc modifications. Given that we decided to wait for after the release to merge these, I put that work to one side. I need to go back and integrate the tag pipeline processing in a way that doesn't degrade spike performance so much.