lowRISC/lowrisc-chip #38: vcd2vpd not found

ninode, 2017-04-07:
Hello, 

I'm running the make run-vpd in the emulator. 
I get the following: 
make run-vpd
```
rm -rf output/rv64ui-p-add.vpd.vcd && mkfifo output/rv64ui-p-add.vpd.vcd
vcd2vpd output/rv64ui-p-add.vpd.vcd output/rv64ui-p-add.vpd > /dev/null &
./emulator-DefaultCPPConfig-debug +dramsim +max-cycles=100000000 +verbose -voutput/rv64ui-p-add.vpd.vcd +loadmem=output/rv64ui-p-add.hex none 3>&1 1>&2 2>&3 | /home/alpha/lowrisc-chip/riscv/bin/spike-dasm  > output/rv64ui-p-add.out && [ $PIPESTATUS -eq 0 ]
/bin/bash: vcd2vpd: command not found

```
Could you please help me fix it ? Is it necessary to simulate RISCV on Zynq ? Or is there another way ?

Thank you

wsong83, 2017-04-07:
You are using an old release of lowRISC as this vpd make target has been removed when we replace VCS with Verilator as the default simulator.

This vcd2vpd command is a utility command provide by Sysnopsys VCS simulator, which is a commercial tool. It should not be necessary as long as the vcd file is generated. You can use gtkWave (a free tool available on most Linux platforms) to see vcd file.

Instead of using the vpd make target, use the vcd make target instead (if I recall it right, we did provide this choice).

closed 2017-04-07 by wsong83

ninode, 2017-04-07:
Thank you. 
Are you referring to untetherv2.0 or LowRISC with trace debugger ?

wsong83, 2017-04-08:
Yes, untether-v0.2 and debug-v0.3 has replaced VCS with verilator.
There is no C++ emulator in these two.
In the tagged-memory-v0.1 version, make target run-vcd should give you vcd files in emulator.