lowRISC/lowrisc-chip #45: Is L2 inclusive or not?

sxu55, 2017-04-25:
It seems to me that the L2 cache implementation in the Uncore is neither inclusive nor exclusive, because the eviction policy doesn't distinguish cached/uncached lines in tile caches. Could you confirm if my understanding is correct or not? Thanks.

wsong83, 2017-04-25:
It is inclusive. Why do you believe "the eviction policy doesn't distinguish cached/uncached lines"?
When a line is replaced, if I recall it right, the writeback unit will probe all L1s who may have the latest value.

sxu55, 2017-04-25:
In the definition of replacement policy (cache.scala), neither random nor PLRU replacement takes the cache line metadata into account (random simply uses LFSR and PLRU simply uses its own state_reg) when generating the way information. And it seems L2MetadataArray class simply uses the replacer to get the way to evict, and therefore makes me wondering if it is possible to evict a line that has been cached in L1. 
The other thing that I don't quite understand is when the L2AcquireTracker tries to determine if it needs to make a transition to writeback state when it is in meta_resp state (line 1074 in cache.scala). When generating the flag for _needs_writeback_, it has logic like (coh.outer.requiresVoluntaryWriteback() ||
                             coh.inner.requiresProbesOnVoluntaryWriteback()), which makes me confused. I would appreciate if you could shed some light on my confusion here.