lowRISC/lowrisc-chip #46: Verilator error

ninode, 2017-05-03:
Hello, 

when trying to run make sim in ~lowrisc-chip/vsim (untether-v2.0), I get the following error:

```
alpha@alpha-VirtualBox:~/lowrisc-chip/vsim$ make sim
cd verilator && make -f Vchip_top.mk
/bin/bash: line 0: cd: verilator: No such file or directory
make: *** [DefaultConfig-sim] Error 1

```
My environment variables are:
```
alpha@alpha-VirtualBox:~/lowrisc-chip/vsim$ which verilator
/usr/bin/verilator

```
Can anyone please tell why this isn't working ?

Thank you !

jrrk, 2017-05-03:
Our simulator requires a version of verilator which is much newer than 
comes with most distributions.

Follow the instructions on our website lowrisc.org to install verilator 
first, including setting the correct

environment variable. You may need to run git clean in the vsim 
directory as well after a failure of this kind.

wsong83, 2017-05-03:
If you get an error during the make process, for example a syntax error or compiler error, after you have fixed the error, you probably need to make clean before make again.

The issue here is that, ventilator.log is always generated but the verilator directory is only created if the make runs ok.

closed 2017-06-05 by wsong83

baemyoungsub, 2017-10-25:
I have suffered from the same problem.  This is Text of verilator.log :
`sh: 1: /local/tool/verilator/verilator_bin: not found
%Error: Command Failed /local/tool/verilator/verilator_bin --cc /home/bae/lowRISC/lowrisc-chip/src/main/verilog/chip_top.sv /home/bae/lowRISC/lowrisc-chip/vsim/generated-src/Top.DefaultConfig.v /home/bae/lowRISC/lowrisc-chip/src/test/verilog/nasti_ram_behav.sv /home/bae/lowRISC/lowrisc-chip/src/test/verilog/host_behav.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/channel.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/lite_nasti_reader.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/lite_nasti_writer.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_buf.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_combiner.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_crossbar.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_demux.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_lite_bridge.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_lite_reader.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_lite_writer.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_mux.sv /home/bae/lowRISC/lowrisc-chip/socip/nasti/nasti_slicer.sv /home/bae/lowRISC/lowrisc-chip/socip/util/arbiter.sv /home/bae/lowRISC/lowrisc-chip/src/test/cxx/common/globals.cpp /home/bae/lowRISC/lowrisc-chip/src/test/cxx/common/dpi_ram_behav.cpp /home/bae/lowRISC/lowrisc-chip/src/test/cxx/common/dpi_host_behav.cpp /home/bae/lowRISC/lowrisc-chip/src/test/cxx/veri/veri_top.cc -DVERILATOR_GCC +incdir+/home/bae/lowRISC/lowrisc-chip/vsim/generated-src +incdir+/home/bae/lowRISC/lowrisc-chip/src/main/verilog --top-module chip_top --unroll-count 256 --trace -Wno-lint -Wno-style -Wno-STMTDLY -CFLAGS '-std=c++11' -CFLAGS '-I/home/bae/lowRISC/lowrisc-chip/src/test/cxx/common -I/home/bae/lowRISC/lowrisc-chip/src/test/cxx/veri' --exe --assert --Mdir ./verilator -O3 -CFLAGS -O1 -o ../DefaultConfig-sim` 

and this is verilator/bin dirctory
`verilator           verilator_difftree  verilator_profcfunc
verilator_coverage  verilator_includer
`
I don't know what went wrong. PLEASE, somebody tell me solution....

jrrk, 2017-10-25:
It looks like you did not follow the instructions for installing 
Verilator correctly. When the documentation refers to

/local/tool it really means "the directory you decided to install 
verilator", which by default will be /usr/local, but you can choose a 
suitable

directory when you install verilator. For example a user without admin 
privileges might choose /home/$USER/verilator.

Also be sure to attend to the creation of the links as explained in the 
documentation on lowrisc.org.

Do not attempt to use the system version of verilator, on most 
distributions it will be far too old to understand the latest syntax 
that we use.

You also need to create an environment variable, example:

export VERILATOR_ROOT=/usr/local/share/verilator # or wherever you put it

baemyoungsub, 2017-11-11:
jrrk : thank you. I reinstalled ubuntu and packages because my computer is totally mass up. so now I faced same error comment. i don't understand your first answer "Our simulator requires a version of verilator which is much newer than comes with most distributions.". I use  verilator-3.914. (Oct 14 2017) version. This is latest version of verilaor i found and i used. May i asy you how can i download satisfied version for RTL simulation??

asb, 2017-11-11:
Jonathan was just pointing out that if you're using a distribution like Ubuntu 16.04 LTS you might find that the distribution-provided verilator is too old. A version from October should be just fine.

codelec, 2017-11-11:
wget http://mirrors.kernel.org/ubuntu/pool/universe/v/verilator/verilator_3.914-2_amd64.deb
sudo dpkg -i verilator_3.914-2_amd64.deb

jrrk, 2017-11-11:
If you install from binary dpkg the instructions on our website need to be modified. In particular you need to run verilator in a verbose mode to find the VERILATOR_ROOT that’s expected. You might also need to recompile some library files such as verilated.o and friends. With this type of tool a universal binary version is not practical. On the other hand if you compile from source (which is slow but not hard) our instructions should apply.

baemyoungsub, 2017-11-12:
asb   I currently use verilator-3.914 (Oct 14 2017). the error still appears to me. this is mt problem

FranDuqueAyachi, 2022-04-27:
> asb I currently use verilator-3.914 (Oct 14 2017). The error still appears to me. This is my problem

Hi, I have the same Issue, did u fix it ?

jrrk, 2022-04-27:
I’m sorry, there is no maintainer on this project any more. It is highly likely to fail as tools and operating systems move on.