lowRISC/lowrisc-chip #47: FPGA boards

ninode, 2017-05-03:
Hello, 

Is it possible to do an FPGA demo in a zynq ? 
I got this error when generating a bitstream for the default fpga kc705:

```
source script/make_project.tcl
# set mem_data_width {128}
# set axi_id_width {9}
# set origin_dir "."
# set base_dir "../../.."
# set common_dir "../../common"
# set project_name [lindex $argv 0]
# set CONFIG [lindex $argv 1]
# set orig_proj_dir [file normalize $origin_dir/$project_name]
# create_project $project_name $origin_dir/$project_name
# set proj_dir [get_property directory [current_project]]
# set obj [get_projects $project_name]
# set_property "default_lib" "xil_defaultlib" $obj
# set_property "board_part" "xilinx.com:kc705:part0:1.1" $obj
ERROR: [Board 49-71] The board_part definition was not found for xilinx.com:kc705:part0:1.1. The project's board_part property was not set, but the project's part property was set to xc7k70tfbv676-1. Valid board_part values can be retrieved with the 'get_board_parts' Tcl command. Check if board.repoPaths parameter is set and the board_part is installed from the tcl app store.
INFO: [Common 17-206] Exiting Vivado at Wed May  3 15:37:10 2017...
make: *** [lowrisc-chip-imp/lowrisc-chip-imp.xpr] Error 1
```
And then, 

```
Vivado% get_board_parts
em.avnet.com:zed:part0:0.9 em.avnet.com:zed:part0:1.0 em.avnet.com:zed:part0:1.1 em.avnet.com:zed:part0:1.2 em.avnet.com:zed:part0:1.3 xilinx.com:ac701:part0:1.0 xilinx.com:ac701:part0:1.1 xilinx.com:ac701:part0:1.2 xilinx.com:zc702:part0:0.9 xilinx.com:zc702:part0:1.0 xilinx.com:zc702:part0:1.1 xilinx.com:zc702:part0:1.2
```
Is there something I can do to generate bitsream in a zedboard ?

Thank you !

jrrk, 2017-05-03:
zynq is not supported in untether-v2.0. It is supported via an external 
contributor

in debug-v0.3. You need to select the board type using an environment 
variable

as per the instructions on the website. The error below is unrelated and 
is most

likely due to the use of an incompatible version of Vivado (we use 2015.4)

wsong83, 2017-05-03:
It does seems your Vivado has a limited support of boards. As described by Jonathan, you need to change the FPGA_BOARD macro. The make_project script is not supposed to be run standalone.

@jrrk2. Jonathan, would you point him to the right tutorial URL please.

jrrk, 2017-05-03:
Dear ninode,

The zedboard instructions were provided by Furkan Turan 
<turanfurkan@outlook.com>

and are available at the address below:

https://github.com/furkanturan/lowrisc-zed

Our repository was modified to port to this board, but we do not have a 
board to try just now.

I suggest you try to follow the instructions at the above link as 
closely as possible, and then

address any issues with the original author of the port. You will see 
from the instructions that

to do anything more useful than just learn about it you need to buy two 
extra PMODs.

Needless to say, you should attach all relevant details such as Vivado 
version and O/S to your query.

Regards,

Jonathan

ninode, 2017-05-04:
I am using Vivado 2015.4. I will try the zedboard instruction that you have provided. 
Thank you for your reply and your suggestions.

closed 2017-06-05 by wsong83