lowRISC/lowrisc-chip #49: Where is dev_map.h?

neocogent, 2017-06-05:
I've been trying to get a build environment set up for fpga testing. I've tried meticulously to follow the info here and on lowrisc.org/docs but have hit a wall. 

I'm trying to run the riscv-tools/make_root.sh script to create a root.bin. It sets a variable for the intended fpga board eg. nexys_ddr. It tries to build bbl for the fpga bootloader but dev_map.h is missing. I've searched everywhere and this file appears to not exist.

```
build bbl...
configure: WARNING: using cross tools not prefixed with host triplet
In file included from ../machine/mtrap.h:4:0,
                 from ../bbl/bbl.c:2:
../machine/encoding.h:6:21: fatal error: dev_map.h: No such file or directory
 #include "dev_map.h"
                     ^
compilation terminated.
In file included from ../machine/mtrap.h:4:0,
                 from ../bbl/kernel_elf.c:3:
../machine/encoding.h:6:21: fatal error: dev_map.h: No such file or directory
 #include "dev_map.h"
                     ^
compilation terminated.
make: *** [bbl.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [kernel_elf.o] Error 1

```
I tried just commenting it out but it actually appears to need the value 

    #define DRAM_BASE          DEV_MAP__mem__BASE

For several places in that build. 

Either I've missed a step somewhere, or some code is missing, or the current docs aren't complete in how to build everything. I'd appreciate any suggestions. It does seem to me that there ought to be a dev_map.h included with the board specific files but none is present for me.

Thx.

jrrk, 2017-06-05:
This file gets auto-generated when you build the Chisel subsystem. See the simulation or FPGA instructions, and carry out this action before trying to build Linux.

neocogent, 2017-06-05:
Ok. Thx. Will do.

wsong83, 2017-06-06:
Some extra notice has now been added to the tutorial near the Linux kernel build process.

As for your new question, the biggest issue I can see for a35t is the size. You might want to significantly cut down the whole design. Such as removing all IO peripherals not used, removing L2, removing trace debugger, removing FPU, etc.

The default configuration consumes ~70% of the LUTs on a100t, that is around 200% of the LUTs on a35t!

I am closing this for now as the original question has been resolved.

closed 2017-06-06 by wsong83