lowRISC/lowrisc-chip #55: RISCV RTOS port

ninode, 2017-06-28:
Hello, 

Is it possible to push another small rtos with rocket cores to the fpga instead of riscv-linux ? The rtos in question has its own gcc. 
Can an RTOS be ported on top of the BBL for the untethered (or tethered (ucb-bar)) systems ?
If so, could you please explain what would be a good place to start ? 

Thank you

wsong83, 2017-06-28:
I do not see much benefit to port a RTOS on the top of BBL. Why not give the RTOS the full control of the SoC.
The interface between riscv-linux and BBL is the SBI APIs. Use BBL if you would like to reuse these SBIs (which may bring future compliance if the SBI is standardized).

I think the easiest way to boot Rocket with a RTOS, is to use a simple bootloader, such as the boot example program, to load the RTOS image to a fixed location then jump to it. RTOS can have the full address information at compile time. Once control is transferred from the first stage bootloader to RTOS, RTOS is the only programming running on the SoC.

ninode, 2017-06-28:
Thank you, very clear and helpful.

So, to boot ROCKET with RTOS  in case of a zedboard I use Vivado to generate the BOOT.bin file which contains the FPGA Bitstream, the First Stage Bootloader, and the U-Boot Bootloader. Can these be generated specifically for the RTOS by provifing the .elf file of the RTOS ?

wsong83, 2017-06-28:
You are talking about the tethered system which use the ARM core (which runs the U-Boot) to load a image to Rocket. We do not support tethered system anymore.

For a tethered system, one problem is that the Rocket does not have direct accesses to many peripherals, which are handled by the ARM core and then emulated through the host target interface. We need to ask the people behind https://github.com/ucb-bar/fpga-images-zedboard and https://github.com/ucb-bar/fpga-zynq

asb, 2017-06-28:
I don't know off-hand which systems they have been tested with (I assume RV32 primarily), but you might want to look at the Zephyr RISC-V port.

closed 2017-07-10 by wsong83

heshamelmatary, 2017-07-14:
For the record, there have been [some work](https://github.com/heshamelmatary/rtems-riscv/tree/priv-1.10) to port RTEMS to RV32. There's also a Google Summer of Code student working on [porting RTEMS to SiFive's HiFive board](https://devel.rtems.org/wiki/GSoC/2017) this year, so you might want to track the progress. I wouldn't consider it mature/stable though until it gets upstream.