lowRISC/lowrisc-chip #60: vcs make file error

clare7, 2017-08-10:
Hi when I tried to 
`make`
for `/lowrisc-chip/vcs`
I get the following errors:

`Error-[IND] Identifier not declared
.../lowrisc-chip/minion_subsystem/verilog/minion_soc.sv, 66
  Identifier 'one_hot_rdata' has not been declared yet. If this error is not 
  expected, please check if you have set `default_nettype to none.`

any advice?

jrrk, 2017-08-10:
Suppprt for vcs is experimental. Try moving the declaration of one_hot_rdata and friends earlier in the file (before first use).

clare7, 2017-08-10:
What are the function of these one_hot_rdata and friends actually?
So I just need to comment them out in the files?

clare7, 2017-08-10:
Alright noted! thanks

closed 2017-08-10 by clare7

jrrk, 2017-08-10:
No they should not be commented out. VCS has a stricter interpretation of Verizon ordering of statements than Vivado, so you need to move the declarations that VCS complains about earlier in the file. I have a feeling this bug was already fixed on some branch, but you do not say what branch you are working on.

The function of the signals is to multiplex between the control and data signals of the different Minion peripherals.

clare7, 2017-08-10:
Thanks! I have did what you have suggested and it works!
With that I close the topic :)