lowRISC/lowrisc-chip #61: RoCC with Lowrisc

clare7, 2017-08-10:
Any guideline on using rocc with lowrisc rocket chip?
would the procedure similar to that of the rocketchip developed by Berkeley?

jrrk, 2017-08-10:
The goal of lowRISC is to make a security enhanced and usable Linux 
system based on Berkley Chisel and Xilinx FPGA IP.

As such the source base diverged quite a bit from their methodology, and 
we have yet to see a new version from them

which has all the functionality of the previous code base that our chip 
is based on. For example the L2 cache is missing

in the rewritten code. At the same time there have been quite a few 
changes in CSRs which make it difficult to track the

latest compiler technology.

An investigation is underway to solve the problems of rebasing at the 
moment, but as of now I would suggest avoiding any

modifications other than the ones we explicitly suggest on our website, 
until you reach experienced user status.

wsong83, 2017-08-10:
As Jonathan said, RoCC is not something you can easily plug and play. The RoCC in the upstream Rocket-Chip is not stable and lag from latest development anyway. If RoCC is your main goal to use lowRISC, I think it is unavoidable that you need to understand the interface to a level of feeling comfortable with revising the Chisel code yourself. When you reach there, probably configuring lowRISC to enable your own RoCC component is no longer an issue.

Right now for lowRISC, we use the old Rocket code around Summer 2016 and it does come along with a RoCC interface, although this interface is disabled by all configurations we provide right now. You need to change the top level Chisel configuration to enable it. As I mentioned earlier, it is not plug and play, so proving instructions to enable the interface before you understand the interface itself is useless. Once you understand it, enabling it will be easy peasy.

I would suggest you to have a careful look of this repository: https://bitbucket.org/taylor-bsg/bsg_riscv

The document and code there will help you understand the RoCC interface. I believe the RoCC interface there is not much different with ours.

Since we do not use the RoCC interface (we do not understand it much to be frank), there is very little support from lowRISC.

wsong83, 2017-08-22:
Feel free to open it again.

closed 2017-08-22 by wsong83