lowRISC/lowrisc-chip #62: Mode of satp(or sptbr) register in lowrisc

Ddnirvana, 2017-08-10:
Hi, guys:

From the riscv-privileged-v1.10, chapter 4.1.12, I find that the "Supervisor Address Translation and Protection (satp)" register contains a Mode and ASID and PPN。

Also, from the code in https://github.com/riscv/riscv-linux/blob/riscv-next/arch/riscv/kernel/head.S, there are some codes to configure the sptbr register using the SPTBR_MODE.
<pre>
/* Compute sptbr for kernel page tables, but don't load it yet */
	la a2, swapper_pg_dir
	srl a2, a2, PAGE_SHIFT
	li a1, SPTBR_MODE
	or a2, a2, a1

	/* Load trampoline page directory, which will cause us to trap to
	   stvec if VA != PA, or simply fall through if VA == PA */
	la a0, trampoline_pg_dir
	srl a0, a0, PAGE_SHIFT
	or a0, a0, a1
	sfence.vma
	csrw sptbr, a0
</pre>

However, in the lowrisc's version linux(https://github.com/lowRISC/riscv-linux), the head.S does not contain codes to configure the sptbr...And I don't see any codes in the linux to set the Modes of sptbr too...

So I am just wondering how lowrisc handle the Mode filed in the sptbr(satp). 

Any responses are welcomed! Thanks!

asb, 2017-08-10:
Hi. I'm afraid we don't currently implement the 1.10 privileged spec. We back-ported the most obvious changes in the 1.9 spec (e.g. the page table entry format change), but can't currently guarantee full compatibility with 1.9 or 1.10.

Ddnirvana, 2017-08-10:
Hi @asb  , thanks for your quickly response!

Another question is which version of riscv's specification used by v0.2's lowrisc(untether)?

wsong83, 2017-08-10:
We have not really thought about recording this information at that time.
The untether v0.2 came out around December 2015 probably using the Rocket of October 2015.
My best guess: user spec 2.0 and priv spec 1.7.

Ddnirvana, 2017-08-10:
@wsong83 thanks for your response. It's very helpful!

closed 2017-08-10 by wsong83