lowRISC/lowrisc-chip #64: Merge upstream Rocket-Chip to support Chisel3+TileLink2+Diplomacy+PLIC-Debug [WIP]

wsong83, 2017-08-15:
TODO list:

- [x] clean up a branch
- [ ] set up the initial lowrisc-chip chisel top for make
- [ ] reimplement trait `HasMasterAXI4MMIOPort` to allow configuration time expansion of multiple slaves to the `AXI3SlaveParameters`.

wsong83, 2017-10-19:
pass ISA regression in both vcs and vsim.
Branch merge becomes update.

asb, 2017-10-19:
Congratulations Wei! This a fantastic milestone.

wsong83, 2017-10-25:
Currently the uart bare-metal test simulated OK. Right now I am going to test it with the real Nexys4-DDR board.

Some inconvenient findings during the process:
- The latest rocket-chip cannot be simulated using Vivado 2015.4, or even any version lower than 2017. The SystemBus module would crash the xelab causing segment error.
- Using Vivado 2017.3, I can run simulation. However, the DPI interface somehow does not work yet (I was forced to disable all DPI calls to avoid run-time exceptions) also the synthesis will crash without a reason.

That is to say, there is current NO stable version of Vivado that can do implementation and simulation at the same time.

For the sake of debugging, I leave the simulatible version in a branch called vivado-2017.3 of the lowrisc-nexys4 repo.

wsong83, 2017-10-28:
Close this issue for now.

closed 2017-10-28 by wsong83