lowRISC/lowrisc-chip #68: Compilation error after Enable Debug at lowRISCv0.4

jawadhy, 2017-08-24:
Hello,

I have enabled the debug on the Configs.Scala (**case UseDebug => true** ) and tried to generate verilog using : **cd $TOP/vsim ; make verilog** , the following compilation errors occurs, could you please help figure out what is the issue:

 [jawad@localhost vsim]$ make verilog
cd /home/jawad/lowrisc-chip && mkdir -p /home/jawad/lowrisc-chip/vsim/generated-src && java -Xmx2048M -Xss8M -XX:MaxPermSize=256M -jar sbt-launch.jar "run Top DefaultL2Config --W0W --minimumCompatibility 3.0.0 --backend lowrisc_chip.LowRISCBackend --configName DefaultL2Config --compileInitializationUnoptimized --targetDir /home/jawad/lowrisc-chip/vsim/generated-src --configDump --noInlineMem"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /home/jawad/lowrisc-chip/project
[info] Set current project to lowrisc_chip (in build file:/home/jawad/lowrisc-chip/)
[info] Compiling 1 Scala source to /home/jawad/lowrisc-chip/target/scala-2.11/classes...
[info] Running lowrisc_chip.Run Top DefaultL2Config --W0W --minimumCompatibility 3.0.0 --backend lowrisc_chip.LowRISCBackend --configName DefaultL2Config --compileInitializationUnoptimized --targetDir /home/jawad/lowrisc-chip/vsim/generated-src --configDump --noInlineMem
Generated Address Map
	io:ext:host 4000 - 403f
	mem 40000000 - bfffffff
	io:int:prci0 3000 - 3fff
	io:int:rtc 2000 - 2fff
	io:int:bootrom 0 - 1fff
Generated Configuration String
platform {
  vendor lowRISC;
  arch rocket;
};
rtc {
  addr 0x2000;
};
ram {
  0 {
    addr 0x40000000;
    size 0x80000000;
  };
};
core {
  0 {
    0 {
      isa rv64imafd;
      timecmp 0x2008;
      ipi 0x3000;
    };
  };
};

[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at lowrisc_chip.Run$$anonfun$40.apply(LowRISCChip.scala:322)
	at lowrisc_chip.Run$$anonfun$40.apply(LowRISCChip.scala:319)
	at Chisel.Driver$$anon$$$$215427f2c411fef219772bc14f46c4$$$$nfun$apply$1.apply(Driver.scala:102)
	at Chisel.Driver$$anon$$$$215427f2c411fef219772bc14f46c4$$$$nfun$apply$1.apply(Driver.scala:102)
	at Chisel.Module$.Chisel$Module$$init(Module.scala:65)
	at Chisel.Module$.apply(Module.scala:50)
	at Chisel.Driver$$anonfun$executeUnwrapped$2.apply(Driver.scala:102)
	at Chisel.Driver$$anonfun$executeUnwrapped$2.apply(Driver.scala:102)
	at Chisel.Driver$.execute(Driver.scala:107)
	at Chisel.Driver$.executeUnwrapped(Driver.scala:102)
	at Chisel.Driver$.apply(Driver.scala:41)
	at Chisel.chiselMain$.run(hcl.scala:67)
	at lowrisc_chip.Run$.delayedEndpoint$lowrisc_chip$Run$1(LowRISCChip.scala:325)
	at lowrisc_chip.Run$delayedInit$body.apply(LowRISCChip.scala:303)
	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.App$$anonfun$main$1.apply(App.scala:76)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
	at scala.App$class.main(App.scala:76)
	at lowrisc_chip.Run$.main(LowRISCChip.scala:303)
	at lowrisc_chip.Run.main(LowRISCChip.scala)
[... 118 lines cut ...]
	at sbt.BuildCommon$class.toError(Defaults.scala:1943)
	at sbt.Defaults$.toError(Defaults.scala:38)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:719)
	at sbt.Defaults$$anonfun$runTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:717)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:226)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:235)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:226)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:748)
[error] (lowrisc_chip/compile:run) Nonzero exit code: 1
[error] Total time: 8 s, completed Aug 24, 2017 12:05:53 PM
/home/jawad/lowrisc-chip/Makefrag-build:9: recipe for target '/home/jawad/lowrisc-chip/vsim/generated-src/Top.DefaultL2Config.sv' failed
make: *** [/home/jawad/lowrisc-chip/vsim/generated-src/Top.DefaultL2Config.sv] Error 1

wsong83, 2017-08-24:
I think this line tells you what was wrong:

~~~
Caused by: cde.ParameterUndefinedException: Parameter DebugCtmScorBoardSize undefined.
~~~

Turing on `UseDebug` along is not enough to enable the Trace debugger. For a complex project, I think it is normally to expect multiple configuration parameters working together. This is the case for the trace debugger.

Actually we have provided configurations for this purpose. You are not supposed to change anything inside the `BaseConfig` unless you need to add some new peripherals or even more complex extensions.

For the trace debugger, all existing configuration targets derived from `WithDebugConfig` will have the trace debugger enabled. So I suggest you to see this `WithDebugConfig` class to understand how to properly config it.

wsong83, 2017-08-29:
Feel free to open it again.

closed 2017-08-29 by wsong83