lowRISC/lowrisc-chip #70: Question about release merging in the acqTracker

sxu55, 2017-09-21:
I don't think this is related to the deadlock issue last time.. But just a bit confused by the irel_can_merge signal in line 895 of acqTracker:

> val irel_can_merge = io.irel().conflicts(xact_addr_block) &&
>                          io.irel().isVoluntary() &&
>                          !Vec(s_idle, s_meta_read, s_meta_resp, s_meta_write).contains(state) &&
>                          !all_pending_done &&
>                          !io.outer.grant.fire() &&
>                          !io.inner.grant.fire() &&
>                          !pending_vol_ignt

It doesn't exclude s_busy state, which I think it should.

Consider such scenario: there are multiple idle cycles between voluntary release beats. Suppose the acquire tracker is in s_busy and has just finished the inner grant, waiting for the core finish signal. Now that a voluntary release of the same address comes, so the acqTracker accepts the release. Later, before the second release beat comes, the inner finish signal comes so the acqTracker returns to s_idle (should this happen?). In this case, I think one release beat is consumed by acqTracker, and the remaining 7 beats will go to voluntary relTracker, forcing the voluntary relTracker to wait for the last missing beat forever.

Do you think if this will happen?

wsong83, 2017-09-22:
You are now beginning to talk about some parts which I will be very hesitated to make changes.

I think I cannot rule it out. In theory, it might happen. However, I believe, in practice, it is nearly impossible? If such a Release will come after all Grants are sent back, it means:

1. The one who sends this later Release should be the one causing the Acquire previously and it is the owner of the cache line.
2. After all Grants are sent back to L1, L1 needs to replay the store operation. For it to release (replace) it, some other cache block in the same set is missed and the recent update cache block is to be replaced. Considering currently L1 is multi-way and LRU, it is not going to happen because the Acquire will be blocked and this recently update cache block should not be the one to be replaced? Unless the pseudo LRU is working badly right now or the number of way is actually 1.
3. There is not much to delay the s_busy state. Once all Grants are sent to L1, the only thing I can figure of that might delay s_busy -> s_meta_write/s_idle is that the data array is busy.

Possible fix:
Well, I am kind agree that this might be a problem. The original design might be too aggressive on parallelizing the various operations during the s_busy state. A safer choice might be delaying the inner Grant after the data array is allocated. Therefore, the s_busy state is guaranteed to be finished in time.  The original design seems to send out inner Grant as soon as possible whenever the outer Grant is received, reducing the L2 missing penalty, I believe.

As I said in our previous issue track, lowRISC is right now merely a maintainer of this L2 implementation. We try to avoid introducing any new bug if we are going to fix existing bugs. Also this L2 might be replaced in our new releases. So it would be better to keep the maintaining effort to the minimal. If you can make a test case that manifests such loss of Release beat, I would try to fix it, as then we have a test case to verify at least.

Thank you very much to raise this question!

sxu55, 2017-09-22:
Thanks for the detailed analysis, your analysis is really helpful. I agree with you that this actually shouldn't happen with proper coherence protocol. After a careful look at the waveform, I realized when I play with the coherence protocol design, I mistakenly force lines in shared status to also do data eviction, and violates SWMR constraint. Seems the L2 state machine is not designed to handle this special incoherent situation.

closed 2017-09-22 by sxu55