lowRISC/lowrisc-chip #71: L1 nbdcache bug?

sxu55, 2017-09-26:
In a potential corner case, I observed that L1 cache might drop a write:
Suppose we have MESI between L1 and L2, and a specific cache block in core 0 is in ExclusiveClean state. Then, a write to this cache block and a invalidateData probe of this cache block from other core come to the L1 cache simultaneously. What is happening is shown below as a timeline:

cpu_req write arrives --> probe_req arrives --> probe_req read meta (E-clean state) --> cpu_req allocate MSHR since E-clean needs to be changed into E-Dirty --> MSHR change meta from E-Clean to E-Dirty --> probeUnit change meta from E-Dirty to Invalid and send releaseAck --> MSHR replay the instr and write to data array.

The issue is in the last step: when MSHR has finished updating the meta from E-Clean to E-Dirty, the replay doesn't read the meta again but directly write to data array instead. At this time, due to the probe invalidation, the written data's corresponding meta has already been changed to invalid. The next time if the core reads the data, the result will be the stale version from L2 cache. I think a more appropriate handling is to fully replay this cpu write instr again (read the meta array again for the same block instead of directly jump to s3_valid), which will raise a cache miss and re-grab the block from L2 cache.

Since I am not fully confident with my observation here, I'd appreciate if someone can help me check if the last step will indeed happen as described here in the current Rocket nbdcache design.

wsong83, 2017-09-27:
Seems like if prober accepts a request, it will block all cpu requests for the same block.
See:
https://github.com/lowRISC/rocket/blob/f03a27c2033c5c3ddccadbbe8e686c377e2be89b/src/main/scala/nbdcache.scala#L1136

sxu55, 2017-09-27:
Sorry if this is a false positive bug report. I indeed saw that both s1_nack and s2_nack are high during this time. However, I also saw this in nbdcache line 1001:

> s3_valid := (s2_valid_masked && s2_hit || s2_replay) && !s2_sc_fail && isWrite(s2_req.cmd) && !s2_tag_xcpt

In this case, even though s2_valid_masked is low due to s2_nack, but since the logic is OR-ed with s2_replay (which is high even though prober is active), and if other conditions are met, s3_valid will still write the data array.

And I saw that s2_replay is taking in s1_replay, and s1_replay has the following assignment:

>  s1_replay := mshrs.io.replay.valid && readArb.io.in(1).ready && replay_enable

In the case I described, I don't think s1_replay will be nacked, and will let the replay slip through the pipeline to s3 if it is a write

wsong83, 2017-09-27:
Replay indeed has a higher priority than probe.

So if replay happens when prober is active, I suppose the prober is blocked by MSHR through this io.mshr_rdy signal.
https://github.com/lowRISC/rocket/blob/f03a27c2033c5c3ddccadbbe8e686c377e2be89b/src/main/scala/nbdcache.scala#L723
https://github.com/lowRISC/rocket/blob/f03a27c2033c5c3ddccadbbe8e686c377e2be89b/src/main/scala/nbdcache.scala#L386

Here the replay should be finished in the meta_hazard count cycles?

Oh, I am trying to make sure both of us are on the same page here.

wsong83, 2017-09-27:
I thin the key is, if prober is blocked, it should go back to read meta again. So if the meta is revised to Dirty by MSHR, the prober should have catched this update when it tries to release.

sxu55, 2017-09-27:
I took a snapshot of the waveform with essential signals for describing what is happening here, hopefully this can help:
[image]

sxu55, 2017-09-27:
Hmm. Just want to make sure we are on the same page here:

So the prober is allocated before the mshr (for promoting the E state) allocation. And unfortunately, mshr is doing the replay (denoted as mshr1 read above) one cycle before the prober update the meta to invalid, so mshr still think it has successfully promoted the block and let the write flow through the pipeline. Pls correct me if my interpretation is incorrect.

wsong83, 2017-09-28:
I think you are right. The blockage from MSHR is one cycle too late.
Would you please check my fix would fix the issue?

wsong83, 2017-09-28:
It seems like the upstream Rocket-Chip use the same logic. You may want to fire a bug report to them as well.
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/NBDcache.scala#L262

sxu55, 2017-09-28:
Cool. I think this fix will work by pulling the probe_rdy down one cycle earilier.
I will run the regression suite again to make sure this is fixed. Will report back soon.

Thanks a lot.

sxu55, 2017-10-03:
The regression suite passed with the proposed fix. I will submit an issue to ucb also.

wsong83, 2017-10-03:
Thanks for the feedback. I will soon have the fix merged to lowRISC master.

closed 2017-10-03 by wsong83