lowRISC/lowrisc-chip #72: PPA Comparison

riscveval, 2017-09-28:
Hi I am evaluating various RISCV Cores. I am getting the following results for Lowrisc V.4.0.

Core | ISA | Bit | Pipeline | LUTs | LUT RAMs | LUT FFs | BRAM | Power µW/MHz | Dhrystone DMIPS/   MHz | Coremark /MHz | Target FPGA
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Lowrisc_64 | RV64IMAFD | 64 | 6 | 49904 | 1943 | 26788 | 71 | 0.063 | 1.72 | 3.06 | Nexys-4 DDR

Is these results are correct or not? Can you share these information to me.

Can you update Dhrystone performance for Rocket Core?

I want to change the cache memory size for LowRISC. So can you share how to change the cache memory for LowRISC?

wsong83, 2017-09-28:
The default cache sizes are defined here:

L1 D$: https://github.com/lowRISC/lowrisc-chip/blob/master/src/main/scala/Configs.scala#L354
L2: https://github.com/lowRISC/lowrisc-chip/blob/master/src/main/scala/Configs.scala#L363

However, the default cache configuration can be overridden by derived configurations.

riscveval, 2017-09-28:
Hi can you share PPA results for lowRisc? Above mentioned results are correct or not? How to measure Dhrystone performance for lowRISC?

wsong83, 2017-10-09:
The design itself is targeting ASIC although perfectly runnable on FPGA. FPGA implementations are our demos for function only. There is no optimization done for improving performance. I have not run Dhrystone benchmark. Therefore, I cannot confirm any of the above either because I do not have the number or I do not think it is ultimately important.

jrrk, 2017-10-10:
> either because I do not have the number nor I do not think it is

With default cache settings the latest version of LowRISC achieves 2535 
dhrystones per MHz on Nexys4_DDR @ 25MHz

A recent PC on the same programme achieves 9681 dhrystones per MHz on 
Nexys4_DDR @ 3400MHz

However, the programme makes little or no use of floating point (apart 
from printing the time results).

Another historical benchmark achieves 1.47 MIPS/MHz on the same PC, and 
1.332 MIPS/MHz on the LowRISC.

In each case the system overhead was about 5%.

The question of why we do not target our FPGA implementations for 
performance is primarily one of cost of the hardware
and having long development build cycles leading to poor productivity. 
There are a number of CPUs out there which are oriented
towards performance, for example the MicroBlaze achieves 120MHz on FPGA. 
I do not know if its FPU can do that speed though.

These figures compare favourably with evaluations made by Chalmers for 
LEON2,MicroBlaze and OpenRISC in area optimized applications,
and is in the same ballpark for optimised configurations of MicroBlaze 
at more than 3 times the frequency.

http://gaisler.com/doc/Evaluation_of_synthesizable_CPU_cores.pdf

However those figures would be different if modern FPGA technology was used.

time ./stanford
Starting
     Perm  Towers  Queens   Intmm      Mm  Puzzle   Quick  Bubble 
Tree     FFT
       83     100      67      67      50     616      67      83 
650     117

Nonfloating point composite is        280

Floating point composite is        347