lowRISC/lowrisc-chip #74: Compile the RISC-V Linux and the ramdisk `root.bin`

thecakim, 2017-10-08:
at http://www.lowrisc.org/docs/untether-v0.2/linux_compile/
I tried to compile linux along this page, but I run the 
$ make ARCH = riscv -j vmlinux
 command in the first script block The following message

------------------------------------------------------------------------------------
 CHK include / config / kernel.release
  CHK include / generated / uapi / linux / version.h
  CC scripts / mod / devicetable-offsets.s
  CC scripts / mod / empty.o
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-m64'
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-msoft-float'
scripts / Makefile.build: 204: recipe for target 'scripts / mod / devicetable-offsets.s' failed
make [2]: *** [scripts / mod / devicetable-offsets.sys] Error 1
make [2]: *** Waiting for unfinished jobs ....
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-m64'
riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-msoft-float'
scripts / Makefile.build: 308: recipe for target 'scripts / mod / empty.o' failed
make [2]: *** [scripts / mod / empty.o] Error 1
scripts / Makefile.build: 455: recipe for target 'scripts / mod' failed
make [1]: *** [scripts / mod] Error 2
Makefile: 527: recipe for target 'scripts' failed
make: *** [scripts] Error 2
------------------------------------------------------------------------------------

I did not progress anymore as the phrase came up.

Maybe you should modify $ TOP / riscv-tools / linux-3.14.41 / arch / riscv / Makefile. Is there a way?

For reference,
on this page http://www.lowrisc.org/docs/untether-v0.2/
I ran 'Compile and install RISC-V cross-compiler' under '2. The development enviromnet' and
I was following 'Compile the RISC-V Linux and the ramdisk root.bin'.

jrrk, 2017-10-08:
You should use the script make_root.sh to compile the kernel and root 
filing system, as described in the Wiki.

If you want to compile manually, do not leave a space between ARCH=riscv 
and -j uses an integer argument which is the number of parallel jobs.

You need to compile our toolchain (an old version), also described in 
the Wiki, because the options have changed.

Also the verilog of the processor must be compiled first to get certain 
header files, before the kernel.

The v0.2 is rather old BTW.

wsong83, 2017-10-09:
Just to provide extra information, these two lines of the error message:

> riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-m64'
> riscv64-unknown-linux-gnu-gcc: error: unrecognized command line option '-msoft-float'

are direct indications that the versions of the compiler and the linux kernel do not match.