lowRISC/lowrisc-chip #78: BlackBox

clare7, 2017-11-17:
Would it be possible to use the blackbox for Verilog integration (such as BlackBoxInline, BlackBoxResource utility from Chisel 3) in this version of LowRiscv chip? If not any advice or alternative that could work for this chip?

wsong83, 2017-11-17:
We do use blackbox for the trace debugger, see this one for an example:
https://github.com/opensocdebug/hardware/blob/master/modules/ctm/riscv/rocket/ctm.scala

However, all the released lowRISC versions use Chisel 2 rather than Chisel 3, so the blackbox utilities mentioned in your questions are not available. Hopefully lowRISC will migrate to Chisel 3 soon (which is in the plan).

clare7, 2017-11-17:
Thanks! manage to get my code works.
Also found another useful reference:
https://groups.google.com/a/groups.riscv.org/forum/#!topic/hw-dev/7ztcExvdGFM

closed 2017-11-17 by clare7