lowRISC/lowrisc-chip #79: lowrisc-chip and rocket-chip

chinmayaPesonal, 2017-11-25:
Hi,

I am a hw developer and relatively new to open source processors.

Recently I started studying https://github.com/freechipsproject/rocket-chip to get access to open source RISC-V processor.  My aim is to understand the core so that I can customize and use it as a part of product idea we have.

As it would be expected, right now I am not very conversant with how everything is organized in the rocket-chip soc and I am relying on any little information I get from web-pages, google group and git-hub Issues page.

I have a general query for you guys.
What are the  'same' and 'different' things between lowrisc-chip and rocket-chip?
Does lowrisc chip in general has more documentation and easier to get hold of?

Thanks
Chinmaya

jrrk, 2017-11-25:
The LowRISC design is based on an old version of Rocket. In those days it had an L2 cache and use tilelink-1 instead of the current 2. This core was extensively modified with security extensions such as stack smashing protection and supplemented with real world peripherals (DDR memory controller, trace debugger, SD-card controller, Minion coprocessor, Ethernet, keyboard, VGA text display, and a corresponding port of Linux, together with various bare-metal tests. It also eliminates the requirement for a host processor.

chinmayaPesonal, 2017-11-25:
Thanks @jrrk .
Please  let me know if my understanding is correct.
For lowrisc-chip, some features were added to the old rocket processor, like stack smashing protection etc.
The lowrisc chip also has DDR memory controller, trace debugger, SD-card controller, Minion coprocessor, Ethernet, keyboard, VGA text display, and a corresponding port of Linux.  These are not part of rocket core/processor but peripherals in the soc, right?
Does rocket-chip has all these peripherals? Also the features that were improved in older version rocket core for lowrisc chip, are now part of new versions of rocket core?

-Chinmaya

jrrk, 2017-11-25:
While we were working on our improvements the Rocket development moved on to include compressed instructions, programmable interrupt, diplomacy, tilelink 2 etc, it would be difficult to retrofit our changes to the new Rocket, but relatively easy to put our peripherals around it. We intend to do the latter soon. There are other SOCs based on Rocket using Chisel for peripherals instead of our approach. There is also the BOOM (out of order machine).

chinmayaPesonal, 2017-11-25:
Thanks for quickly resolving my queries.
In rocket chip chisel environment, one can build different flavors of rocket core - like 32bit/64bit, with or without floating point unit.
Does lowrisc chip lets one choose between different flavours of the rocket core processor?

jrrk, 2017-11-25:
The configurations are stored in a file called Configs.scala, you can easily add more options yourself, I have not tried disabling the FPU, but I believe a similar syntax will work. The FPU is on the critical path for timing.