lowRISC/lowrisc-chip #87: Failed to run programs on FPGA

xubaqian, 2018-03-20:
I have booted Linux on FPGA and want to run my programs on it.
I use `riscv64-unknown-elf-gcc` to compile a simple hello world program and copy it into root.bin through make_root.sh. It successfully runs on FPGA.
But when I failed to run spec2000 and got error message like `Can't open file ***` or `Error opening file *** for a access`.
Does it due to lack some libraries ?
If I use `riscv64-unknown-linux-gnu-gcc` as compiler to run hello world , it would show me `/bin/ash: ./hello: not found`

In fact I encountered the same problem in another respository on vc706. That time I fixed this problem by using `riscv64-unknown-linux-gnu-gcc` to compile programs into filesystem of vmlinux. But this time I failed to boot linux when I used `make ARCH=riscv menuconfig` , and got following messages 
```
[    0.230000] TCP: cubic registered
[    0.230000] Freeing unused kernel memory: 1040K (ffffffff80000000 - ffffffff80104000)
[    0.230000] Failed to execute /init (error -8)
[    0.230000] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[    0.230000] CPU: 0 PID: 1 Comm: swapper Not tainted 3.14.41-g9a25e8d-dirty #6
[    0.230000] Call Trace:
[    0.230000] [<ffffffff80105f54>] walk_stackframe+0x0/0xc8
[    0.230000] [<ffffffff802b4ae4>] panic+0xb4/0x1c4
[    0.230000] [<ffffffff802b44f8>] rest_init+0x80/0x84
[    0.230000] [<ffffffff802b45ec>] kernel_init+0xf0/0xf4
[    0.230000] [<ffffffff802b44f8>] rest_init+0x80/0x84
[    0.230000] [<ffffffff802b44f8>] rest_init+0x80/0x84
[    0.230000] [<ffffffff80104bb8>] ret_from_syscall+0x10/0x14
```
I don't know why failed. Maybe I don't understand which determind the filesystem, root.bin or vmlinux.

wsong83, 2018-03-20:
Have you added the static compile option to gcc?
By default, libraries are not installed on the RISC-V Linux. All programs running in the RISC-V Linux need to be statically linked with all the library functions required.

Not sure about the kernel panic. Probably wrong configuration.

jrrk, 2018-03-20:
The LowRISC release ethernet-v0.5 is the first one that allows you to 
compile and run your own programs directly on the FPGA. Previous 
releases only provided a busybox environment which lacks the userland 
applications and libraries that you are looking for.

These libraries are deliberately omitted from the busybox environment 
because they have to be loaded remotely at startup which wastes RAM and 
causes a bigger delay in booting

User programs should not be copied into your boot filesystem for this 
reason. You will get 'not found' as one possible error due to lack of 
libraries.

If you follow the instructions here:
http://www.lowrisc.org/docs/ethernet-v0.5/fpga2/

Your compiler will be installed as riscv64-poky-linux-gcc

root.bin is a composite binary that contains Berkeley Boot Loader and 
vmlinux as a payload. Vmlinux contains the initial root filesystem as a 
cpio archive. Your compiler or user apps should not be placed in this 
area. Instead prepare the SD-card as described in the link above.

As Wei says, if you need to augment your boot environment for some 
reason this can be done by compiling static executables  or by adding 
more options to busybox if this contains the required functionality.

If you install the poky userland as described above all the usual 
options such as shared libraries will be enabled.

The on-FPGA compilation is quite slow. If you have substantial programs 
to compile this may be done by modifying the poky environment and 
running on the workstation. However this takes some hours (or days if 
you have a slow internet) to bootstrap from scratch.

Regards,
Jonathan

xubaqian, 2018-03-21:
I use `riscv64-unknown-linux-gnu-gcc` with `-static` option and successfully run my programs.
Thank you for your detailed explanation!

closed 2018-03-21 by xubaqian