lowRISC/lowrisc-chip #88: Failure during the toolchain build

noureddine-as, 2018-03-20:
Hi!
I would like to test the untethered version risc-v based SoC.

[image]
When i clone the repo and proceed to build the toolchain, i get this error, not sure if it's due to my network limitations or to something else?

I would like to understand please if the lowRISC SoC depends on a particular version of riscv-tools (gnu compilation toolchain / spike ...etc)? Or can I use a recent compilation toolchain with an old version of lowRISC? I mean by that if i change my hardware configuration (number of cores / accelerators / cache ...etc) do I need to rebuild the software toolchain/simulators? 

Thanks for taking time to respond my questions.
Best regards.

jrrk, 2018-03-20:
> image

That error could be caused by problems with nameservers or slow 
international links.

You cannot use the latest toolchain with lowrisc because it does not 
support compressed instructions and the ISA has changed (in particular 
control and status registers), and privilege instruction set.

However you do not need to recompile if you change the Knobs (number of 
cores etc.) as the ISA will be the same.

noureddine-as, 2018-03-20:
It turned out that it's a network issue, i connected from another and that works.

Thanks for your answers @jrrk I'll dive deeper in the topic.