lowRISC/lowrisc-chip #89: make bitstream failed

xubaqian, 2018-03-21:
My lowrisc-chip version: minion-v0.4, vivado version: 2015.4
I want to generate bitstream in kc705 but failed. I got the following message:
```
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7k325t'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7k325t'
ERROR: [Synth 8-1031] dii_package is not declared [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:3]
ERROR: [Synth 8-1766] cannot open include file consts.vh [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:5]
ERROR: [Synth 8-1766] cannot open include file dev_map.vh [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:6]
ERROR: [Synth 8-2841] use of undefined macro ROCKET_MEM_TAG_WIDTH [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:180]
ERROR: [Synth 8-2841] use of undefined macro ROCKET_PADDR_WIDTH [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:181]
ERROR: [Synth 8-2841] use of undefined macro ROCKET_MEM_DAT_WIDTH [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:182]
ERROR: [Synth 8-2841] use of undefined macro ROCKET_IO_TAG_WIDTH [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:485]
ERROR: [Synth 8-2841] use of undefined macro ROCKET_IO_DAT_WIDTH [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:487]
INFO: [Synth 8-2350] module chip_top ignored due to previous errors [/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv:9]
Failed to read verilog '/home/xqz/tmp/lowrisc-chip/src/main/verilog/chip_top.sv'
INFO: [Common 17-83] Releasing license: Synthesis
3 Infos, 0 Warnings, 0 Critical Warnings and 9 Errors encountered.
```
But I didn't encounter any problem in nexys4_ddr. What's wrong ?

xubaqian, 2018-03-21:
OK, thanks

closed 2018-03-21 by xubaqian

xubaqian, 2018-03-23:
Few days ago I ported LowRISC untether-v0.2 to VC707 board on the basis of kc705 directory, but that was just a trial and I need to port minion v0.4 to VC707.
Using the same method, I copy the directory of nexys4_ddr ,rename to vc707 and modify some files:

Then make vivado, in implementation stage, I got 1 error and 2 critical warnings
```
[Vivado 12-1387] No valid object(s) found for set_false_path constraint with option '-from [get_clocks clk_io_uart_clk_wiz_0]'. ["/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/constraint/timing.xdc":1]
[Vivado 12-1387] No valid object(s) found for set_false_path constraint with option '-from [get_clocks mmcm_clkout0]'. ["/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/constraint/timing.xdc":2]
[Vivado_Tcl 4-23] Error(s) found during DRC. Placer not run.
```
On the terminal,
```
Writing placer database...
Writing XDEF routing.
Writing XDEF routing logical nets.
Writing XDEF routing special nets.
Write XDEF Complete: Time (s): cpu = 00:00:00.15 ; elapsed = 00:00:00.04 . Memory (MB): peak = 2713.605 ; gain = 0.000 ; free physical = 6728 ; free virtual = 21499
write_checkpoint: Time (s): cpu = 00:00:08 ; elapsed = 00:00:06 . Memory (MB): peak = 2713.605 ; gain = 0.000 ; free physical = 6723 ; free virtual = 21497
INFO: [DRC 23-27] Running DRC with 8 threads
INFO: [Coretcl 2-168] The results of DRC are in file /home/xqz/tmp/lowrisc-chip/fpga/board/vc707/lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top_drc_opted.rpt.
INFO: [Vivado_Tcl 4-424] Cannot write hardware definition file as there are no IPI block design hardware handoff files present
Command: place_design
Attempting to get a license for feature 'Implementation' and/or device 'xc7vx485t'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7vx485t'
INFO: [DRC 23-27] Running DRC with 8 threads
INFO: [Vivado_Tcl 4-198] DRC finished with 0 Errors
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
Running DRC as a precondition to command place_design
INFO: [DRC 23-27] Running DRC with 8 threads
INFO: [Vivado_Tcl 4-198] DRC finished with 1 Errors, 1 Critical Warnings
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-23] Error(s) found during DRC. Placer not run.
INFO: [Common 17-83] Releasing license: Implementation
53 Infos, 16 Warnings, 2 Critical Warnings and 1 Errors encountered.
place_design failed
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.

INFO: [Common 17-206] Exiting Vivado at Fri Mar 23 15:09:09 2018...
[Fri Mar 23 15:09:09 2018] impl_1 finished
wait_on_run: Time (s): cpu = 00:00:00.07 ; elapsed = 00:01:22 . Memory (MB): peak = 1041.188 ; gain = 0.000 ; free physical = 8364 ; free virtual = 23144
INFO: [Common 17-206] Exiting Vivado at Fri Mar 23 15:09:09 2018...
```

So I open chip_top_drc_opted.rpt, and get the following message
```
REQP-123#1 Error
connects_CLKINSEL_VCC_connects_CLKIN1_ACTIVE  
dram_ctl/u_mig_7series_0_mig/u_iodelay_ctrl/clk_ref_mmcm_gen.mmcm_i: The MMCME2_ADV with CLKINSEL tied high requires the CLKIN1 pin to be active.
Related violations: <none>

REQP-161#1 Error
connects_CLKINSEL_VCC_connects_CLKIN1_ACTIVE  
dram_ctl/u_mig_7series_0_mig/u_ddr3_infrastructure/plle2_i: The PLLE2_ADV with CLKINSEL tied high requires the CLKIN1 pin to be active.
Related violations: <none>

IOSTDTYPE-1#1 Critical Warning
IOStandard Type  
I/O port clk_n is Single-Ended but has an IOStandard of LVDS which can only support Differential
Related violations: <none>

NSTD-1#1 Critical Warning
Unspecified I/O Standard  
8 out of 155 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: sd_dat[3:0], clk_p, sd_sclk, sd_cmd, sd_reset.
Related violations: <none>

UCIO-1#1 Critical Warning
Unconstrained Logical Port  
7 out of 155 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: sd_dat[3:0], sd_sclk [... cut ...]
It seems problem occurs in the first two line of constraint/timing.xdc.
And I found minion_subsystem directory is related to SD card controller and minion_soc.sv defines some parameters of fpga pins. Note that I didn't assign location constraint of SD card because I failed many times and seems parameter iodrv = 24 defines the drive strength.

In conclusion, maybe I need to modify the clock, pin parameter and minion_subsystem. I need guidance.

Plus: I don't need some peripherals. Boot the linux and run some benchmarks are enough.

jrrk, 2018-03-23:
The KC705 support has not been updated for debug-v0.3 onwards. Most of 
the peripherals (high speed UART mode, VGA, keyboard would need 
substantial adaptation to work with the KC705 or VC707. Even the 
Ethernet PHY is significantly different. If you disable these 
peripherals in your Chisel config you will get invalid constraints 
messages, this is normal. The majority of your errors seem to be 
associated with the clock wizard. If you recustomize the clock wizard 
with defaults that are appropriate for your board (such as differential 
input), then these errors should go away. At the moment it seems clk_n 
which is your differential clock negated is not connecting 
differentially to the clock wizard. This can easily be fixed by 
reference to a known good example design for the VC707. Look out for a 
.xci file that you can import and then customize. If you are running 
benchmarks, you would be better off using the ethernet-v0.5 release, the 
Linux device drivers have been cleaned up a lot, and the performance of 
peripherals improved.

xubaqian, 2018-03-25:
Clock wizard？ I see clk_wiz_0 for clock generators and clk_wiz_1 for SD-card clock generator in /fpga/board/vc707/script/make_project.tcl , but I don't know what parameters should I configure with according to vc707 user guide, is there docs about it ？

jrrk, 2018-03-25:
Search online for VC707 example MIG (memory interface generator) design. You will be able to build a project which you can then do ‘save as tcl’ to create a script. This should then be edited down to just what you need. You can then add any clocks that our design needs with the customise option.

xubaqian, 2018-03-26:
I'm still puzzled and about my errors and haven't customise my clock.
In chip_top_drc_opted.rpt, I got 2 errors:
```
REQP-123#1 Error
connects_CLKINSEL_VCC_connects_CLKIN1_ACTIVE  
dram_ctl/u_mig_7series_0_mig/u_iodelay_ctrl/clk_ref_mmcm_gen.mmcm_i: The MMCME2_ADV with CLKINSEL tied high requires the CLKIN1 pin to be active.
Related violations: <none>

REQP-161#1 Error
connects_CLKINSEL_VCC_connects_CLKIN1_ACTIVE  
dram_ctl/u_mig_7series_0_mig/u_ddr3_infrastructure/plle2_i: The PLLE2_ADV with CLKINSEL tied high requires the CLKIN1 pin to be active.
Related violations: <none>
```
And the first two lines in /fpga/board/vc707/constraint/timing.xdc will cause 2 critical warnings in Run Implematation stage:
```
 [Vivado 12-1387] No valid object(s) found for set_false_path constraint with option '-from [get_clocks clk_io_uart_clk_wiz_0]'. [/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/constraint/timing.xdc:1]
Resolution: Check if the specified object(s) exists in the current design. If it does, ensure that the correct design hierarchy was specified for the object. If you are working with clocks, make sure create_clock was used to create the clock object before it is referenced.
 [Vivado 12-1387] No valid object(s) found for set_false_path constraint with option '-from [get_clocks mmcm_clkout0]'. [/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/constraint/timing.xdc:2]
Resolution: Check if the specified object(s) exists in the current design. If it does, ensure that the correct design hierarchy was specified for the object. If you are working with clocks, make sure create_clock was used to create the clock object before it is referenced.
```
If I delete the two lines, these warings would disappear.

You said my errors seems to be associated with the clock wizards, and I guess they are in make_project.tcl
```
# Clock generators
create_ip -name clk_wiz -vendor xilinx.com -library ip -module_name clk_wiz_0
set_property -dict [list \
                        CONFIG.PRIMITIVE {PLL} \
                        CONFIG.CLKOUT1_REQUESTED_OUT_FREQ {200.000} \
                        CONFIG.RESET_TYPE {ACTIVE_LOW} \
                        CONFIG.CLKOUT1_DRIVES {BUFG} \
                        CONFIG.MMCM_DIVCLK_DIVIDE {1} \
                        CONFIG.MMCM_CLKFBOUT_MULT_F {10} \
...
```
How do I customize my clock wizards? I'm a beginner.

jrrk, 2018-03-26:
Use make vivado to open the gui. Then click on the IP tab, then right click on the appropriate IP, the choose customise. Based on the error messages I would say you need to switch to a differential clock input.

xubaqian, 2018-03-28:
Previous error originates from my incorrect dram_ctrl. Now I still get the two critical warning ,but can generate bitstream. Download it to VC707 board, I just get a messy code �  from the UART when I click ‘Program device’ in vivado. I don't know why.

I'm not so sure about pin_plan.xdc because I didn't find detailed user guide for nexy4 , but [VC707 does](https://www.xilinx.com/support/documentation/boards_and_kits/vc707/ug885_VC707_Eval_Bd.pdf). According to [xc7a100tcsg324pkg.txt](https://www.xilinx.com/support/packagefiles/a7packages/xc7a100tcsg324pkg.txt) and [xc7vx485tffg1761pkg.txt](https://www.xilinx.com/support/packagefiles/v7packages/xc7vx485tffg1761pkg.txt), I can find the pin and pin_name for nexy4, then search the same pin_name in vc707 and finally get the corresponding pin. However, the pin in vc707 didn't match what I thought it should be in [vc707 user guide](https://www.xilinx.com/support/documentation/boards_and_kits/vc707/ug885_VC707_Eval_Bd.pdf). If I use the pin from the vc707 pdf , nothing would be output. If I use the pin from the txt, I got a a messy code � output.

jrrk, 2018-03-28:
You probably need to change the UART baud rate divider, verilog simulation (make simulation) will tell you if the clock wizard is set correctly, bearing in mind the master clock will be higher on the VC707. Nexys4ddr documentation is available on the digilent website, naming between manufacturers is not always consistent.

xubaqian, 2018-03-29:
`make hello`, `make simulation`, then got an error
```
Makefile:182: recipe for target 'lowrisc-chip-imp/lowrisc-chip-imp.sim/sim_1/behav/elaborate.log' failed
make: *** [lowrisc-chip-imp/lowrisc-chip-imp.sim/sim_1/behav/elaborate.log] Error 1
```
in lowrisc-chip-imp/lowrisc-chip-imp.sim/sim_1/behav/elaborate.log, error message:
`ERROR: [VRFC 10-2063] Module <sd_card> not found while processing module instance <sdflash1> [/home/xqz/tmp/lowrisc-chip/src/test/verilog/chip_top_tb.sv:213]`
so I delete sd_card and sdflash1 mudule in chip_top_tb.sv
```
/**
   sd_card
   sdflash1
      (
      .sdClk ( sd_sclk ),
      .cmd   ( sd_cmd  ),
      .dat   ( sd_dat  )
      );
*/
```

 `make simulation` again
```
source xsim.dir/tb_behav/xsim_script.tcl
# xsim {tb_behav} -maxdeltaid 10000 -autoloadwcfg -tclbatch {/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/script/simulate.tcl} -key {Behavioral:sim_1:Functional:tb}
Vivado Simulator 2015.4
xsim.dir/tb_behav/xsimk: /home/xqz/xilinx/Vivado/2015.4/lib/lnx64.o/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/xqz/tmp/lowrisc-chip/fpga/board/vc707/lowrisc-chip-imp/lowrisc-chip-imp.sim/sim_1/behav/xsim.dir/xsc/dpi.so)
ERROR: [Simtcl 6-50] Simulation engine failed to start: Simulation exited with status code 1.
Please see the Tcl Console or the Messages for details.
xsim%
```
I find libstdc++.so.6 in /usr/lib/x86_64-linux-gnu/, and replace /home/xqz/xilinx/Vivado/2015.4/lib/lnx64.o/libstdc++.so.6 with it, but still fail to start simulator
```
source xsim.dir/tb_behav/xsim_script.tcl
# xsim {tb_behav} -maxdeltaid 10000 -autoloadwcfg -tclbatch {/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/script/simulate.tcl} -key {Behavioral:sim_1:Functional:tb}
Vivado Simulator 2015.4
ERROR: [Simtcl 6-50] Simulation engine failed to start: Simulation exited with status code 1.
Please see the Tcl Console or the Messages for details.
xsim%
```
If click run simulation in GUI, failure occurs in elaborate stage:
` [USF-XSim-62] 'elaborate' step failed with error(s) while executing '/home/xqz/tmp/lowrisc-chip/fpga/board/vc707/lowrisc-chip-imp/lowrisc-chip-imp.sim/sim_1/behav/elaborate.sh' script. Please check that the file has the correct 'read/write/execute' permissions and the Tcl console output for any other possible errors or warnings.`

jrrk, 2018-03-29:
This error arises because dpi.so was compiled with a different libstdc++ from the Verilog modules. Once you start modifying the Xilinx installation its game over I’m afraid because of the fragile base class problem, libstdc++ from different distributions cannot be mixed. You will need to reinstall the Xilinx Vivado software from a backup. 

To proceed I would remove dpi.so from your project and just use a simple behavioural RAM model instead.

Alternatively if you have access to a commercial simulator such as Synopsys VCS, you can choose this option in Vivado and output a simulation script directly (from the export simulation menu).

A final option if you want to stick with dpi is to run in a virtual machine, something like Centos-6, a (free knock off of a) qualified distribution 

To fix the problem properly you should ensure that dpi.so is compiled in the same environment that Vivado uses, but dpi is not needed for what you are trying to do.

wanming2008, 2019-09-05:
> `make hello`, `make simulation`, then got an error
>
> ```
> Makefile:182: recipe for target 'lowrisc-chip-imp/lowrisc-chip-imp.sim/sim_1/behav/elaborate.log' failed
> make: *** [...]

hi:徐八千.
Did you successfully run lowrisc on vc707

xubaqian, 2019-09-05:
> hi:徐八千.
> Did you successfully run lowrisc on vc707

I gave up.

jrrk, 2019-09-05:
I’m sorry you are having problems with isim. This code hasn’t been developed to the same extent as the synthesis. You can use the export to external simulator option of Vivado, but again you need to be careful of library paths.