lowRISC/lowrisc-chip #91: Segmentation Error

clare7, 2018-05-21:
Hi, I keep facing the error when I run  the 

<<CONFIG=Nexys4DebugConfig make selftest>>
it has no problem with <<make bitstream>>
so the problems only came at make target. Any advice?

INFO: [Common 17-83] Releasing license: Implementation
write_bitstream: Time (s): cpu = 00:01:05 ; elapsed = 00:01:13 . Memory (MB): peak = 3507.039 ; gain = 0.000 ; free physical = 8960 ; free virtual = 117162
INFO: [Vivado_Tcl 4-395] Unable to parse hwdef file chip_top.hwdef
INFO: [Common 17-206] Exiting Vivado at Mon May 21 10:06:51 2018...
[Mon May 21 10:06:55 2018] impl_1 finished
wait_on_run: Time (s): cpu = 00:00:00.26 ; elapsed = 00:10:15 . Memory (MB): peak = 1052.715 ; gain = 0.000 ; free physical = 11447 ; free virtual = 119649
INFO: [Common 17-206] Exiting Vivado at Mon May 21 10:06:55 2018...
vivado -mode batch -source ../../common/script/search_ramb.tcl -tclargs lowrisc-chip-imp > search-ramb.log
python ../../common/script/bmm_gen.py search-ramb.log src/boot.bmm 128 65536
data2mem -bm src/boot.mem -bd src/boot.mem -bt lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.bit -o b lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit
/opt/Xilinx/SDK/2015.4/bin/loader: line 164: 16352 Segmentation fault      (core dumped) "$RDI_PROG" "$@"
Makefile:200: recipe for target 'lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit' failed
make[1]: *** [lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit] Error 139
make[1]: Leaving directory '/home/mmwong/Documents/lowrisc-chip/fpga/board/nexys4_ddr'
Makefile:230: recipe for target 'selftest' failed
make: *** [selftest] Error 2

wsong83, 2018-05-21:
The error is related to the data2mem command provided by Xilinx SDK
> data2mem -bm src/boot.mem -bd src/boot.mem -bt lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.bit -o b lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit
> /opt/Xilinx/SDK/2015.4/bin/loader: line 164: 16352 Segmentation fault (core dumped) "$RDI_PROG" "$@"

I suggest you to check the grammar for this command and whether all files needed are accessible and valid (not corrupted somehow). Regenerate each file if necessary. Run the command alone in shell (without the Make/Python) can be helpful as well. If all those failed, try another version of Xilinx, (downgrade is more promising than upgrade)

clare7, 2018-05-21:
all the files are valid and available and running the command alone is successful

`data2mem -bm src/boot.mem -bd src/boot.mem -bt lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.bit -o b lowrisc-chip-imp/lowrisc-chip-imp.runs/impl_1/chip_top.new.bit`

i have check the `make bit-update` command and the syntax/grammar of the command is correct

clare7, 2018-05-21:
Ok thanks. It works after fixing the LD_LIBRARY_PATH issue.

closed 2018-05-21 by clare7