lowRISC/lowrisc-chip #93: iomshr stores to axi-lite slave

GuillemCabo, 2018-08-31:
Hi,
I'm using the old untether-v0.2.I have added an axi-lite peripheral to the simulation and I'm having problems to write to it. Reads seem fine.

After writing, the registers of the peripheral change, and the bvalid signal is emitted. After this instruction the processor hangs. No exceptions are risen. This happens with all the store instructions. 
In the image below you can see a _store double_ instruction to the address 0x8fff0000.
[image]

I have noticed that io_replay_valid doesn't change after the write. That is intended to be like that?
[image]

I have seen that in v0.3 some bugs were fixed in the IOMSHR state machine, [and this one](https://github.com/lowRISC/rocket/commit/2b5831d9226258f15d28d16135da464f30aac603),address finishes of stores. I was wondering if is the same problem. 

Can you give more ideas of what to try, possible sources of the problem, and which is the expected behaviour of the IOMSHR?
I think that the obvious solution is move to v0.4 or v0.5 but i will like to stick with the old one for the moment.

Thank you in advance.

jrrk, 2018-08-31:
You could try instantiating an AXI protocol checker in Vivado. Our small team lacks the resources to retrofit bug fixes to old releases,
So I’m interested to understand why you don’t want to upgrade (since we are preparing the refresh-v0.6 release shortly).

GuillemCabo, 2018-09-04:
Hi,
Thanks for the advice. Turn out that the problem was with the memory map.
Now is working fine. Io_replay_valid was working as expected. 

I have noticed that depending of  what are you doing VIVADO generates unencrypted or encrypted files for the Protocol checker.

About upgrade it, I'm uncertain. I tried V.04 but i couldn't compile the risc-v toolchain with opensuse Tubleweed but that is an other history. I guess that i have to check the new modules and  resources.
Will it be compatible with KC705?

Thank you.

jrrk, 2018-09-04:
There are many differences in KC705 especially in Ethernet and Video hardware that did not make it into the v0.3-v0.5 releases. The ongoing work on Ethernet, equivalent to the 0.6 release is happening on the kc705_mii branch. No work has been done on video support and it is not sufficiently polished to merge into v0.6 at the moment. In principle this release can support two cores from an FPGA area constraint point of view but we still need to solve the lack of L2 cache issue and any software coherency issues. If that is something that interests you then feel free to work on it and we can try to merge support for this later.

Regards,
Jonathan

GuillemCabo, 2018-09-05:
How much work will be required to add video support? I might be interested to give a try. 
Is possible disable the debug components and instantiate more cores? 

Best,
Guillem

jrrk, 2018-09-05:
The video on KC705 is HDMI and as such is proprietary. Xilinx have a non-free IP, there is some doubt about the specifications of the driver chip, whether they are freely available. Bearing in mind the content protection features of HDMI, it seems unlikely the full speculations would come into common knowledge. However there are Verilog modules such as 

https://github.com/cliffordwolf/SimpleVOut?files=1

which on further investigation might help you reach this goal.

closed 2018-09-12 by GuillemCabo