lowRISC/lowrisc-chip #94: Does refresh-v0.6 support multicore on fpga?

LuXuKun, 2018-09-03:
Hi,
I wonder which kind of board refresh-v0.6 supports and whether the board supports multicore configs. I have tried minion-v0.4 which use nexys4_ddr. It seems nexys4_ddr can only support single core rocket while bitstream generated with multicore config cannot be written into fpga board.
Thanks in advance.

jrrk, 2018-09-03:
You are correct, nexys4_ddr is too small to support more than one core. The kc705 board has the capacity to support multiple cores but we don’t plan to merge support for this configuration into refresh-v0.6 at the moment, partly due to the lack of an L2 cache and partly due to all the implications for software.

jrrk, 2018-10-30:
> Since I cannot find any relevant information on the lowRISC page,
> could you please tell me which version officially supports multicore
> Linux booting on the kc705 board? Is there any relevant guide to
> follow or anyone that had already attempted the specific "experiment"?

We don't support multicore Linux booting at the moment. However if we 
were to support it something like the kc705 would be

needed to cope with the increased size of the netlist. The main feature 
that is missing is a coherent L2 cache to support atomic

synchronization between cores. If you feel like experimenting with this 
environment then you need to checkout the kc705_mii branch.

You then need to adjust your Chisel parameters to invoke the 
DualCoreConfig instead of the DefaultConfig, these settings are in

rocket-chip/src/main/scala/system/Configs.scala. Then you can build the 
FPGA, launch Linux, and watch it crash. That's as far as I've got.

If you want to get further I think it will be a steep learning curve, 
and you may prefer to checkout SiFive's products where these issues have

already been ironed out. Or perhaps if you investigate FireSim they will 
have something useful to help you make progress.

The older version of Rocket, used on LowRISC releases v0.1-v0.5, did 
have an L2 cache. It cannot easily be ported to the latest Rocket

because it uses TileLink-1 and the latest is TileLink-2. You would have 
to go right back to release v0.2 to find a version that runs on the kc705.

It was certainly possible with this version to dial up a dual-core 
configuration, but I don't know if it ever booted Linux because the main

motivation for moving to TileLink-2 was overcoming bugs in the 
multi-processor support.

beatsnbytes, 2018-10-31:
I see....thank you for your very informative reply! I was actually working with the "untethered v2.0" release of LowRISC and after specifying a dual-core configuration (through the .scala file "NTILES=2 & NBANKS=2") the linux boot did not even start. (stuck at the printf specified at the boot.c file).
What about bare metal multicore support? Should all N cores operate normally or it would need some tinkering (entry points .etc?)

jrrk, 2018-10-31:
That version (v0.2) is so far out of date in terms of limitations, ABI 
changes and CSR register number changes, it's not really worth

investing much time on in my opinion, but if you insist on using it then 
the way it is supposed to work is both harts (hardware threads)

start at the same entry point, some very early code checks the hart ID, 
and parks the core if the hart ID is non-zero.

Subsequent unparking and cooperation occurs using IPI commands, which 
are essentially software interrupts between cores.

We don't think it is effective to support this usage. There is a reason 
why almost three years worth of updates have happened upstream

and in our own repositories.

beatsnbytes, 2018-10-31:
Yes, I have indeed noted that behaviour of parking one of the cores in the initialization process. So to clarify it, the latest versions of LowRISC even though they do not support multicore linux booting can run bare metal benchmarks with the cooperation of more than one cores.

jrrk, 2018-10-31:
I don't think I am quite saying that. I am saying that the LowRISC 0.6 
release contains a recent (Mar-18) Rocket Chip Generator.

This generator has the capability for multiple cores. With a large FPGA 
such as the kc705 (preliminary support for which is

included on the kc705_mii branch) there is sufficient physical capacity 
to incorporate a multi-core Rocket. This Rocket will

have IPI but it won't have a TileLink-2 coherent level-2 cache. This 
prevents SMP Linux from running. You might be able to get

non-SMP Linux running (for example with one core dedicated to I/O), but 
none of these options (apart from single core on the kc705)

has been tested by us, and as such you are welcome to contribute, but I 
fear we can not be much help.

beatsnbytes, 2018-10-31:
Ok thank you again! I have a much more demystified picture of the whole SMP subject now.

closed 2019-03-04 by LuXuKun

nicolast0604, 2019-04-25:
Thanks for your discussion above.
Hi I am trying to configure SMP in Linux and porting to a quard core rocket chip to some FPGA boards.
Does it mean that KC705 lack of L2 cache so it is not workable.

Are there any FPGA board which is workable to port a quard rocket chip and boot Linux SMP on it?

jrrk, 2019-04-25:
The L2 cache is not a property of the FPGA board, it is a property of the RTL 
design. For an example of

a RISCV SMP Linux design, you can consult 
https://github.com/PrincetonUniversity/openpiton

This supports a Genesys2 board, which is the same FPGA as the KC705 board. You 
will need to make some

adaptation such as adjusting the board constraints file. I haven't actually 
tried this code myself yet.

This RTL does not use Rocket, it uses Ariane which is a European Rocket written 
in System Verilog rather than Chisel.

It has no connection to the European Space Agency as far as I am aware, it comes 
from ETHZ in Switzerland.

LuXuKun, 2019-04-25:
> Thanks for your discussion above.
> Hi I am trying to configure SMP in Linux and porting to a quard core rocket chip to some FPGA boards.
> Does it mean that KC705 lack of L2 cache so it is not workable.
> 
> Are there any FPGA board which is workable to port a quard rocket chip and boot Linux SMP on it?

LowRISC supports L2Cache. The matter is nexys_4 board doesn't have enough resource for multi-core. 

Maybe you can have a try with sifive/freedom project. This project uses VC707 board and supports quad core rocket chip, as well as running Linux SMP on it. Link is here: https://github.com/sifive/freedom/.

Sifve/freedom project doesn't support L2 Cache. And VC707 board is a little expensive.

jrrk, 2019-04-25:
An older version of LowRISC did support an L2 cache and the possibility of 
multiple CPUs.

But this was based on Tilelink-1 which was known to be buggy and did not support 
SMP correctly.

The only open RISCV L2 SMP project that I am aware of is the OpenPiton one (they 
call it L1.5).

The L2 cache in the new Rocket is commercial and hasn't been released as open 
source.

LuXuKun, 2019-04-25:
Oh, sorry I gave a wrong answer. Did not notice your replay that lowRISC v0.6 doesn't support L2 Cache now. How this won't mislead others.

LuXuKun, 2019-04-25:
Oh, sorry I gave a wrong answer. Did not notice jrrk's replay that lowRISC v0.6 doesn't support L2 Cache now. Hope this won't mislead others.

nicolast0604, 2019-04-26:
Hi Jrrk and LuXuKun 
Thanks for your help answering this question.
Does it mean that if I would like to a boot a quad core rocket chip, I should add 
L2 cache RTL design by myself? 

How much about the complexity and effort do you estimate?

Is there any company or personal who can do this design service ?

Regards,
Nicolas

jrrk, 2019-04-26:
Absolutely, SiFive have a four core (actually 5, but the 5th core is non SMP), 
development board called the HiFive unleashed.

you can see their core selection page here:

https://scs.sifive.com/core-designer/

Obviously this a commercial offering, did I interpret your question correctly ?

My understanding is (this may change) SiFive have no plans to open-source their 
L2 cache,

but their low-end U54-MC 64-bit quad core is derived from, and very similar to 
Rocket.

If you are looking for an open-source solution Ariane+Openpiton is the closest 
thing that I am aware of,

but it is not really stable yet.