lowRISC/lowrisc-chip #95: boot procedure for kc705_mii branch

GuillemCabo, 2018-09-14:
Hi,
I have executed some baremetal examples and i want to give a go to boot linux.
I have seen that by default it tries too boot through Ethernet. Is that right?
```
�Calling main with MAC = eee1:e2e3e4e0
Selftest iteration 1, next buffer = 0, rx_start = 4000
rx_frq = 125002230
tx_frq = 2500040
```
I have take a look to the ethernet-v0.5 release notes but seems that is not exactly the same procedure.

Thank you

jrrk, 2018-09-14:
This is an experimental branch, I don't have a solution for the case 
where the host end is trying

to enforce gigabit communications (125MHz rx frequency). If your host is 
Linux you can force it to

only advertise 100BaseT using eth-tool and/or mii-tool, which might 
solve the problem.

You should be able to build the SD-card booter using make mmc 
program-updated.

I am in the process of documenting the procedures for refresh-v0.6 
branch of which kc705_mii is a variant.

It needs more development to reach stability though it is an interesting 
project. Part of the problem is the 1000BaseT

PHY which seems to have some undocumented behaviour that needs 
investigating. In addition it uses a separate

master clock which is asynchronous to the system clock, and transmit and 
receive frequencies are slightly different

also. If some willing developer had a brainstorm they could probably 
solve many of these problems relatively easily.

GuillemCabo, 2018-09-18:
Hi,
Thank you for your help.
I have seen your recent work documenting the[ boot process for refresh 0.6.](https://www.cl.cam.ac.uk/~jrrk2/docs/)
That guide made my day. The only problem that I have found has been with the recipes for **mkfs,fatdisk,extdisk**. In OpenSUSE they were not getting the names of the device right. I just replaced with VIM as:

```
104 mkfs: $(CARDMEM).log
105     sudo mkfs -t msdos /dev/$(USB)1
106     sudo mkfs -t ext4 /dev/$(USB)2
107     sudo mkswap /dev/$(USB)3
108     sudo mkfs -t ext4 /dev/$(USB)4
```

jrrk, 2018-09-18:
Dear Guillem,
  That particular case would work, but you also need to be able to iterate over all mounted partitions...

closed 2018-10-05 by GuillemCabo