lowRISC/lowrisc-chip #97: Issue building Debian for riscv-64.

marthinwurer, 2018-11-03:
I'm trying to build everything in the baremetal toolchain for v0.6, and I'm having some problems with the Debian step.

I follow all the commands here: https://www.lowrisc.org/docs/overview/

The commands that I ran are as follows (I checked out a new repo and everything):
```
sudo apt update
sudo apt upgrade

sudo apt-get install autoconf automake autotools-dev curl \
libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison \
flex texinfo gperf libncurses5-dev libusb-1.0-0-dev libboost-dev \
swig git libtool libreadline-dev libelf-dev python-dev \
microcom chrpath gawk texinfo nfs-kernel-server xinetd pseudo \
libusb-1.0-0-dev hugo device-tree-compiler zlib1g-dev libssl-dev \
debootstrap debian-ports-archive-keyring qemu-user-static iverilog \
openjdk-8-jdk-headless iperf3 libglib2.0-dev libpixman-1-dev

git clone -b refresh-v0.6 --recursive https://github.com/lowrisc/lowrisc-chip.git

sudo ln -s /mnt/c/nobackup/Xilinx/ /opt/Xilinx

cd /mnt/c/nobackup/lowrisc-chip

source /opt/Xilinx/Vivado/2018.2/settings64.sh
unset LD_LIBRARY_PATH

source set_env.sh

# https://www.lowrisc.org/docs/generate-the-bitstream/
cd $TOP/fpga/board/nexys4_ddr
make cleanall
make bitstream

make cfgmem
make program-cfgmem

# https://www.lowrisc.org/docs/build-bare-metal-toolchain/
cd $TOP/rocket-chip/riscv-tools/
bash ./build.sh
cd $TOP/rocket-chip/riscv-tools/riscv-gnu-toolchain/build
../configure --prefix=$RISCV
make -j$(nproc) linux

cd $TOP/qemu
./configure --static --disable-system --target-list=riscv64-linux-user
make

cd $TOP/debian-riscv64
sudo update-binfmts --import work/qemu-riscv64
make cpio
```

When I run the final `make cpio` I get an error with the script as follows:
```
Time zone: 105

Current default time zone: 'America/New_York'
Local time is now:      Sat Nov  3 09:27:19 EDT 2018.
Universal Time is now:  Sat Nov  3 13:27:19 UTC 2018.

sudo tar cJf - -C work/debian-riscv64-chroot . > ../rootfs.tar.xz
rm -rf bin etc dev home lib proc sbin sys tmp usr mnt nfs root run init
mkdir -p bin etc dev home lib proc sbin sys tmp usr mnt nfs root run usr/bin usr/lib usr/sbin usr/share/perl5 usr/share/udhcpc lib/riscv64-linux-gnu usr/lib/riscv64-linux-gnu # usr/share/sysvinit
cp -p work/usr-share-udhcpc-default.script usr/share/udhcpc/default.script
cp -p work/debian-riscv64-chroot/sbin/mount.nfs ./sbin
cp: cannot stat 'work/debian-riscv64-chroot/sbin/mount.nfs': No such file or directory
work/makefile.inc:17: recipe for target 'init' failed
make: *** [init] Error 1
```

Am I doing something wrong, or are you missing a step in your documentation about the mount.nfs file?

jrrk, 2018-11-04:
I can’t see the output of the make cpio, there is the possibility of an error as it is the unstable distribution as explained in the docs. That is why we provide the pre-made distribution.

ChaoWao, 2019-10-04:
> I checked again, and Debain sid riscv64 is indeed broken at the moment: The following packages have unmet dependencies:  isc-dhcp-client : Depends: libdns-export1102 but it is not installable                    Depends: libisc-export169 but it is not installable                    Recommends: isc-dhcp-common but it is not going to be installed E: Unable to correct problems, you have held broken packages. This could be worked perhaps around by excluding isc-dhcp-client from the install, but the easier solution is to just wait for sid to stabilise again (That is why we provide the pre-built Debian root filing system). A possible alternative would be to use Fedora instead. This is further advanced in maturity but it is not viable to install updates locally due to the dnf dependency (apt-rpm is broken and deprecated on this distribution). No point in starting a lively debate, but intensive Python scripts should not be run on 50MHz CPUs IMHO.
> […](#)

I'm meeting the exactly seem question. What do you mean by saying "This could be worked perhaps around by excluding isc-dhcp-client from the install"? I'd like to have a try.

OS: Ubuntu 18.04

jrrk, 2019-10-07:
I checked again today and the result was, that part of the flow is working again. The riscv64 port failed to reach the percentage of working packages to migrate to the stable base at the last fork.