lowRISC/lowrisc-chip #101: kc705_mii branch status VS 0.6 refresh

GuillemCabo, 2018-12-13:
Hi,
I'm looking at the kc705_mii branch.
In the readme it says:
Current version: Release version 0.4 (05-2017)

I know that it is a experimental branch and you use to add features there from newer releases even if it is not 100% up to date.
There is a list of all the features available? I will like to have a better idea of how this compare to the official releases. 

Thank you.

jrrk, 2018-12-13:
The README is out of date. kc705_mii was is based on a pre-release of 
refresh-v0.6

Updating the README was more or last the last thing that was done.

The status is, SD-Card is working but it is highly advisable to use a 
micro-SD Card in an adapter

because older, slower, cards tend to be full size.

The Ethernet is sort of working but there are issues with clock domain 
synchronization errors on this board.

Also it does not auto-detect 1000BaseT

The video and HID keyboard are not supported at all.

1G of RAM memory is available.

Ethernet booting is possible but a greater layer of error checking is 
needed which slows it down quite a bit.

If the Ethernet was fully debugged it would be comparable to the 
refresh-v0.6 release.

GuillemCabo, 2018-12-14:
Great to see that is almost everything there.
I hope to have some time to play with it soon.
Thank you.

closed 2018-12-14 by GuillemCabo