lowRISC/lowrisc-chip #102: Eth 8packets

clare7, 2018-12-24:
Hi, I'm wondering what does the value `{3'H7,{7{3'H5}}}` indicates? It was found in Line 98

	 ` if ((rx_nxt == {3'H7,{7{3'H5}}}) && (byte_sync == 0) && (sync == 0))`

Based on the codes given, it seems like is checking the end of 8th (collected) RX packets which each packet is 3 bits size. However, I couldnt figure out the how did it comes to value H7 and H5?

jrrk, 2018-12-24:
In this case the coding is RMII, so it is checking two data bits and receive enable.

clare7, 2018-12-26:
In that case, why does it has to be 7 of ,{3'H5} ? In order to check for
the SFD 1101 0101 isnt that 3 of {3'H5}  will suffice for coding in RMII?

jrrk, 2018-12-26:
You could certainly adapt the code to only check a single byte but thereby the chance of a false sync could be increased. Bearing in mind the chance of a bit error is greater early on in the packet due to the possibility of contention from multiple transmitters.

clare7, 2018-12-28:
Noted. Thank you. I also notice that there is no signal that triggers
*tx_axis_tready
. *Did I miss out something?

  if (tx_axis_tready)
    begin
       tx_frame_addr <= tx_frame_addr + 1;
       tx_axis_tlast <= (tx_frame_addr == tx_packet_length-2) &
tx_axis_tvalid_dly;
    end

jrrk, 2018-12-28:
You did. The port  input_axis_tready of axis_gmii_tx.v is actually an 
output despite its name.

This indicates in a convoluted way when the transmitter is ready for 
more input. If you have

issues with the coding style you could take it up with the original 
author but I would advise

against this because we have a phrase 'Don't look a gift horse in the 
mouth'.

This is entirely contradictory to the Trojan horse saying 'Beware of 
Greeks bearing gift [horses]'.

So you decide if you received a gift horse or trojan horse, that's what 
open source is all about!

In the same way you can judge this overall release to see if it more 
closely fits the description

'many hands make light work' or 'too many cooks spoil the broth'.

Either way, have a prosperous New Year.