lowRISC/lowrisc-chip #103: Simulating lowrisc chip

bnkla, 2019-02-16:
I want to simulate the lowrisc chip and I wonder how to do so. After freshly cloning the 0.6 repository I tried this:
```
export VERILATOR_ROOT=/home/user/verilator/install
export PATH=$PATH:$VERILATOR_ROOT/bin                                               
export RISCV=/home/user/riscv                 
export PATH=$PATH:$RISCV/bin                                                        
cd vsim 
make sim
```
Now, I get this error:
```
[info] [25.630] Done elaborating.
[success] Total time: 190 s, completed Feb 16, 2019 9:03:17 PM
mkdir -p /home/user/lowrisc-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/user/lowrisc-chip/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.fir -o /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.sv -X sverilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.conf
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
------------------------------------------------------------------------------
Error: Input file /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.fir not found
------------------------------------------------------------------------------
Usage: firrtl [options] [<arg>...]

common options
  -tn, --top-name <top-level-circuit-name>
                           This options defines the top level circuit, defaults to dut when possible
  -td, --target-dir <target-directory>
                           This options defines a work directory for intermediate files, default is .
  -ll, --log-level <Error|Warn|Info|Debug|Trace>
                           This options defines a work directory for intermediate files, default is .
  -cll, --class-log-level <FullClassName:[Error|Warn|Info|Debug|Trace]>[,...]
                           This options defines a work directory for intermediate files, default is .
  -ltf, --log-to-file      default logs to stdout, this flags writes to topName.log or firrtl.log if no topName
  -lcn, --log-class-names  shows class names and log level in logging output, useful for target --class-log-level
  --help                   prints this usage text
  <arg>...                 optional unbounded args
firrtl options
  -i, --input-file <firrtl-source>
                           use this to override the default input file name , default is empty
  -o, --output-file <output>
                           use this to override the default output file name, default is empty
  -faf, --annotation-file <input-anno-file>
                           Used to specify annotation files (can appear multiple times)
  -foaf, --output-annotation-file <output-anno-file>
                           use this to set the annotation output file
  -X, --compiler <high|middle|low|verilog|sverilog>
                           compiler to use, default is verilog
  --info-mode <ignore|use|gen|append>
                           specifies the source info handling, default is append
  -fct, --custom-transforms <package>.<class>
                           runs these custom transforms during compilation.
  -fil, --inline <circuit>[.<module>[.<instance>]][,..],
                           Inline one or more module (comma separated, no spaces) module looks like "MyModule" or "MyModule.myinstance
  -firw, --infer-rw <circuit>
                           Enable readwrite port inference for the target circuit
  -frsq, --repl-seq-mem -c:<circuit>:-i:<filename>:-o:<filename>
                           Replace sequential memories with blackboxes + configuration file
  -clks, --list-clocks -c:<circuit>:-m:<module>:-o:<filename>
                           List which signal drives each clock of every descendent of specified module
  -fsm, --split-modules    Emit each module to its own file in the target directory.
  --no-check-comb-loops    Do NOT check for combinational loops (not recommended)
  --no-dce                 Do NOT run dead code elimination

/home/user/lowrisc-chip/rocket-chip/scripts/vlsi_mem_gen /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.conf > /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.behav_srams.sv.tmp && mv -f /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.behav_srams.sv.tmp /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.behav_srams.sv
Traceback (most recent call last):
  File "/home/user/lowrisc-chip/rocket-chip/scripts/vlsi_mem_gen", line 195, in <module>
    main()
  File "/home/user/lowrisc-chip/rocket-chip/scripts/vlsi_mem_gen", line 191, in main
    for line in open(sys.argv[1]):
IOError: [Errno 2] No such file or directory: '/home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.conf'
make: *** [/home/user/lowrisc-chip/Makefrag-build:22: /home/user/lowrisc-chip/vsim/generated-src/TestHarness.DefaultFPGAConfig.behav_srams.sv] Error 1

```
I have file freechips.rocketchip.system.DefaultFPGAConfig.fir in generated-src, but not TestHarness.Default

jrrk, 2019-02-18:
Dear bnkla,

The current release of lowrisc moved all the rocket functionality back into 
lowrisc-chip-refresh-v0.6/rocket-chip/vsim.

This is to improve modularity, ease of upgrading the Rocket IP, and to make it 
easier to see what has changed.

Very sorry for the confusion caused by still having a toplevel directory called 
vsim from previous releases.

Now there is an issue with ISA simulations, Rocket expects a cut-down JTAG 
interface to be used to speed up simulation

and program loading. This cut down interface does not work on our chosen 
hardware so we use a modified Xilinx JTAG

chain using the USER chain functionality. A side-effect of this is that the full 
Xilinx JTAG chain chain model is not published

so cannot be used in simulation. If you just want to simulate Verilog 
peripherals then the facilities provided by Vivado for

external simulation (the internal ISim is not adequate) may be used but it is 
not straightforward, because of issues with

incompatible library versions between Vivado and VCS (Verilator does not 
correctly simulate Vivado libraries, which in any

case are encrypted). So the summary is, if you have a Nexys4DDR board, this 
release will be useful to you, if not the standard

Rocket release without JTAG will be 99.9% compatible in a simulation flow.

Regards,

Jonathan

PS If you want to know the exact differences in the Rocket code, Gabriel Somlo 
has rebased and carefully documented them here:

https://github.com/gsomlo/rocket-chip/tree/gls-lowrisc-v02

bnkla, 2019-02-18:
Hi,
Thank you for your reply. Yes, I have a Nexys4DDR board, so my options are to simulate the FPGA using make simulation in fpga/board/nexys4_ddr?

Simulating the Rocket itself should work running the makefile in rocket-chip/vsim? Or this is not working because of the modified XILINX JTAG?

vsim and vcs in main directory I can ignore?

My goal is to simulate the complete lowrisc chip.

BR

jrrk, 2019-02-18:
The make simulation target also has lagged behind because it relies on Vivado 
functionality. Certainly you can run ISA tests

in the vsim directory having disabled the JTAG modification. But for whole chip 
simulation we do not have a good solution, and

the only option I can suggest is using VCS and the export simulation option of 
Vivado.

Compiling the encrypted IPs is a problem because of the incompatible 
LD_LIBRARY_PATH settings, I can only offer the following unsupported solution:

cd fpga/board/nexys4_ddr

source /opt/Xilinx/Vivado/2015.4/settings64.sh
unset LD_LIBRARY_PATH
export SNPSLMD_LICENSE_FILE=27023@lmserv-synopsys.cl.cam.ac.uk
export VCS_HOME=/opt/synopsys/M-2017.03-SP2-5
export PATH=$VCS_HOME/bin:$PATH
rm -rf compile_simlib
sed -e "s=~=$PWD=" -e "s=@=$VCS_HOME/bin=" > compile_simlib.tcl <<EOF
#close_project
create_project -force "vcs" .
set_property "default_lib" "xil_defaultlib" [current_project]
set_property "part" "xc7a100tcsg324-1" [current_project]
set_property "simulator_language" "Mixed" [current_project]
set_property target_simulator VCS [current_project]
set_property compxlib.vcs_compiled_library_dir ~/compile_simlib [current_project]
compile_simlib -force -language all -dir {~/compile_simlib} -simulator vcs_mx 
-simulator_exec_path {@} -library all -family artix7
EOF

You will notice an older version of Vivado is selected for this particular 
operation. You can then select this directory, when invoking the export 
simulation function in Vivado.

We do not have other simulators to try, perhaps they would be more tolerant of 
the Vivado version. Depending on your install you might have to manually modify 
the script

generated by export simulation (for example to add -full64 to the command line). 
Unfortunately by its nature HDL simulation can be a bit of learning curve. Do 
not use vsim or

vcs in the main directory, and be sure to respect all our instructions regarding 
initial synthesis (to get the required models).

closed 2019-03-15 by bnkla