lowRISC/lowrisc-chip #106: Nasti Lite transactions make processor hang

beatsnbytes, 2019-02-22:
Hi all, 

I am  experimenting with sending lw and sw signals for addresses residing in the IOSPACE (0x80000000 - 0x8fffffff) (untethered version v 0.2). 
If I issue any command (lw or sw on its own then the data are loaded or stored respectively and everything goes smooth).
On the other hand, any combination of the aforementioned commands the processor hangs.
- I have made the following observations. The signal "address read ready" after the  first 1->0 transition never goes back to '1' thus blocking the second  ld command since the ar_valid signal  infinitely waits for the ar_ready signal. This scenario is presented in the image below.
[image]

 (Note that the same thing happens only if I issue one lw command but since after this no other command is following the processor never hangs)

[image]

All the other combinations also make the processor hang but I haven't figured out yet why. 
Summing up,  is there some limitation that I am missing regarding the access to IO devices through the uncached TileLink protocol ? Has anyone experienced this issue before?

Best regards

jrrk, 2019-02-22:
>     lw_lw_nasti_lite

Dear Koba,

This version isn't supported any more but I have the following observations:

1.     By default the valid DRAM address range is 0x80000000 - 0x87FFFFFF (you 
do not say what board you are using)

2.    The DRAM controller from Xilinx expects AXI4 transactions of which NASTI 
is a subset, but NASTI Lite is not

3.    The TileLink protocol would be to use cached accesses to DRAM, so the main 
traffic on this interface would drain and refill operations.

4.    NASTI Lite does not allow multiple outstanding transactions.

beatsnbytes, 2019-02-22:
Dear jrrk, 
Sorry for the incomplete question. 
1) I am using the kc705 board but that is an RTL simulation using verilator.
2) These addresses are actually addressing IO  since the is_IO flag is set.
3) Is the Tilelink also using cached accesses to the IO?
4) That means that if I give another load instruction to NASTI Lite while it serves the previous one then I am not guaranteed safe operation?
Best regards,

jrrk, 2019-02-22:
Dear Koba,

This version of Rocket/TileLink is rather old and known to have bugs. If you 
want to upgrade you should use

the kc705_mii branch which is almost equivalent to release v0.6. In the latest 
version of Rocket you can

independently set for any address range, whether memory is executable, 
cacheable, writable etc.

Any NASTI Lite peripheral can connect to Rocket using an NASTI to NASTI Lite 
bridge which takes care of any necessary delays.

This will just block the processor until the previous transaction is complete. 
If this is not happening I suggest you rerun

your simulation with a known good NASTI Lite peripheral.

Whatever support I can give will be strictly limited for this version, but in 
order to avoid wasted time be sure to include:

1.    The version (v0.2)

2.    Why you can't use the latest version (kc705_mii)

3.    What you changed and why

4.    What you are trying to achieve

You would not expect to use caching in conjunction with I/O because of readable 
status flags which will be volatile. This is distinct from volatile

keywords in C which just mean do not keep memory values in registers.

beatsnbytes, 2019-02-22:
Dear jrrk, 

Thank you for the feedback. 
I am based on a legacy project (v0.2) hence the choice of the specific version. I am trying (for bare metal testing) to bypass the cached transactions making the core to effectively "talking" just to the IO. this can be done in theory with remapping the whole address space to IO but with the recent findings I am not really sure that such a goal is a feasible one.

Best regards

beatsnbytes, 2019-02-28:
At the end, I was accessing memory that was out of the address space of the peripheral devices effectively making the processor to hang since it could not get a valid AXI Lite response back. Thanks for the help anyway jrrk.

closed 2019-02-28 by beatsnbytes