lowRISC/lowrisc-chip #107: OpenOCD and GDB debugging with Spike

ChafikSohaib, 2019-03-08:
Hi, I was trying to use OpenOCD and GDB debugging with spike but apperently the OpenOCD crash:

```
$make spike-ocd-launch 
-------------------  Starting OpenOCD -------------------
/home/chafik/lowrisc-chip-refresh-v0.6/riscv/bin/openocd -f ./spike.cfg
Open On-Chip Debugger 0.10.0+dev-00165-gdc312f5 (2019-03-06-18:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:9824
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0xdeadbeef (mfg: 0x777 (<unknown>), part: 0xeadb, ver: 0xd)
Makefile:129 : la recette pour la cible « spike-ocd-launch » a échouée
make: *** [spike-ocd-launch] Abandon (core dump créé)
```

with the configuration: (spike.cfg)

```
interface remote_bitbang

remote_bitbang_host localhost
remote_bitbang_port 9824

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 3735928559

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
halt

```

Am I doing something wrong?

jrrk, 2019-03-08:
The lowrisc version of openocd uses different scan chains to access the debug 
block of the RISCV, this is due to having to share the scan chain that is 
already used for XiIinx purposes. One of the changes is an irlen of 6 instead of 
5. The latest version of openocd supports flexible allocation of the scan chain 
numbers, but this came out after our release. We have not included in this 
release a version of spike that models our lowrisc peripherals, so you might be 
better off just connecting directly to the FPGA, or running an unmodified 
version of openocd.

ChafikSohaib, 2019-03-08:
I see, Thanks a lot

closed 2019-03-08 by ChafikSohaib