lowRISC/lowrisc-chip #108: Generating emulators

ChafikSohaib, 2019-03-15:
Hi, I tried folowing the tutorial on this link:  [https://www.lowrisc.org/docs/](url), so after building the tool-chain and all I needed to generate the C emulator, but I get this error.

`chafik@chafik-OptiPlex-7060:~/lowrisc-chip/rocket-chip/emulator$ make clean
rm -rf *.o *.a emulator-* /home/chafik/lowrisc-chip/rocket-chip/emulator/generated-src /home/chafik/lowrisc-chip/rocket-chip/emulator/generated-src-debug DVEfiles ./output
chafik@chafik-OptiPlex-7060:~/lowrisc-chip/rocket-chip/emulator$ make
mkdir -p /home/chafik/lowrisc-chip/rocket-chip/emulator/generated-src/
cd /home/chafik/lowrisc-chip/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/chafik/lowrisc-chip/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator /home/chafik/lowrisc-chip/rocket-chip/emulator/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultConfig"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/chafik/lowrisc-chip/rocket-chip/project
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
[info] Loading settings from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/chafik/lowrisc-chip/rocket-chip/)
[info] Packaging /home/chafik/lowrisc-chip/rocket-chip/macros/target/scala-2.11/macros_2.11-1.2.jar ...
[info] Done packaging.
[info] Packaging /home/chafik/lowrisc-chip/rocket-chip/chisel3/target/scala-2.11/chisel3_2.11-3.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Compiling 27 Scala sources to /home/chafik/lowrisc-chip/rocket-chip/hardfloat/target/scala-2.11/classes ...
[warn] there were 867 feature warnings; re-run with -feature for details
[warn] one warning found
[info] Done compiling.
[info] Packaging /home/chafik/lowrisc-chip/rocket-chip/hardfloat/target/scala-2.11/hardfloat_2.11-1.2.jar ...
[info] Done packaging.
[info] Compiling 222 Scala sources to /home/chafik/lowrisc-chip/rocket-chip/target/scala-2.11/classes ...
[warn] /home/chafik/lowrisc-chip/rocket-chip/src/main/scala/devices/debug/Debug.scala:941:42: method != in class UInt is deprecated: Use '=/=', which avoids potential precedence problems
[warn]     abstractCommandBusy := (ctrlStateReg != CtrlState(Waiting))
[warn]                                          ^
[warn] /home/chafik/lowrisc-chip/rocket-chip/src/main/scala/devices/debug/DebugTransport.scala:182:85: method != in class UInt is deprecated: Use '=/=', which avoids potential precedence problems
[warn]   nonzeroResp := stickyNonzeroRespReg | (io.dmi.resp.valid & (io.dmi.resp.bits.resp != UInt(0)))
[warn]                                                                                     ^
[warn] /home/chafik/lowrisc-chip/rocket-chip/src/main/scala/devices/tilelink/BusBypass.scala:106:25: method != in class UInt is deprecated: Use '=/=', which avoids potential precedence problems
[warn]     val stall = (bypass != io.bypass) && a_first
[warn]                         ^
[warn] /home/chafik/lowrisc-chip/rocket-chip/src/main/scala/subsystem/InterruptBus.scala:16:36: class IntXing in package interrupts is deprecated: IntXing does not ensure interrupt source is glitch free. Use IntSyncSource and IntSyncSink
[warn]     val asyncXing = LazyModule(new IntXing(sync))
[warn]                                    ^
[error] Nom de fichier trop long
[warn] four warnings found
[error] one error found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 43 s, completed 15 mars 2019 14:09:02
Makefrag-verilator:12 : la recette pour la cible « /home/chafik/lowrisc-chip/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.fir » a échouée
make: *** [/home/chafik/lowrisc-chip/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.fir] Erreur 1
`
I tried to change the argument in /project/build.scala the configure the lenght of the file name but I'm still getting the same problem:

`// See LICENSE for license details.

import sbt._
import Keys._
import complete._
import complete.DefaultParsers._
import xerial.sbt.Pack._

object BuildSettings extends Build {

  override lazy val settings = super.settings ++ Seq(
    organization := "berkeley",
    version      := "1.2",
    scalaVersion := "2.11.7",
    parallelExecution in Global := false,
    traceLevel   := 30,
    scalacOptions ++= Seq("-deprecation","-unchecked"),
    scalacOptions ++= Seq("-Xmax-classfile-name", "140"),
    libraryDependencies ++= Seq("org.scala-lang" % "scala-reflect" % scalaVersi$
     addCompilerPlugin("org.scalamacros" % "paradise" % "2.1.0" cross CrossVers$
 )

  lazy val chisel = project in file("chisel3")
  lazy val hardfloat  = project.dependsOn(chisel)
  lazy val rocketchip = (project in file(".")).dependsOn(chisel, hardfloat)

}
`

what am I doing wrong?

jrrk, 2019-03-15:
> Hi, I tried folowing the tutorial on this link: https://www.lowrisc.org/docs/, so after building the tool-chain and all I needed to generate the C emulator, but I get this error.

I think I’m right in saying that direct C generation from Chisel is obsolete now and the recommendation is to generate Verilog and then use Verilator to generate the C model from the Verilog. You will adopt a similar approach except with the vsim directory instead of emulator. If you want to run ISA tests you should use the unmodified Rocket branch because we have customised the scan chain because of FPGA limitations.

I don’t think the tutorial you mentioned encourages C simulation in the manner you describe below.

ChafikSohaib, 2019-04-02:
@jrrk thanks for your time. Do you have any idea about the "name too long" error ?

I face this error even when we generate the bitstream for FPGA implementation.

(As you can see in the following screenshot, in french, "Nom de fichier trop long")

[image]

Regards,

jrrk, 2019-04-02:
> image

That's new to me. This is the report I get for the same operation:

[info] Compiling 222 Scala sources to 
/local/scratch-legacy/jrrk2/lowrisc-chip-refresh-v0.6/rocket-chip/target/scala-2.11/classes 
...
[warn] 
/local/scratch-legacy/jrrk2/lowrisc-chip-refresh-v0.6/rocket-chip/src/main/scala/devices/debug/Debug.scala:941:42: 
method != in class UInt is deprecated: Use '=/=', which avoids potential 
precedence problems
[warn]     abstractCommandBusy := (ctrlStateReg != CtrlState(Waiting))
[warn]                                          ^
[warn] 
/local/scratch-legacy/jrrk2/lowrisc-chip-refresh-v0.6/rocket-chip/src/main/scala/devices/debug/DebugTransport.scala:182:85: 
method != in class UInt is deprecated: Use '=/=', which avoids potential 
precedence problems
[warn]   nonzeroResp := stickyNonzeroRespReg | (io.dmi.resp.valid & 
(io.dmi.resp.bits.resp != UInt(0)))
[warn] ^
[warn] 
/local/scratch-legacy/jrrk2/lowrisc-chip-refresh-v0.6/rocket-chip/src/main/scala/devices/tilelink/BusBypass.scala:106:25: 
method != in class UInt is deprecated: Use '=/=', which avoids potential 
precedence problems
[warn]     val stall = (bypass != io.bypass) && a_first
[warn]                         ^
[warn] 
/local/scratch-legacy/jrrk2/lowrisc-chip-refresh-v0.6/rocket-chip/src/main/scala/subsystem/InterruptBus.scala:16:36: 
class IntXing in package interrupts is deprecated: IntXing does not ensure 
interrupt source is glitch free. Use IntSyncSource and IntSyncSink
[warn]     val asyncXing = LazyModule(new IntXing(sync))
[warn]                                    ^
[warn] there were 2081 feature warnings; re-run with -feature for details
[warn] 5 warnings found
[info] Done compiling.

Perhaps the internationalization module is failing somehow. I would suggest you 
try to emulate exactly the environment we have:

LANG=en_GB.UTF-8
LANGUAGE=en_GB:en
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=

You should also check the version of java that you are using. We are using 
openjdk-8-jre-headless.

Regards,

Jonathan

ChafikSohaib, 2019-04-04:
@jrrk , Thanks for your time. I believe it was a language configuration problem like you suggested.

closed 2019-05-03 by ChafikSohaib