lowRISC/lowrisc-chip #110: BRAM Size

furkanturan, 2019-04-11:
Hello, 

This is not an issue in fact, but a question. I see in [the 554th line of chip_top.v](https://github.com/lowRISC/lowrisc-chip/blob/7090b7b70f33aa3763e360d1c5b79fa3dd8a611c/src/main/verilog/chip_top.sv#L554) that you are inferring a BRAM of 128-bit size, even though the memory port of the controller is 32-bits (`LOWRISC_IO_DAT_WIDTH`) in width. You could have preferred a block ram that is shorter in width but deeper. Even more, you could have used the block memory generator from the IP Catalog for that rather than dealing with the corresponding code piece to parse the memory words. But you apparently did not prefer those. I feel like this is a trick to reduce resource utilisation or something like that, but could not get the exact reasoning. 

May I kindly ask the idea behind that decision?

Kind Regards,

jrrk, 2019-04-11:
The original goal was a bus that could be variable in width to explore the 
performance impact. This could be significant for boards that support wider DDR 
memory subsystems. We did not want to have a converter that could generate the 
contents for all sorts of different shapes and sizes of RAM, so went for the 
widest expected option. As for the Xilinx IP catalogue its use has been 
minimized because the type of syntax it outputs is incompatible with Verilator, 
and of course for ASIC synthesis (that we propose in the future if we get the 
funding), technology specific files should be minimized. Another complication is 
that we actually deliberately increase resource usage with our solution. The 
stock IP generator will pack resources into the parity bits, this makes it 
difficult if not impossible to update the ROM at the bitstream (data2mem) stage.

furkanturan, 2019-04-11:
Thank you very much for this detailed explanation, and prompt reply. It clarified things in my mind.

closed 2019-04-11 by furkanturan