lowRISC/lowrisc-chip #113: Deiban boot mode and Switch selection

wanming2008, 2019-04-28:
Hello.
I made a board. Almost the same as nexy4-DDR.But for the sake of portability. I removed the switch on the board.This also brings problems.
I see the Docs that  SW0-SW3 to configure the boot mode。SW15-SW11 to configure the mac address.
Then I removed SW10-SW5.But this caused the debian system to fail to boot.

Log says:
-  Switch selecion is 07E2
- Boot selection 07E2 not executed, droppoing to ash.

I used the official nexuy board.The correct log should be said

- Switch selecion is 0002

I know that modifying the board can correct this problem. But these SW10-SW5 gpio are not pulled out. There is also no test point

I wonder if I can start the normal entry into the debian system by modifying the software. thanks.
[image]

jrrk, 2019-04-28:
> image

Dear Wanming,

Thank you for sharing your exciting hardware cost reduction plans. The 
switch inputs are used in several places.

On a non-standard hardware you will want to edit the file 
debian-riscv64/work/busyboxinit.sh to accomodate your

requirements. This file was originally used to distinguish between 
SD-card root and NFS-root, but you can use it for

any other purpose if you have different hardware. Do you plan to follow 
an open-source route for your

new hardware? This is entirely up to you, but if you have further issues 
it would be helpful if you could send us a hardware

prototype to the address below, together with schematics to allow us to 
comment in more detail on your proposals.

In particular if you intend to produce a batch of units you will want to 
apply for a range of MAC addresses from the IEEE

( https://standards.ieee.org/faqs/regauth.html#2 ) and have a convenient 
way of administering this in production.

Dr Jonathan Kimmitt

University of Cambridge
Department of Computer Science and Technology
The Computer Laboratory
William Gates Building
15 JJ Thomson Avenue
Cambridge CB3 0FD

ENGLAND

jrrk2@cam.ac.uk

wanming2008, 2019-05-06:
Hello jrrk:
1: Thanks for your guidance， now it is available on the Debian system.
2: Regarding open source, we actually have a core board of artix-7 long time ago. It has an artix-7, ram, and 3.3V, 1.8V, 1.0V power supply on the core board. Then it brings out all the gpio of artix-7.
So contrast nexy-ddr. I redone a daughter board.
On this daughter's board. I designed Ethernet, switch, key, LED, TF card, VGA, and qspi-flash.
without the usb to PS/2 interface.because i do not understand the PS/2  design on the nexy DDR.
I removed the G-sensor and temperature sensing, usb to jtag, some keys, some switchess and pmod interfaces.So only two layers of board are needed to complete this design.
Now I have only done 2 such boards for an experimental test to study lowrisc.
If we can make the hardware stable. We will be open source at that time.

closed 2019-05-06 by wanming2008

jrrk, 2019-05-06:
The original PS/2 keyboard used a round connector, the USB converter came later. If you can get hold of the original green/purple PS/2 connector, that should work fine with our RTL.