lowRISC/lowrisc-chip #114: booting linux on the nexys4 ddr with the refresh-v0.6

ChafikSohaib, 2019-05-03:
Hi,

I'm currently using lowrisc refresh-v0.6 on the nexys 4 ddr for a C++ application that I need to implement. So, to do so, I wanted to boot the RISC-V linux on the FPGA with the latest version of lowrisc via SD-card. I generated the linux kernel, the ramdisk and the bootloader but when I try to simulate the behaviour on spike I get the error: 

``$spike +disk=root.bin ~/lowrisc-chip-refresh-v0.6/fpga/board/nexys4_ddr/boot.bin vmlinux ``

``
terminate called after throwing an instance of 'std::invalid_argument'
  what():  --disk/+disk unsupported (use a ramdisk)
``

I tried also to boot it directly with the three files but after loading the boot.bin successfully, it was stuck in the part when booting the loaded program, so the kernel is never booted.

I used busybox-1.30.1 for the ramdisk.

Could you please refer me to any documentation available regarding this issue. Or am I doing something wrong?

Best regards,

jrrk, 2019-05-03:
The aim of the bootloader is to extract the kernel from SD-card (or the 
network). These peripherals are hardware specific to the FPGA board, and not 
available under spike. As to the rest I am not quite clear why you have changed 
versions or not followed the instructions in our documentation in the first 
instance. For sure you can compile C++ programs under Linux on the FPGA, but it 
will be a little slow (possibly about 200 times slower than a PC).

You might want to try the lowrisc-quickstart repository if you are having 
trouble generating a valid sd-card and kernel.

ChafikSohaib, 2019-05-20:
Hello Dr. Jonathan,

Thank you for your help. I succeded in doing that my application is running perfectly on top of the FPGA. I get the results I wanted. My other problem if you don't mind, the results I have are some images that I want to display via VGA I started to use what is given in the hello application the hid driver in order to print colors on the LCD but the dimensions of each block printed are pretty huge to fulfill my perpose. Can I change theses dimensions? if yes, how can I do that? 

I used : 
``
hid_vga_ptr to print the blocks on the vga output 
``
``
hid_reg_ptr[LOWRISC_REGS_PALETTE+ i] to control the colors.
``

Thanks again.
Best regards,

jrrk, 2019-05-20:
If you execute git clone -b frame-buffer --recursive 
https://github.com/lowrisc/lowrisc-chip.git lowrisc-chip-frame-buffer

this will checkout an alternative VGA driver with higher resolution. However 
this FPGA lacks sufficient spare BRAM to produce

a full size image. There is a small bare-metal demonstration of this capability 
which is not really documented, but might be a better starting point for you.

ChafikSohaib, 2019-05-22:
Hi Dr. Jonathan,

Is there any chance that you can help me with the details of the main part in drawing the logo algorithm. How the logo palette for the colours and the logo array are linked? I'm having a hard time understanding how it works with all the values and the numbers in the`draw_logo(int ghlimit)` function. 

Actually, my target is some small png images, I succeeded in generating the equivalent of logo[ ] but I need to understand how it's linked with the colours palette and drawn with:
 `hid_fb_ptr[i * ghlimit + ((j - 15) >> 4)] = cline;` and 
`hid_reg_ptr[LOWRISC_REGS_PALETTE + i] = palette_logo2[i];`

Thank you for your help.
Best regards,

jrrk, 2019-05-22:
The draw_logo() function is part of the boot loader and as such is compressed in 
an artificial way. You would not use this method for real images. You first need 
to reduce your image to 16 colours and then you draw with the equivalent of 4 
bits per pixel. It's a bit like the way a versatile graphics adapter works. Each 
64-bit word represents 16 pixels.

ChafikSohaib, 2019-06-07:
Hi Dr. Jonathan,

Thank you again for you help. I have a last question regarding the VGA. Would you please tell me how can I, starting from a normal picture generate the a file similar to logo.h? Because of the fact that the image is compressed I couldn't understand how it's done. I successfuly used the low quality VGA driver but for the higher resolution I still have problems.

Best regards,

jrrk, 2019-06-07:
The logo uses run length encoding. GIMP can generate this type of file in its export C mode.

jrrk, 2019-06-07:
Follow up, the next release will have X-windows support which should make your life easier.

closed 2019-07-01 by ChafikSohaib