lowRISC/lowrisc-chip #117: Hello.c compilation error with riscv64-unknown-elf-gcc

jalezeta, 2019-06-18:
I am aiming to run Linux in the lowrisc-chip and I am using this git repository as a reference. 
I have installed the riscv-tools from https://github.com/riscv/riscv-gnu-toolchain. 
I had no problem on generating and uploading the bitsream into the Zedboard, but I am having some troubles when running the "make hello" command to test the hello.c program on the RISC-V platform. 
The related error to the compilation error appeared as:

_riscv64-unknown-elf-gcc: error trying to exec 'cc1': execvp: No such file or directory_

It looks like I am missing 'cc1', but I can locate it under "/opt/riscv/libexec/gcc/riscv32-unknown-linux-gnu/8.3.0" and under "/opt/riscv/libexec/gcc/riscv64-unknown-elf/8.3.0" so I honestly do not know what can be happening. 

I tried to add "/opt/riscv/libexec/gcc/riscv64-unknown-elf/8.3.0" into the $PATH environment variable but I am still getting errors related to 'cc1'.

_cc1: error: unrecognized command line option '-mhard-float'_

jrrk, 2019-06-18:
Dear Jalezeta,
  We don’t support the zedboard officially, this section of the repository was contributed by a third party. However your problem doesn’t seem to be related to that aspect of your query. The recommended compiler for LowRISC is the one given in the tutorial on lowrisc.org, which is actually part of the repository you downloaded. Depending on the version of LowRISC (which you don’t mention) certain parameters such as CSR numbers and flag names could have changed relative to the compiler you tried to use.

Ideally there would be only one compiler for everybody but at the moment this is not possible where older releases of code that have been customised are concerned. Also due to auto-configuration there can be operating system dependencies also which can cause the symptoms you describe.

Please have a go at following our tutorial as written before branching off and trying other compilers.

Regards,
Jonathan

jalezeta, 2019-06-18:
Dear Jonathan,

I am following one of the tutorials in lowrisc.org (https://www.lowrisc.org/docs/debug-v0.3/zedboard/). As it can be seen, the used lowrisc version is the V0.3, thus, that is the one I am using. Regarding to compiler, I have all the compilers installed as in "https://www.lowrisc.org/docs/untether-v0.2/riscv_compile/".

Regards,
Jon Ander

jrrk, 2019-06-18:
If that’s the case, the latest 8.3.0 compiler will not work because v0.3 is based on an old version of Rocket. The version of compiler that comes with this is rather old (v6.1.0):

https://github.com/lowRISC/riscv-gnu-toolchain/tree/master/gcc

This doesn’t seem to be the version you have installed.

 I would suggest rebasing the zedboard patches on the latest release (v0.6), if you want to use compressed instructions and JTAG gdb debugging.

jalezeta, 2019-06-21:
I reinstalled all the tools in the suggested version and the problem is solved! thanks!

closed 2019-06-21 by jalezeta