lowRISC/lowrisc-chip #118: issues to use FPU

ChafikSohaib, 2019-07-01:
Hello,
I'm currently trying to port a CNN application to lowRISC in which I need the FPU, I'm having a problem getting the results in the output of the FPU because all I get is a zero as a result.
This is a simple demo to my issue:

I added these code lines to hello.c to test it: 

[image]

and as a results I got:
int: 1234
float int: 0

I used the entry file crt.S as it is giving where the FPU is activated and initialized.
And I used gdb to see the contents of the variables:

[image]

Can I have some help or clarification on what I may be doing wrong? Thanks.

jrrk, 2019-07-01:
> image

Dear Chafik,

I see no variables a,x and y in my version of the code so I can't really 
comment. In general it is a bad

idea to post fragmentary information even if you are convinced the recipient 
knows everything that you know.

Also you have confused the issue by mentioning the FPU, but not mentioning the 
definitions of any floating-point variables.

Regards,

Jonathan

ChafikSohaib, 2019-07-01:
Dear Dr, Jonathan,

I'm truly sorry I got confused between two captures

[image]

This is the one I should have posted. Thank you again.

Best regards,

jrrk, 2019-07-01:
Dear Chafik,

The bare metal test program uses a customised and simplified C-library to 
minimise the program footprint. This is a standard

technique to minimise on-chip ROM area (64K by default).

It does not support a writable data segment because this would result in 
non-portable behaviour (ROM would behave differently to RAM).

If you want to support non-trivial full ANSI-C behaviour, the recommended method 
(if you are allergic to Linux) is to cross-compile your code in the manner of 
BBL. This will allow your programs an essentially unlimited amount of memory (up 
to 128MBytes anyway), and if you stick to machine mode you will still have full 
access to all the GDB state.

If you use GDB to load the program, there are no restrictions on start address 
or API, but GDB does not initialise writable data or BSS by default. I mentioned 
this on the mailing list but the response was that GDB should not be doing 
anything that the programmer can do themselves.

LowRISC doesn't really support bare-metal operation since it requires a lot more 
knowledge than working under Linux, and is also much more laborious, but there 
is nothing to stop advanced users from figuring out the details themselves.

ChafikSohaib, 2019-07-02:
I got your idea, Dr Jonathan, and it's working using a linker file similar to the one used in BBL, starting from 0x8000_0000.
Thanks a lot for your help!

Best regards,

closed 2019-07-02 by ChafikSohaib