lowRISC/lowrisc-chip #119: make boot loader failed :Error: junk at end of line, first unrecognized character is `-'

mathmax12, 2019-09-04:
I'd like to generate the systemverilog for the boot loader using the "lowrisc-chip/fpga/src/etherboot".  Then I got a below error.
 Any help will be appreciated.
=======================================================
lowrisc-chip/fpga/src/etherboot$make all

make: Warning: File 'Makefile' has modification time 6.1 s in the future
sed -e "s/\(compatible = \"\)\(\"\)/\1\2/" ../genesys2.dts | dtc -I dts -O dtb -o genesys2_ariane.dtb
echo -e .section .dtb \; .align 4,0 \; .globl _dtb \; _dtb: \; .incbin \"genesys2_ariane.dtb\" | riscv64-unknown-elf-gcc -Os -ggdb -march=rv64imac -mabi=lp64 -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=  -c -xassembler - -o genesys2_ariane.o
{standard input}: Assembler messages:
{standard input}:1: Error: junk at end of line, first unrecognized character is `-'
Makefile:40: recipe for target 'genesys2_ariane.o' failed
make: *** [genesys2_ariane.o] Error 1

jrrk, 2019-09-05:
This command expects BITSIZE to be set in the environment, as per the higher level Makefile

mathmax12, 2019-09-05:
@jrrk 
Thanks for that!

I added the below into the **Makefile** under the "lowrisc-chip/fpga/src/etherboot".

` BITSIZE=0xB00000
  BOARD="genesys2"
  CPU="ariane"
  COMPATIBLE=ethz, ariane
`

The error persists.
Do I miss some thing else?

`CC    <= src/hid.c
CC    <= src/logo.c
CC    <= startup.S
sed -e "s/\(compatible = \"\)\(\"\)/\1ethz, ariane\2/" ../genesys2.dts | dtc -I dts -O dtb -o genesys2_ariane.dtb
echo -e .section .dtb \; .align 4,0 \; .globl _dtb \; _dtb: \; .incbin \"genesys2_ariane.dtb\" | riscv64-unknown-elf-gcc -Os -ggdb -march=rv64imac -mabi=lp64 -Wall -mcmodel=medany -fno-builtin-printf -fno-builtin-memcpy -fno-builtin -ffreestanding -fshort-wchar -DBITSIZE=0xB00000  -c -xassembler - -o genesys2_ariane.o
{standard input}: Assembler messages:
{standard input}:1: Error: junk at end of line, first unrecognized character is `-'
Makefile:46: recipe for target 'genesys2_ariane.o' failed
make: *** [genesys2_ariane.o] Error 1
`

Thanks.

jrrk, 2019-09-05:
Have you tried compiling from the top level Makefile as intended? If that fails, it could be an issue with the installed version of GCC. To debug, dump the intermediate files with the -save-temps option of GCC.

jrrk, 2019-09-05:
Moreover the version of echo on the path might be important. Try calling /bin/echo explicitly.

mathmax12, 2019-09-05:
@jrrk 
Thanks for your quick reply.
The /bin/echo works for the "lowrisc-chip/fpga/src/etherboot" make.

I tried to build from the top "lowrisc-chip" but got the following error.
"
On branch linux-5.1.3-lowrisc
nothing to commit, working directory clean
make -C debian-riscv64 cpio
make[1]: Entering directory '/home/xiaojia.song/riscv_project/temp/lowrisc-chip/debian-riscv64'
make[1]: *** No rule to make target 'cpio'.  Stop.
make[1]: Leaving directory '/home/xiaojia.song/riscv_project/temp/lowrisc-chip/debian-riscv64'
Makefile:55: recipe for target 'linux-5.1.3-lowrisc/initramfs.cpio' failed
make: *** [linux-5.1.3-lowrisc/initramfs.cpio] Error 2
"

mathmax12, 2019-09-16:
This resolved the issue.

Thanks.

closed 2019-09-16 by mathmax12