lowRISC/lowrisc-chip #120: Question about the new version of the Nexys 4 DDR

noureddine-as, 2019-09-05:
Hello Dr. Jonathan,

I wanted to buy the Nexys 4 DDR FPGA board [here](https://store.digilentinc.com/nexys-4-ddr-artix-7-fpga-trainer-board-recommended-for-ece-curriculum/) but apparently "The Nexys 4 DDR has been renamed to the Nexy A7". I don't if it's just a rebrand, or if there is any potential changes compared to the old version.

Has someone ever tested LowRISC (at least refresh-v0.6) on that new board? Or those will need modifications? If yes, what should be changed in the current project?

Thank you very much.
Best regrads,
Noureddine.

jrrk, 2019-09-05:
I have not tried it. I think there may be some minor peripheral changes. If so, you can cater for this by reviewing and, if necessary, updating the constraints file.

noureddine-as, 2019-09-05:
Comparing the `Nexys-A7-100T-Master.xdc` and `Nexys-4-DDR-Master.xdc` files from https://github.com/Digilent/digilent-xdc

The only difference is in `Pmod Header JXADC` where `IOSTANDARD LVDS` is changed  to `IOSTANDARD LVCMOS33`. So i guess this does not affect the actual project, isn't it ?

jrrk, 2019-09-05:
You are correct, JXADC is not used.

noureddine-as, 2019-09-05:
Also, I noticed there is a difference in the Block RAM size,
from Nexys 4 DDR has 4,860 Kbits and Nexys A7 has 1,188 Kb.
And this constraints the maximum size of the bootloader, isn't it? do you thing it's going to be a problem?

https://reference.digilentinc.com/reference/programmable-logic/nexys-4-ddr/start
https://reference.digilentinc.com/reference/programmable-logic/nexys-a7/start

jrrk, 2019-09-06:
You should ignore the A7-50 variant, it is unsuitable.

noureddine-as, 2019-09-11:
Okay, I ordered the Nexys A7-100T. It may take a while before it arrives (4-5 weeks at least), and will post my updates here whenever I have news.
Thanks!

noureddine-as, 2019-09-24:
Just received the Nexys A7 100-T, and didn't need to change anything, they are the same.

Apparently the links I posted earlier contain different information. here is the one we aquired:
https://store.digilentinc.com/nexys-a7-fpga-trainer-board-recommended-for-ece-curriculum/

closed 2019-09-24 by noureddine-as