lowRISC/lowrisc-chip #121: ariane-v0.7 build issue

mathmax12, 2019-09-06:
Dear Dr. Jonathan,
I tried to clone the ariane-v0.7 with  "git clone -b ariane-v0.7 --recursive https://github.com/lowRISC/lowrisc-chip.git"
And then got this issue:

Submodule 'ui/keycodemapdb' (git://git.qemu.org/keycodemapdb.git) registered for path 'ui/keycodemapdb'
Cloning into 'capstone'...
fatal: unable to connect to git.qemu.org:
git.qemu.org[0: 172.99.69.163]: errno=Connection timed out

fatal: clone of 'git://git.qemu.org/capstone.git' into submodule path 'capstone' failed
Failed to recurse into submodule path 'qemu'

I checked the qemu folder under "https://github.com/lowrisc/qemu/tree/7ae23bc1974f8738b5009125873e43e6a8b1d848"
The capstone @ 22ead3e and dtc @ e543880 are not accessible.  
It seems these two links are expired.

Where can I get them?

jrrk, 2019-09-06:
Any recent version of QEMU would do here. The QEMU user executable is needed for the Debian bootstrap, originally the thread was about using buildroot. You can also download a prerelease of the complete root file system from the release area of GitHub.

mathmax12, 2019-09-06:
Thanks for the prompt reply.
I manually download the qemu and remove the update operation.
The qemu build passed.

Then another error happened.
Did I do something wrong?

"make -C debian-riscv64 cpio
make[1]: Entering directory '/home/xiaojia.song/riscv_project/temp/lowrisc-chip2/lowrisc-chip/debian-riscv64'
make[1]: Warning: File 'work/makefile.inc' has modification time 34 s in the future
sudo debootstrap --arch=riscv64 --variant=minbase --keyring=/etc/apt/trusted.gpg \
      --include=gnupg --exclude=cgmanager sid work/debian-riscv64-chroot \
          http://deb.debian.org/debian-ports || exit 1
I: Retrieving InRelease 
I: Checking Release signature
E: Release signed by unknown key (key id DA1B2CEA81DCBC61)
work/makefile.inc:28: recipe for target 'work/debian-riscv64-chroot/usr/bin/apt' failed
make[1]: *** [work/debian-riscv64-chroot/usr/bin/apt] Error 1
make[1]: Leaving directory '/home/xiaojia.song/riscv_project/temp/lowrisc-chip2/lowrisc-chip/debian-riscv64'
Makefile:55: recipe for target 'linux-5.1.3-lowrisc/initramfs.cpio' failed
make: *** [linux-5.1.3-lowrisc/initramfs.cpio] Error 2
"

jrrk, 2019-09-07:
There should be a command in the Makefile or subsidiary shell scripts to import any necessary keys. You can add any missing keys here. I’m away just now and not in a position to update it.

mathmax12, 2019-09-09:
It already added that key ( key id DA1B2CEA81DCBC61) in line 24 of https://github.com/lowRISC/debian-riscv64/blob/031543c7f555c23d8dfa588983d5e404d7fc7fbf/work/makefile.inc#L24.

jrrk, 2019-09-09:
Perhaps your version of O/S stores its keyring in a different location.

mathmax12, 2019-09-09:
Thank you for always giving my help.

It seems these key are not exist on the keyserver or expired?
`lowrisc-chip3$sudo apt-key adv --recv-keys --keyserver keyserver.ubuntu.com 8B48AD6246925553 7638D0442B90D010 04EE7237B7D453EC DA1B2CEA81DCBC61
Executing: /tmp/tmp.PX9r48K7Op/gpg.1.sh --recv-keys
--keyserver
keyserver.ubuntu.com
8B48AD6246925553
7638D0442B90D010
04EE7237B7D453EC
DA1B2CEA81DCBC61
gpg: requesting key 46925553 from hkp server keyserver.ubuntu.com
gpg: requesting key 2B90D010 from hkp server keyserver.ubuntu.com
gpg: requesting key B7D453EC from hkp server keyserver.ubuntu.com
gpg: requesting key 81DCBC61 from hkp server keyserver.ubuntu.com

gpg: keyserver timed out
gpg: keyserver receive failed: keyserver error
`
I tried to disable the key checking with `--no-check-gpg` and it passed the key checking.

Then I got another error:
`I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot /home/riscv_project/lowrisc-chip_ariane0d7/lowrisc-chip3/debian-riscv64/work/debian-riscv64-chroot **mount -t proc proc /proc**
W: See /home/riscv_project/lowrisc-chip_ariane0d7/lowrisc-chip3/debian-riscv64/work/debian-riscv64-chroot/debootstrap/debootstrap.log for details
work/makefile.inc:28: recipe for target 'work/debian-riscv64-chroot/usr/bin/apt' failed
make[1]: *** [work/debian-riscv64-chroot/usr/bin/apt] Error 1
make[1]: Leaving directory '/home/riscv_project/lowrisc-chip_ariane0d7/lowrisc-chip3/debian-riscv64'
Makefile:55: recipe for target 'linux-5.1.3-lowrisc/initramfs.cpio' failed
make: *** [linux-5.1.3-lowrisc/initramfs.cpio] Error 2
`
Here is the content of the debootstrap.log file:
`chroot: failed to run command 'mount': Exec format error`

How could I around this issue?
Is there any user documentation for the ariane-v0.7 branch at this moment?

jrrk, 2019-09-09:
Sorry, I haven’t written the v0.7 specific stuff but the Debian configuration stuff is similar to the v0.6 release. Perhaps you forgot the riscv misc filesystem stuff?

jrrk, 2019-09-10:
If you follow the step by step instructions for the Debian build on v0.6, you should be good for v0.7, or in the first instance use it instead. However both releases were based on Debian unstable so there is always going to be an issue with keys being updated. The latest fork was in August but I don’t think the riscv architecture made it into the mainline.

mathmax12, 2019-09-10:
Thanks, Dr. Jonathan.
I followed the v0.6 and here is what I got,
```
sudo cp work/sources.list work/debian-riscv64-chroot/etc/apt
sudo chroot work/debian-riscv64-chroot apt-key adv --recv-keys --keyserver keyserver.ubuntu.com DA1B2CEA81DCBC61
Warning: apt-key output should not be parsed (stdout is not a terminal)
Executing: /tmp/apt-key-gpghome.uaN3Ah9ZVX/gpg.1.sh --recv-keys --keyserver keyserver.ubuntu.com DA1B2CEA81DCBC61
gpg: keyserver receive failed: Connection timed out
work/makefile.inc:28: recipe for target 'work/debian-riscv64-chroot/usr/bin/apt' failed
make: *** [work/debian-riscv64-chroot/usr/bin/apt] Error 2

root@msl-mip-dev18:/home/riscv_project/lowrisc-chip_ariane0d7/lowrisc-chip3/debian-riscv64$make cpio
rm -rf bin etc dev home lib proc sbin sys tmp usr mnt nfs root run init
mkdir -p bin etc dev home lib proc sbin sys tmp usr mnt nfs root run usr/bin usr/lib usr/sbin usr/share/perl5 usr/share/udhcpc lib/riscv64-linux-gnu usr/lib/riscv64-linux-gnu # usr/share/sysvinit
cp -p work/usr-share-udhcpc-default.script usr/share/udhcpc/default.script
cp -p work/debian-riscv64-chroot/sbin/mount.nfs ./sbin
cp: cannot stat 'work/debian-riscv64-chroot/sbin/mount.nfs': No such file or directory
work/makefile.inc:272: recipe for target 'init' failed
make: *** [init] Error 1

```
It seems the key hasn't been put into the mainline.

jrrk, 2019-09-10:
There are other key servers you can consult.

mathmax12, 2019-09-10:
Sorry for borthering you so many times.
But is still doesn't work...

jrrk, 2019-09-11:
You can use the lowrisc-quickstart repository to download pre-built images of Debian and anything else you need instead of building from upstream repositories. I will investigate the issues you have noticed on my return. Sorry for any instability.

mathmax12, 2019-09-11:
Thank you very much for always giving me help.

I have an idea to achieve my goal but not sure about it.
The goal is booting the Linux system from the QSPI flash on the FPGA.

I know the lowrisc-chip/ariane-v0.7 has achieved this.
Reading from the Makefile of the pulp-platform/ariane, I found the **FPGA bitstream building** and the **building of bbl + Linux kernel** are two independent steps. Is that correct?
For my case, I already build the bbl + Linux kernel before and it works with the pulp-platform/ariane FPGA bitstream. The only thing that doesn't work for me is pulp-platform/ariane FPGA bitstream doesn't support booting OS from QSPI flash.

So I am thinking to build the bitstream with the lowrisc-chip/ariane-v0.7.
After that, I can use /write_cfgmem.tcl to generate the final msc for(bitstream + bbl + linux image).

Do you think this work?

Thanks.

jrrk, 2019-09-12:
You can mix and match the Ariane buildroot based riot filing system with the kernel from the LowRISC ariane-v0.7 release, provided that you ensure that the combined size is less than the maximum QSPI size. You need to use my bitstream or  copy my QSPI peripheral to the main Ariane repository. You need to use my boot loader also. After that you can make the modifications for your target board. There is no requirement to use the Debian root file system, and indeed it will be difficult to cut it down to a suitable size.

mathmax12, 2019-09-17:
Thanks.
I set up a DHCP server on the host.
Now the connection between the GenesysII board and host is connected.

````
DHCPREQUEST for 10.1.30.44 on eth0 to 255.255.255.255 port 67
DHCPACK of 10.1.30.44 from 10.1.30.250
bound to 10.1.30.44 -- renewal in 19760 seconds.
done.
[ ok ] Starting RPC port mapper daemon: rpcbind.
[FAIL] Starting NFS common utilities: statd idmapd failed!
[ ok ] Cleaning up temporary files....
[FAIL] startpar: service(s) returned failure: nfs-common ... failed!
INIT: Entering runlevel: 2
[info] Using makefile-style concurrent boot in runlevel 2.
[ ok ] Starting NTP server: ntpd.
Recovering nvi editor sessions.
none found.
[ ok ] Starting OpenBSD Secure Shell server: sshd.
```
But it stopped there. **No login line.**
I can ssh root@10.1.30.44 to the board and it required a password to log in.
I know the first time we can not ssh to log in the board system.
But this at least tells me the Ethernet connection is good.

Any suggestions will be appreciated.

jrrk, 2019-09-17:
sshd does not allow root to login without a secret key exchange. You could copy your public key to /root/.ssh/authorized_keys on the SD-Card (partition 2) to see if it helps. Also test your configuration with genesys2_rocket_xilinx.bit to see if you have an older version of Ariane with the floating point divider bug. You can also modify your root filing system to boot straight into a shell. And you will in due course want to build your own bitstream from the latest sources, in which case you can substitute your own custom startup script. Finally I haven’t got X windows working in Ariane yet, it only boots to the command line. You might prefer to work with the root filing system that comes with the ariane-sdk repository.

mathmax12, 2019-09-18:
Thanks a lot for always giving me suggestions.  I appreciate that.
I replaced the root filing system with the one from ariane-sdk and it worked.

I want to build the FPGA RTL and regenerate the bitstream.
But I can not find the definition of $(ariane_pkg) $(util) $(src) $(fpga_src) in line
https://github.com/lowRISC/lowrisc-chip/blob/64986933820090fb647ac7066079a100e62b66f4/Makefile#L72.
What should I assign to them in the branch ariane-v0.7?

Thanks.

jrrk, 2019-09-19:
This Makefile contains the statement 

include sources.inc

It should get included and defined automatically if you do 

make genesys2_ariane

in this directory. If this doesn't work perhaps your version of make is incompatible.

mathmax12, 2019-09-20:
Thanks.
I just notice that I missed that file on my local.

When I run the "make genesys2_ariane", an error was thrown at the [line ](riscv-pk/serial/vmlinux-serial: riscv-pk/serial/Makefile $(LINUX)/.config) in the Makefile.

> make -C linux-5.1.3-lowrisc ARCH=riscv CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf- CONFIG_SERIAL_8250_CONSOLE=y CONFIG_VT_CONSOLE=n CONFIG_LOWRISC_VGA_CONSOLE=n -j 4 
make[1]: Entering directory '/home/xiaojia.song/riscv_project/lowrisc-chip_ariane0d7/lowrisc-chip4/linux-5.1.3-lowrisc'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
...
...
  VDSOLD  arch/riscv/kernel/vdso/vdso-dummy.o
  CC      init/init_task.o
  CC      arch/riscv/kernel/cpu.o
  VDSOLD  arch/riscv/kernel/vdso/vdso.so.dbg
/opt/riscv/lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: -shared not supported
collect2: error: ld returned 1 exit status
  LD      arch/riscv/kernel/vdso/vdso-syms.o
  OBJCOPY arch/riscv/kernel/vdso/vdso.so
/opt/riscv/bin/riscv64-unknown-elf-objcopy: 'arch/riscv/kernel/vdso/vdso.so.dbg': No such file
arch/riscv/kernel/vdso/Makefile:50: recipe for target 'arch/riscv/kernel/vdso/vdso.so' failed
make[3]: *** [arch/riscv/kernel/vdso/vdso.so] Error 1
scripts/Makefile.build:486: recipe for target 'arch/riscv/kernel/vdso' failed
make[2]: *** [arch/riscv/kernel/vdso] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC      init/version.o
Makefile:1051: recipe for target 'arch/riscv/kernel' failed
make[1]: *** [arch/riscv/kernel] Error 2
make[1]: *** Waiting for unfinished jobs....
  AR      init/built-in.a
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[1]: *** wait: No child processes.  Stop.
Makefile:48: recipe for target 'riscv-pk/serial/vmlinux-serial' failed
make: *** [riscv-pk/serial/vmlinux-serial] Error 2

jrrk, 2019-09-20:
My version of the tool chain incorporates a hack for this problem (on certain host O/S) which unfortunately is not worthy to upstream.

mathmax12, 2019-09-20:
From the Ariane-SDK make file I found the **riscv64-unknown-linux-gnu-**  is recommended.

CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf- is an embedded elf toolchain. riscv64-unknown-linux-gnu- instead should be used from Linux toolchain.

This works.

jrrk, 2019-09-21:
The Unknown-linux-gnu- variant is intended to compile programs that run under Linux, the unknown-elf- variant is for bare metal programs such as the boot loader and Linux itself. This distinction prevents programs from making use of facilities such as system calls that are not available under bare metal environments. Of course the Linux kernel never does this, so it is safe to use the same compiler for kernel and userland. Linux under RISCV occasionally traps into machine mode, so it is not totally true to describe it as running on bare metal.

mathmax12, 2019-09-23:
I see.
Thanks a lot!

mathmax12, 2019-09-24:
I wonder is there a way to emerge the [boot.bin](https://github.com/lowRISC/lowrisc-chip/releases/tag/v0.7-rc1) and [rootfs.tar.xz
](https://github.com/lowRISC/lowrisc-chip/releases/tag/v0.7-rc1) to one file.

I want to use the [write_cfgmem.tcl](https://github.com/lowRISC/lowrisc-chip/blob/64986933820090fb647ac7066079a100e62b66f4/fpga/scripts/write_cfgmem.tcl#L28) to combine the above file and the FPGA bitstream file into one *.mcs file.

jrrk, 2019-09-24:
The existing build environment already merges the .bit file and boot.bin into a single file, you probably noticed that already. When you build Linux, you can replace the .cpio file with any root filing system of your choice. However, beware size limits on the combined .mcs file, on the Genesys2, the total including bitstream will be limited to 16MBytes or so if it needs to fit in QSPI memory.

mathmax12, 2019-09-25:
Thanks a  lot.

I didn't find the command line to generate the "boot.bin" , but in the [Build Berkeley boot loader](https://www.lowrisc.org/docs/build-berkeley-boot-loader/) page I noticed that the riscv-pk/serial/bbl
actually is the "boot.bin". 

If this is correct, then the [write_cfgmenm.tcl](https://github.com/lowRISC/lowrisc-chip/blob/64986933820090fb647ac7066079a100e62b66f4/fpga/scripts/write_cfgmem.tcl#L28) in the lowrisc-chip/ariane-v0.7 generate a *mcs file contains everything (bitstream, linux kernel, initramfs,  Berkeley boot loader) except the real rootfs.

I build a bbl in the riscv-pk/serial and cp this to the SD card as the boot.bin instead of the [release one](https://github.com/lowRISC/lowrisc-chip/releases/tag/v0.7-rc1).
But the booting stopped at the below:
```
`Hello from Ariane! Please wait a moment...
Relocating to DDR memory
Hello World!
swp[0] = 0
swp[1] = DEADBEEF
swp[2] = 0
swp[3] = 0
swp[4] = 0
Setup MAC addr
QSPI OEM[0] = 0
QSPI OEM[1] = 18
QSPI OEM[2] = 3e
QSPI OEM[3] = 2
QSPI OEM[4] = e3
QSPI OEM[5] = c1
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
SD boot

u-boot based first stage boot loader
MMC:   mmc created at 86800248, host = 86800200
lowrisc_sd: 0
Device: lowrisc_sd
Manufacturer ID: 3
OEM: 5344
Name: SC16G 
Bus Speed: 5000000
High Capacity: Yes
Capacity: 14.8 GiB
Bus Width: 4-bit

0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
80:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
e0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
100:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
120:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
140:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
160:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
180:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
1a0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ef be ad de 00 00 80 00
1c0:  01 01 0c 3f 1f 20 00 08 00 00 ff ff 00 00 00 00 01 21 83 3f 20 20 00 08 01 00 00 00 40 00 00 00
1e0:  01 21 82 3f a0 20 00 08 41 00 00 00 10 00 00 00 81 21 83 1f e0 59 00 08 51 00 00 c4 89 01 55 aa
Load boot.bin into memory
load elf to DDR memory
|Section[0]: elfn(80000000,0x1000,0x6f38);
|Section[1]: elfn(80007000,0x8000,0x10a0);
|memset(800080a0,0,0x9fe0);
|Section[2]: elfn(80200000,0x200000,0x820f2c);
|Boot the loaded program at address 80000000...

````

Am I right?

jrrk, 2019-09-25:
The boot sequence looks to be valid, but BBL never prints anything. I would expect the machine configuration to be printed if that option is enabled. I cannot comment further as I don’t know what, if anything, has been changed. For example, if the wrong version of riscv-pk was checked out, it might not recognise the UART hardware configuration, and so stuff could be happening, but it would be invisible.

jrrk, 2019-09-25:
You could try building BBL with the dummy payload (just remove the kernel configuration argument) and see if that works. That would establish that your build procedure is OK.

jrrk, 2019-09-25:
You talk about mixing and matching release files with files you have built yourself. Please bear in mind these binaries are not released, they are just candidates and may not be full compatible with the latest source code. The v0.7 is not released yet.

mathmax12, 2019-09-25:
Thanks a lot for always give suggestion to me. I greatly appreciate that!

> You could try building BBL with the dummy payload (just remove the kernel configuration argument) and see if that works. That would establish that your build procedure is OK.

Is [$(LINUX)/.config](https://github.com/lowRISC/lowrisc-chip/blob/64986933820090fb647ac7066079a100e62b66f4/Makefile#L46) the configureation that I should remove?

Thanks.

jrrk, 2019-09-25:
No, I meant the configure command line inside riscv-pk/serial. This has an auto-generated Makefile but you can regenerate this without a kernel selection. This is covered in the riscv-pk documentation (which is a good idea to re-read, to ensure adequate understanding of the process)

mathmax12, 2019-09-26:
> The boot sequence looks to be valid, but BBL never prints anything. I would expect the machine configuration to be printed if that option is enabled. I cannot comment further as I don’t know what, [...]

The issue is solved.
I used an older version of riscv-pk.
After update it to yours, the problem sloved.
Now the booting from SD card with my self-build boot.bin works.

Next I will try to boot the system from flash. 

Thanks.

mathmax12, 2019-09-26:
> The existing build environment already merges the .bit file and boot.bin into a single file, you probably noticed that already. When you build Linux, you can replace the .cpio file with any root filing system of your choice. However, beware size limits on the combined .mcs file, on the Genesys2, the total including bitstream will be limited to 16MBytes or so (from memory, check this figure) if it needs to fit in QSPI memory.

Now use the **write_cfgmem.tcl** to merge the **bbl** and **genesys_ariane.bit** to generate the *.mcs file. (I did change the ariane_peripherals_xilinx.sv according to our previous talk). After configuring the flash with this *.mcs file, the system can find the bbl but stoped at below.
 
```
Hello from Ariane! Please wait a moment...
Relocating to DDR memory
Hello World!
swp[0] = 0
swp[1] = DEADBEEF
swp[2] = 0
swp[3] = 0
swp[4] = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
QSPI boot
load QSPI to DDR memory
load ELF to DDR memory
00000000 7F 45 4C 46 02 01 01 00 00 00 00 00 00 00 00 00 
00000010 02 00 F3 00 01 00 00 00 00 00 00 80 00 00 00 00 
00000020 40 00 00 00 00 00 00 00 B0 3C A2 00 00 00 00 00 
00000030 05 00 00 00 40 00 38 00 03 00 40 00 0B 00 0A 00 
00000040 01 00 00 00 05 00 00 00 00 10 00 00 00 00 00 00 
00000050 00 00 00 80 00 00 00 00 00 00 00 80 00 00 00 00 
00000060 A8 6F 00 00 00 00 00 00 A8 6F 00 00 00 00 00 00 
00000070 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[0]: elfn(80000000,0x1000,0x6fa8);
00000078 01 00 00 00 06 00 00 00 00 80 00 00 00 00 00 00 
00000088 00 70 00 80 00 00 00 00 00 70 00 80 00 00 00 00 
00000098 61 10 00 00 00 00 00 00 88 B0 00 00 00 00 00 00 
000000A8 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[1]: elfn(80007000,0x8000,0x1061);
memset(80008061,0,0xa027);
000000B0 01 00 00 00 04 00 00 00 00 00 20 00 00 00 00 00 
000000C0 00 00 20 80 00 00 00 00 00 00 20 80 00 00 00 00 
000000D0 2C 0F 82 00 00 00 00 00 2C 0F 82 00 00 00 00 00 
000000E0 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[2]: elfn(80200000,0x200000,0x820f2c);
Boot the loaded program at address 80000000...
bbl loader
 {
  #address-cells = <0x00000002>;
  #size-cells = <0x00000002>;
  compatible = "eth,ariane-bare-dev";
  model = "eth,ariane-bare";
  chosen {
    stdout-path = <0x2f736f63 0x2f756172 0x74403431 0x30303030 0x30303a31 0x31353230>;
  }
  cpus {
    #address-cells = <0x00000001>;
    #size-cells = <0x00000000>;
    timebase-frequency = <0x0007a120>;
    cpu@0 {
      clock-frequency = <0x02faf080>;
      device_type = "cpu";
      reg = <0x00000000>;
      status = "okay";
      compatible = "ethz, ariane", "riscv";
      riscv,isa = "rv64imafdc";
      mmu-type = "riscv,sv39";
      tlb-split;
      interrupt-controller {
        #interrupt-cells = <0x00000001>;
        interrupt-controller;
[... 151 lines cut ...]
swp[1] = DEADBEEF
swp[2] = 0
swp[3] = 0
swp[4] = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
QSPI boot
load QSPI to DDR memory
load ELF to DDR memory
00000000 7F C5 4C 46 02 01 01 00 00 00 00 00 00 00 00 00 
00000010 02 00 F3 00 01 00 00 00 00 00 00 80 00 00 00 00 
00000020 40 00 00 00 00 00 00 00 00 DE DD 00 00 00 00 00 
00000030 05 00 00 00 40 00 38 00 03 00 40 00 12 00 11 00 
elf read failed with code 2

````

Is my guess and what I did correct?
Thanks a lot!

jrrk, 2019-09-26:
The message you got was consistent with the .cpio file not being read properly. This could be due to the boot loader not knowing how big the code that it is trying to read is. I set this value to a simple default for QSPI mode, you might need to increase it in the boot loader source. For some reason the QSPI fails unless you read the entire image in one go (another mystery).

mathmax12, 2019-09-26:
Thanks.
Here the "message" in "The **message** you got was consistent with the .cpio file not being read properly" is **[    1.122890] ---[ end Kernel panic - not syncing: no cpio magic ]---** or **elf read failed with code 2** ?

jrrk, 2019-09-27:
The first example, where it actually boots, the cpio magic is a problem. There is more than one format of cpio, Linux likes the new format only. You can copy the syntax in my build scripts or put your files in a directory tree and let the Linux build process do it as per the kernel documentation. I don’t know what is wrong in your second example, it looks like the first stage boot loader didn’t understand the QSPI format.

mathmax12, 2019-09-27:
In the first example, I didn't change the initramfs.cpio file and use the one from this the [v0.7-rc1](https://github.com/lowRISC/lowrisc-chip/releases/tag/v0.7-rc1). The bbl build from the make file under the [lowrisc-chip-ariane-v0.7](https://github.com/lowRISC/lowrisc-chip/tree/ariane-v0.7).
If I copy this bbl to SD card as boot.bin and with a rootfs, the linux booting works well.    But without of rootfs just copy the merged bbl + bit to the flash throws the above issue. I think the initramfs is enough for booting and no need the rootfs, right? 

I just notice one thing the size of genesys_ariane_xilinx.bit is 5MB and the size of bbl is around 15MB.
But the size merged ariane_xilinx.mcs file is around 54MB. The MCS file is a HEX file where two ASCII chars are used to represent each byte of data.  So I think the size increase is normal.

However, the capacity of the flash on the genesys board is 32 MB < 54 MB.  Is this a possible reason?
I will try to use bin file insead of mcs tomorrow for a try.

Thanks.

jrrk, 2019-09-27:
The Xilinx tool will give an error if you try to burn an MCS file that is too big, you would not get the cpio magic error due to lack of rootfs. You could try modifying the first stage boot loader to print the MD5SUM of the BBL that it reads, before booting. You will need to use the genesys2_ariane_new target to change it, or CB perhaps the clock rate to the QSPI needs to be reduced (but it worked on my board)

mathmax12, 2019-09-28:
Update:
I tried to program the Flash with *.bin instead of *.mcs. The same issue persists (Kernel panic).  The **panic message randomly varies** when **programming the flash mut-times** with the same *.bin/MCS file.
 
```
---[ end Kernel panic - not syncing: no cpio magic ]---
or
---[ end Kernel panic - not syncing: junk within compressed archive ]---
or
---[ end Kernel panic - not syncing: broken padding ]---
```
I tried to slow the clock of qspi flash but got a timing violation.

I wonder do you mind I use your flash configuration file (i.e. *.mcs) to program my board.
Or you can try with mine on your board if possible.
[ariane_xilinx.zip](https://github.com/lowRISC/lowrisc-chip/files/3664895/ariane_xilinx.zip)

See if there some difference.

Thanks.

jrrk, 2019-09-28:
I will be back in the Laboratory 1st October, I can check my latest image, see if it works still, if so I can upload it for you.

mathmax12, 2019-09-28:
Thanks a lot!

Best regards,
Sean

jrrk, 2019-10-01:
I am testing the stable version of the flash image now. It is based on git commit 71efeb8067751ac79dc803c40ad34819e827bde6 (Jul 2nd)

These are the applicable submodules:

jrrk2@sigurd:/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7$ git submodule
 e873743b547fbf39bd08f25a0bfa7db38eb61794 ariane (v4.2.0)
 031543c7f555c23d8dfa588983d5e404d7fc7fbf debian-riscv64 (remotes/origin/ariane-v0.7)
 436c2ee880075942aaaa8852d1e7ecc72d8c2112 fpga/src/apb_uart (v0.0.1-5-g436c2ee)
 72330646196ae7a7bc670db7aadfd7d34f4cb76f fpga/src/ariane-ethernet (remotes/origin/ariane-v0.7)
 a8af1bb99ad4fbc2c500c08f6a2b990d62fbad68 fpga/src/spi_mem_programmer (remotes/origin/ariane-v0.7)
 7ae23bc1974f8738b5009125873e43e6a8b1d848 qemu (heads/master)
 b70ef508cd5e8508ad5c43e4582ebda094baa925 riscv-gnu-toolchain (v20180629-67-gb70ef50)
 45b057965881b6698e8277c5f294a0d73cb2dbcd riscv-pk (remotes/origin/ariane-v0.7)
 e6feeac0e9cebf00de84252a7c1e1696662288aa rocket-chip (remotes/origin/ariane-v0.7)
 4a04455b1ecd79c978f645fda34f79f125e959e8 src/OpenIP (remotes/origin/ariane-v0.7)
jrrk2@sigurd:/local/scratch/jrrk2/lowrisc-chip-ariane-v0.7$ 

This is the log on the serial port when no SD-Card is installed

Hello from Ariane! Please wait a moment...
Relocating to DDR memory
Hello World!
swp[0] = 20
swp[1] = DEADBEEF
swp[2] = 0
swp[3] = 0
swp[4] = 0
Setup MAC addr
QSPI OEM[0] = 0
QSPI OEM[1] = 18
QSPI OEM[2] = 3e
QSPI OEM[3] = 2
QSPI OEM[4] = e3
QSPI OEM[5] = 7f
Switch setting = 20,20
Random seed = 0
Switch setting = 20,20
Random seed = 0
Switch setting = 20,20
Random seed = 0
Switch setting = 20,20
Random seed = 0
QSPI boot
load QSPI to DDR memory
load ELF to DDR memory
00000000 7F 45 4C 46 02 01 01 00 00 00 00 00 00 00 00 00 
00000010 02 00 F3 00 01 00 00 00 00 00 00 80 00 00 00 00 
00000020 40 00 00 00 00 00 00 00 A8 3C A2 00 00 00 00 00 
00000030 05 00 00 00 40 00 38 00 03 00 40 00 0B 00 0A 00 
00000040 01 00 00 00 05 00 00 00 00 10 00 00 00 00 00 00 
00000050 00 00 00 80 00 00 00 00 00 00 00 80 00 00 00 00 
00000060 C8 6F 00 00 00 00 00 00 C8 6F 00 00 00 00 00 00 
00000070 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[0]: elfn(80000000,0x1000,0x6fc8);
00000078 01 00 00 00 06 00 00 00 00 80 00 00 00 00 00 00 
00000088 00 70 00 80 00 00 00 00 00 70 00 80 00 00 00 00 
00000098 61 10 00 00 00 00 00 00 88 B0 00 00 00 00 00 00 
000000A8 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[1]: elfn(80007000,0x8000,0x1061);
memset(80008061,0,0xa027);
000000B0 01 00 00 00 04 00 00 00 00 00 20 00 00 00 00 00 
000000C0 00 00 20 80 00 00 00 00 00 00 20 80 00 00 00 00 
000000D0 2C 0F 82 00 00 00 00 00 2C 0F 82 00 00 00 00 00 
000000E0 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[2]: elfn(80200000,0x200000,0x820f2c);
Boot the loaded program at address 80000000...
bbl loader
 {
  #address-cells = <0x00000002>;
  #size-cells = <0x00000002>;
[... 196 lines cut ...]
-rw-rw-r--    1 0        0               26 May 23  2019 Makefile
drwxrwxr-x    2 0        0             1520 Jan  1 00:00 bin
drwxrwxr-x    2 0        0              240 Jul  2  2019 dev
drwxr-xr-x    2 0        0               40 Jan  1 00:00 dos
drwxrwxr-x    2 0        0               40 Jul  2  2019 etc
drwxrwxr-x    2 0        0               40 Jul  2  2019 home
-rwxrwxr-x    1 0        0              668 Jul  2  2019 init
drwxrwxr-x    3 0        0               80 Jul  2  2019 lib
lrwxrwxrwx    1 0        0               12 Jan  1 00:00 linuxrc -> /bin/busybox
drwxrwxr-x    2 0        0               40 Jul  2  2019 mnt
drwxrwxr-x    2 0        0               40 Jul  2  2019 nfs
drwxrwxr-x    2 0        0               40 Jul  2  2019 proc
drwxrwxr-x    2 0        0               40 Jul  2  2019 root
drwxrwxr-x    2 0        0               40 Jul  2  2019 run
drwxrwxr-x    2 0        0              980 Jan  1 00:00 sbin
drwxrwxr-x    2 0        0               40 Jul  2  2019 sys
drwxrwxr-x    2 0        0               40 Jul  2  2019 tmp
drwxrwxr-x    6 0        0              120 Jul  2  2019 usr
/ # 

The associated .bit and .mcs files are here:

[71efeb8067751ac79dc803c40ad34819e827bde6.zip](https://github.com/lowRISC/lowrisc-chip/files/3675195/71efeb8067751ac79dc803c40ad34819e827bde6.zip)

In addition, if you have a VGA monitor, you will see the lowRISC logo during booting.

mathmax12, 2019-10-01:
Thanks for that. I appreciate it.

Does this information come from another board rather than Genesys II?
I can see the **switch value is** different from mine on the GenesysII board:
```
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
```

mathmax12, 2019-10-02:
Before I used the latest version of the submodules.
Now I switched to the above-suggested versions. 
The booting made some pregresss 
```
Hello from Ariane! Please wait a moment...
Relocating to DDR memory
Hello World!
swp[0] = 0
swp[1] = DEADBEEF
swp[2] = 0
swp[3] = 0
swp[4] = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
Switch setting = 0,0
Random seed = 0
QSPI boot
load QSPI to DDR memory
load ELF to DDR memory
00000000 7F 45 4C 46 02 01 01 00 00 00 00 00 00 00 00 00 
00000010 02 00 F3 00 01 00 00 00 00 00 00 80 00 00 00 00 
00000020 40 00 00 00 00 00 00 00 B0 3C A2 00 00 00 00 00 
00000030 05 00 00 00 40 00 38 00 03 00 40 00 0B 00 0A 00 
00000040 01 00 00 00 05 00 00 00 00 10 00 00 00 00 00 00 
00000050 00 00 00 80 00 00 00 00 00 00 00 80 00 00 00 00 
00000060 A8 6F 00 00 00 00 00 00 A8 6F 00 00 00 00 00 00 
00000070 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[0]: elfn(80000000,0x1000,0x6fa8);
00000078 01 00 00 00 06 00 00 00 00 80 00 00 00 00 00 00 
00000088 00 70 00 80 00 00 00 00 00 70 00 80 00 00 00 00 
00000098 61 10 00 00 00 00 00 00 88 B0 00 00 00 00 00 00 
000000A8 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[1]: elfn(80007000,0x8000,0x1061);
memset(80008061,0,0xa027);
000000B0 01 00 00 00 04 00 00 00 00 00 20 00 00 00 00 00 
000000C0 00 00 20 80 00 00 00 00 00 00 20 80 00 00 00 00 
000000D0 2C 0F 82 00 00 00 00 00 2C 0F 82 00 00 00 00 00 
000000E0 00 10 00 00 00 00 00 00 7F 45 4C 46 02 01 01 00 
Section[2]: elfn(80200000,0x200000,0x820f2c);
Boot the loaded program at address 80000000...
bbl loader
 {
  #address-cells = <0x00000002>;
  #size-cells = <0x00000002>;
  compatible = "eth,ariane-bare-dev";
  model = "eth,ariane-bare";
  chosen {
    stdout-path = <0x2f736f63 0x2f756172 0x74403431 0x30303030 0x30303a31 0x31353230>;
  }
  cpus {
    #address-cells = <0x00000001>;
    #size-cells = <0x00000000>;
    timebase-frequency = <0x0007a120>;
    cpu@0 {
      clock-frequency = <0x02faf080>;
      device_type = "cpu";
      reg = <0x00000000>;
      status = "okay";
      compatible = "ethz, ariane", "riscv";
      riscv,isa = "rv64imafdc";
      mmu-type = "riscv,sv39";
      tlb-split;
      interrupt-controller {
        #interrupt-cells = <0x00000001>;
        interrupt-controller;
        compatible = "riscv,cpu-intc";
[... 165 lines cut ...]
149     }
**150** /*  switch (sw >> 5)
**151**     {
**152**     case 0x0: printf("SD boot\n"); sd_main(sw); break;
**153**     case 0x1: printf("QSPI boot\n"); qspi_main(sw); break;
**154**     case 0x2: printf("DRAM test\n"); dram_main(sw); break;
**155**     case 0x4: printf("TFTP boot\n"); eth_main(); break;
**156**     case 0x6: printf("Cache test\n"); cache_main(); break;
**157**     case 0x7: printf("Keyboard test\n"); keyb_main(); break;
**158**     }
**159**  */
**160**     printf("QSPI boot\n");
**161**     qspi_main(sw);
162 
163     while (1)
164     {
165       // do nothing
166     }
167 }
168 
```

Below is the change for the  lowrisc-chip_ariane0d7/lowrisc-chip/fpga/src/ariane_peripherals_xilinx.sv file. (**[remove the Ethernet+SPI part between 296-487](https://github.com/lowRISC/lowrisc-chip/blob/ariane-v0.7/fpga/src/ariane_peripherals_xilinx.sv)**)

Is my modification correct?

jrrk, 2019-10-02:
The bitstream/MCS file I sent to you is for the GenesysII. You should remove all peripherals that aren't available in your target system from the fpga/src/genesys2.dts file, this will prevent Linux from getting confused about any peripherals that you aren't using/are missing in your target configuration.

I'm not totally clear what the outcome of running my bitstream on your genesysII board was, did it boot OK, or are there other issues (you need to set the correct switches (0x20) to boot from QSPI) as you already figured out.

mathmax12, 2019-10-03:
Hey Dr. Jonathan,
Thanks a lot!
After modified the *.dts file, the soc can boot successfully.
Thank you very much for alway give me help.
I highly appreciate that.

Best regards,
Sean

monrick0303, 2019-12-17:
> sshd does not allow root to login without a secret key exchange. You could copy your public key to /root/.ssh/authorized_keys on the SD-Card (partition 2) to see if it helps. Also test your [...]

I also stuck at `[ ok ] Starting OpenBSD Secure Shell server: sshd.`
how do modify my root filing system to boot straight into a shell?
Thanks your help.

jrrk2, 2019-12-17:
You can always modify your filing system live on the host. Just insert the SD-Card in its adapter into the host PC, change directory to the mount point (or if not mounted automatically, mount it yourself on /mnt), and then you can just:

    rm etc/rc?.d/S01ssh

You can add temporary debugging commands such as /bin/bash in etc/rc.local

If you need more extensive modifications such as adding or removing packages, this can be done using:

    sudo chroot <mount-point>
    apt add/remove/etc...