lowRISC/lowrisc-chip #123: error while building riscv-tools

AnouarNechi, 2019-10-11:
Hello
I am trying to work with lowrisc So I followed this link : https://www.lowrisc.org/docs/debug-v0.3/zedboard/ 
So, I have set up the **env** as following :
```
export TOP=$PWD
export RISCV=$TOP/riscv-tools/riscv-gnu-toolchain
export PATH=$PATH:$RISCV/bin
export OSD_ROOT=$TOP/opensocdebug
export PATH=$OSD_ROOT/bin:$PATH
export FPGA_BOARD=$TOP/fpga/board/zed
```
Then I moved to the **riscv-tools** and I used the build script to set up the cross compiler:
```
cd riscv-tools
./build.sh
```
But , I keep getting the following error when reaching **newlib-gcc/gcc** :
```
/lowrisc-chip/riscv-tools/riscv-gnu-toolchain/build/src/newlib-gcc/gcc/ubsan.c:1473:23: error: ISO C++ forbids comparison between pointer and integer [-fpermissive]
       || xloc.file == '\0' || xloc.file[0] == '\xff'

```
Is there a solution please?

jrrk, 2019-10-12:
It looks like you are compiling a very old release with a much newer host gcc. You could try the suggestion of adding -fpermissive to the host compiler flags, or install a legacy version of host gcc.

AnouarNechi, 2019-10-14:
Yes I am using the debug v3 version because I want to work on ZedBoard and this is what the official tutorial saying.
Does the newer version work on zed board also?

KB-Q, 2020-12-30:
Hello... Any updates on this issue?

jrrk, 2020-12-30:
The zed board port was carried out by a third party and was never maintained by LowRISC. Unless you install the exact OS that was documented  at the time, you are out of luck. You could also try installing an older gcc on your current OS but this is outside the scope of the support I can offer because it is open ended back porting.

hgurbuzz, 2022-02-13:
Hi,

I have set up the env as following :
export RISCV=/opt/riscv
export PATH=$PATH:$RISCV/bin 

Then I moved to the riscv-tools and I used the build script to set up the cross compiler:
cd riscv-tools
sudo ./build.sh

But , I keep getting the following error 
"Please set the RISCV environment variable to your preferred install path"

what am i missing?
Thank you