lowRISC/lowrisc-chip #129: opocd Error: Unsupported DTM version: 9

shirlynan, 2019-11-04:
I 'm trying debugging riscv with jtag. but always get the following error in openocd. Has anyone idea how to make it work....
Open On-Chip Debugger 0.10.0+dev-00818-g7c82a7b (2019-10-27-12:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
riscv.cpu
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x43651093 (mfg: 0x049 (Xilinx), part: 0x3651, ver: 0x4)
Error: Unsupported DTM version: 9
Info : Listening on port 3333 for gdb connections

jrrk2, 2019-11-04:
You should be using openocd commit dc312f57e626958ce2e3b8edfe5445695b37d58b from 	https://github.com/lowRISC/riscv-openocd.git, which reports version 0.10.0+dev-00165-gdc312f57

This patch is necessary to support the JTAG chain length in Xilinx FPGAs and other discrepancies as described in the lowrisc.org documentation.

shirlynan, 2019-11-04:
> You should be using openocd commit dc312f57e626958ce2e3b8edfe5445695b37d58b from https://github.com/lowRISC/riscv-openocd.git, which reports version 0.10.0+dev-00165-gdc312f57
>
> This patch is [...]

Thanks for your reply. I tried to build this openocd version, but failed with errors: 

src/target/riscv/riscv-011.c:1961:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);

I guess this should be caused by version mismatch of pre-built riscv tools and openocd? So i have to change my prebuilt riscv tools? but to which version?

jrrk2, 2019-11-04:
Newer versions of gcc are stricter, try adding  -Wno-error to your flags

shirlynan, 2019-11-04:
> Newer versions of gcc are stricter, try adding -Wno-error to your flags

unfortunately it doesn't work...

jrrk2, 2019-11-04:
You need to give more detail.

shirlynan, 2019-11-04:
> You need to give more detail.
I added -Wno-error flag to make command but the make errors are still there:
src/target/riscv/riscv-011.c: In function ‘deassert_reset’:
src/target/riscv/riscv-011.c:1950:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);
                                ^
In file included from src/target/riscv/riscv-011.c:18:0:
./src/jtag/jtag.h:345:6: note: expected ‘struct scan_field *’ but argument is of type ‘struct scan_field * (*)(struct target *)’
 void jtag_add_ir_scan(struct jtag_tap *tap,
      ^
src/target/riscv/riscv-011.c: In function ‘read_memory’:
src/target/riscv/riscv-011.c:1961:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);

src/target/riscv/riscv-011.c: In function ‘write_memory’:
src/target/riscv/riscv-011.c:2128:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);

Environment Info:
git branch xilinx-boom, KC705, xilinx usb-jtag cables.

shirlynan, 2019-11-04:
> You need to give more detail.

I added -Wno-error flag to make command but the make errors are still there:

src/target/riscv/riscv-011.c: In function ‘deassert_reset’:
src/target/riscv/riscv-011.c:1950:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);
^
In file included from src/target/riscv/riscv-011.c:18:0:
./src/jtag/jtag.h:345:6: note: expected ‘struct scan_field ’ but argument is of type ‘struct scan_field * ()(struct target *)’
void jtag_add_ir_scan(struct jtag_tap *tap,
^
src/target/riscv/riscv-011.c: In function ‘read_memory’:
src/target/riscv/riscv-011.c:1961:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);

src/target/riscv/riscv-011.c: In function ‘write_memory’:
src/target/riscv/riscv-011.c:2128:32: error: passing argument 2 of ‘jtag_add_ir_scan’ from incompatible pointer type [-Werror=incompatible-pointer-types]
jtag_add_ir_scan(target->tap, &select_dbus, TAP_IDLE);

Environment Info:
git branch xilinx-boom, KC705, xilinx usb-jtag cables.

closed 2019-12-11 by shirlynan