lowRISC/lowrisc-chip #146: documentation page not found

lan0017, 2019-11-30:
The page is not found.So can anyone provide the documentation.
https://www.lowrisc.org/docs/

jrrk, 2019-11-30:
My apologies on behalf the lowrisc.org.

During site maintenance you can always access the docs section as follows:

git clone https://github.com/lowrisc/lowrisc-site

cd lowrisc-site/

hugo server

The information you require will appear at localhost:1313/docs in your 
web browser.

IMVector, 2019-12-04:
> My apologies on behalf the lowrisc.org. During site maintenance you can always access the docs section as follows: git clone https://github.com/lowrisc/lowrisc-site cd lowrisc-site/ hugo server The information you require will appear at localhost:1313/docs in your web browser.
> […](#)

sorry  to bother you, when I run the server there are some error, what  should I do to solve them

Building sites … ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "blog/zero-point-six-release.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/refresh-v0.6/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "blog/zero-point-six-release.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/refresh-v0.6/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "blog/zero-point-six-release.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/refresh-v0.6/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/ethernet-v0.5/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/ethernet-v0.5/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/minion-v0.4/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/minion-v0.4/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/debug-v0.3/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/debug-v0.3/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/untether-v0.2/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/untether-v0.2/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/tagged-memory-v0.1/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "docs/lowrisc-releases.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/tagged-memory-v0.1/_index.md".
ERROR 2019/12/04 22:16:08 error processing shortcode "_internal/shortcodes/ref.html" for page "blog/zero-point-five-release.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/ethernet-v0.5/_index.md".
[... cut ...] ternal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/tagged-memory-v0.1/_index.md".
ERROR 2019/12/04 22:17:43 error processing shortcode "_internal/shortcodes/ref.html" for page "blog/lowrisc-tagged-memory-preview-release.md": template: _internal/shortcodes/ref.html:1:73: executing "_internal/shortcodes/ref.html" at <ref .Page (.Get 0)>: error calling ref: No page found with path or logical name "docs/tagged-memory-v0.1/_index.md".

jrrk, 2019-12-04:
For compatibility with our later work we had to upgrade our docs to version 0.55.6 of Hugo ( the one that comes with Ubuntu 18.04 ). Unfortunately there doesn’t seem to be a way to maintain older versions otherwise.

PRieblinger-esatus, 2020-01-23:
> 
> 
> For compatibility with our later work we had to upgrade our docs to version 0.55.6 of Hugo ( the one that comes with Ubuntu 18.04 ). Unfortunately there doesn’t seem to be a way to maintain older versions otherwise.

The version mentioned above appears to be incorrect. The version of Hugo which can be installed in Ubuntu 18.04 at the time of writing is v0.40.1. I just verified this on Ubuntu Mate Ubuntu Mate 18.04.3 LTS 64-bit. This version is not compatible with the lowrisc-site repository and throws the above errors on my system, too.
For future reference: I installed Hugo as a snap package after finding out that the snap version of Hugo at the time of writing was 0.62.2, so quite a bit ahead of the version in the Ubuntu repository.
The exact commands I used to gain access to the docs were the following:
```
sudo snap install hugo # install hugo as a snap package
git clone https://github.com/lowRISC/lowrisc-site.git # clone the lowrisc-site repo
cd lowrisc-site # change directory into the lowrisc-site folder
snap run hugo server # run the hugo snap package as a server using the current folder
```
The last command runs the local Hugo server on which the lowrisc docs can be accessed by a webbrowser on the same machine using the URL `localhost:1313/docs`.

jrrk2, 2020-01-23:
I find snapd doesn't work if /home is an NFS mount (as it is in my case). An alternative is to manually install with
sudo dpkg -i libsass1_3.5.5-2_amd64.deb ~/hugo_0.52-1_amd64.deb (manually downloaded from the disco release)

jrrk, 2020-03-02:
If further comments arise, just re-open the issue.

closed 2020-03-02 by jrrk