lowRISC/lowrisc-chip #147: switch_root to real rootfs

monrick0303, 2019-12-03:
I use `buildroot of ariane-sdk` build image,
and i want to mount `rootfs.tar.xz of lowrsic` as real rootfs.
i modify init script of initramfs:
[image]
it will switch to real rootfs by `switch_root`.

During the linux boot process on Ariane:
`[   27.512532] mmcblk0: mmc0:0000 SS04G 3.69 GiB`
`[   28.270961]  mmcblk0: p1 p2`
`Mounting SD root`
`[   34.961062] random: fast init done`
`[   39.051177] EXT4-fs (mmcblk0p2): recovery complete`
`[   39.691713] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)`
`Mounting proc`
`Mounting sysfs`
`Mounting devtmpfs`
`Mounting devpts`
`Mounting tmpfs`
`Executing switch_root for mmc`
`INIT: version 2.93 booting`
`[   93.129442] random: crng init done`
`[   93.129442] random: crng init done`
`[info] Using makefile-style concurrent boot in runlevel S.`
`[info] Setting the system clock.`
`hwclock: Cannot access the Hardware Clock via any known method.`
`hwclock: Use the --verbose option to see the details of our search for an access method.`
`[warn] Unable to set System Clock to: Thu Jan 1 00:02:56 UTC 1970 ... (warning).`
`[ ok ] Activating swap...done.`
`[  197.397739] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)`
`[  197.397739] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)`
`[warn] Creating compatibility symlink from /etc/mtab to /proc/mounts. ... (warning).`
`[....] Checking file systems...fsck from util-linux 2.33.1`
`done.`
`[ ok ] Cleaning up temporary files... /tmp.`
`[ ok ] Mounting local filesystems...done.`
`[ ok ] Activating swapfile swap...done.`
`[ ok ] Cleaning up temporary files....`
`[ ok ] Starting Setting kernel variables: sysctl.`
`[....] Configuring network interfaces...Internet Systems Consortium DHCP Client 4.4.1`
`Copyright 2004-2018 Internet Systems Consortium.`
`All rights reserved.`
`For info, please visit https://www.isc.org/software/dhcp/`

`[  430.874833] Open device, request interrupt 3`
`[  430.874833] Open device, request interrupt 3`
`Listening on LPF/eth0/23:01:00:89:07:02`
`Sending on   LPF/eth0/23:01:00:89:07:02`
`Sending on   Socket/fallback`
`Created duid "\000\001\000\001\307\222\276/#\001\000\211\007\002".`
`DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 6`
`DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 11`
`DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 17`
`DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 17`
`DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 8`
`DHCPDISCOVER on eth0 to 255.255.255.255 port 67 interval 2`
`No DHCPOFFERS received.`
`No working leases in persistent database - sleeping.`
`done.`
`[ ok ] Starting RPC port mapper daemon: rpcbind.`
`[ ok ] Starting NFS common utilities: statd idmapd.`
`[ ok ] Cleaning up temporary files....`
`INIT: Entering runlevel: 2`

It will stuck at `sbin/int` process,
sometime at `INIT: Entering runlevel: 2`,
or sometime at `[ ok ] Activating swapfile swap...done.`
I don't know why.

If I change `switch_root /mnt /sbin/int` to `chroot /mnt`, 
during the linux boot process on Ariane:
`Executing chroot for mmc`
`bash: cannot set terminal process group (-1): Inappropriate ioctl for device`
`bash: no job control in this shell`
`[   89.290585] random: crng init done`
`[   89.290585] random: crng init done`
`root@ariane-fpga:/#`

It looks successful, except `bash: cannot set terminal process group (-1): Inappropriate ioctl for device`
But I still want to use `switch_root`,
because it will delete all content of old rootfs(initramfs).

Can you help me?
Thanks You.

jrrk, 2019-12-03:
Dear monrick,

I'm going to assume that since you mention Ariane, you are talking about 
the forthcoming ariane-v0.7 release.

First off, why don't you use the rescue kernel that comes with 
lowrisc-chip, this

is much more economical than the size of buildroot.

It's not always a good idea to mix buildroot images with Debian images. 
Each environment has its

own concept of shared libraries, release status, version control etc, 
and you can easily end

up with incompatibilities if processes need to talk to each other, or 
different versions of

header files. The newer lowrisc releases just extract the minimum stuff 
needed to boot from a Debian image,

and then puts this into an initramfs. A further point is that the 
startup script assumes some sort of network

connection, if you do not have this it could cause the sysvinit process 
to fail. The new release uses systemd

which has a higher memory overhead, but launches threads in parallel 
such that a stuck network connection does not

prevent the overall boot from completing.

I raised a further point yesterday in respect of Ariane treatment of I/O 
devices.

https://github.com/pulp-platform/ariane/issues/356

This could be relevant to your problems.

Meanwhile you can experiment with the instructions here for the 
forthcoming official lowRISC Ariane release:

https://kimmitt.uk/docs/docs/

If you use the so-called rescue Linux image, it already does (most of) 
what you are trying to achieve.

Regards,

Jonathan

monrick0303, 2019-12-03:
Thanks you.
So I can use the so-called rescue Linux image run on [ariane of pulp-platform](https://github.com/pulp-platform/ariane) rather than [ariane of lowrisc](https://github.com/lowRISC/lowrisc-chip/tree/ariane-v0.7) ?

jrrk, 2019-12-03:
These Ariane versions are largely the same apart from cache sizes. If 
you build with the pulp-platform

you will get a different set of peripherals, but you can adjust the 
startup scripts to suit.

However I am a little confused, why you are asking on the lowrisc-chip 
issues github, if your intended

use is the pulp-platform ? On the whole it is a bad idea to mix and 
match unless you want to take responsibility

for re-testing everything you use.

monrick0303, 2019-12-04:
Thanks You.
I have posted on [ariane-sdk#28](https://github.com/pulp-platform/ariane-sdk/issues/28)
After that.
Since it stuck at `/sbin/int of lowrisc real rootfs`, so i am asking on here.

In general,
I want to use `lowrisc rootfs` run on `ariane of pulp-platform`.
(because it have more command.)

I will test more.
Thanks You.

monrick0303, 2019-12-17:
> These Ariane versions are largely the same apart from cache sizes. If you build with the pulp-platform you will get a different set of peripherals, but you can adjust the startup scripts to suit. [...]

Hello,
`a different set of peripherals` is mean device tree?
where i can adjust the startup scripts?
`lowrisc-chip-ariane-v0.7/linux-5.3.8-lowrisc/arch/riscv/boot/dts` or `/lowrisc-chip-ariane-v0.7/fpga/src/generic.dts` ?

Thanks your help.

jrrk2, 2019-12-17:
If you want to add or remove peripherals you need to make changes in at least three places:

    1. In the Verilog that gets compiled by the Vivado tool
    2. In the Linux Kernel to make provision for a suitable device driver
    3. In the device tree to specify the address of the device and the driver to be used
    4. Optionally, adding packages to the debian installation script to use the new device

If you only want to remove or disable a peripheral, it is sufficient to remove it from fpga/src/generic.dts
and update the bitstream with one of the relatively fast *_new targets, it will not get loaded in Linux, and any attempt to use it will fail.

monrick0303, 2019-12-17:
So,if i don't want to rebuild the bitsteam,i just
1.disable the device in linux-5.3.8-lowrisc/arch/riscv/configs/defconfig (search for entries that begin with CONFIG_LOWRISC_*), and rebuild Linux.
2.make bbl 
3.install the sd-card by lowrisc-quickstart 
(The no.3 step can use GPT partition table of ariane-sdk ?)

Now i can boot linux-5.3.8-lowrisc with bitstream of pulp-platform.
But it will `mount -t ext4 /dev/mmcblk0p2 /mnt failed`.
it might can't read second partition of sd-card.

Thanks your help!

jrrk2, 2019-12-17:
I haven't tried the pulp-platform bitstream recently. Last time I tried it it used the Xilinx SPI driver for the SD-Card. It works only in one-bit mode, but is supported by Linux. You will need to use a matching device tree blob and kernel. It will not work with the unmodified LowRISC config. You could prepare a kernel config which is a superset of what LowRISC and pulp-platform will require, but I don't understand the motivation for this. Another difference is the partition table, LowRISC uses a DOS partition to store the kernel, pulp-platform stores it in a boot partition. If you can explain why the lowrisc bitstream doesn't suit your purpose, I could be more helpful.

monrick0303, 2019-12-18:
Hello,
I can boot linux and mount rootfs after modify linux config.
but i have some problem.

following is boot process:
[   16.602667] mmcblk0: mmc0:0000 SS04G 3.69 GiB
[   16.823504] Freeing unused kernel memory: 2524K
[   16.833640] This architecture does not have kernel memory protection.
[   16.852885] Run /init as init process
[   18.584001]  mmcblk0: p1 p2
Mounting SD root
[   24.511067] random: fast init done
[   72.367935] EXT4-fs (mmcblk0p2): recovery complete
[   72.468043] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
Mounting proc
Mounting sysfs
Mounting devtmpfs
Mounting devpts
Mounting tmpfs
Executing switch_root for mmc
[   85.132918] random: crng init done
INIT: version 2.93 booting
[info] Using makefile-style concurrent boot in runlevel S.
[info] Setting the system clock.
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --verbose option to see the details of our search for an access method.
[warn] Unable to set System Clock to: Thu Jan 1 00:03:22 UTC 1970 ... (warning).
[ ok ] Activating swap...done.
[  220.914645] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[  220.914645] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[....] Checking file systems...fsck from util-linux 2.33.1
done.
[ ok ] Cleaning up temporary files... /tmp.
[ ok ] Mounting local filesystems...done.
[ ok ] Activating swapfile swap...done.
[ ok ] Cleaning up temporary files....
[ ok ] Starting Setting kernel variables: sysctl.
[....] Configuring network interfaces...Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Cannot find device "eth0"
Failed to get interface index: No such device

If you think you have received this message due to a bug rather
than a configuration issue please read the section on submitting
bugs on either our web page at www.isc.org or in the README file
before submitting a bug.  These pages explain the proper
process and the information we find helpful for debugging.

exiting.
ifup: failed to bring up eth0
failed.
[ ok ] Starting RPC port mapper daemon: rpcbind.
[FAIL] Starting NFS common utilities: statd idmapd failed!
[ ok ] Cleaning up temporary files....
[FAIL] startpar: service(s) returned failure: nfs-common ... failed!
INIT: Entering runlevel: 3
[info] Using makefile-style concurrent boot in runlevel 3.
[ ok ] Starting NTP server: ntpd.
Recovering nvi editor sessions.
none found.
[ ok ] Starting OpenBSD Secure Shell server: sshd.
bash: cannot set terminal process group (959): Inappropriate ioctl for device
bash: no job control in this shell
root@lowrisc:/#

1.why there occur fail?
`[FAIL] Starting NFS common utilities: statd idmapd failed!`
`[FAIL] startpar: service(s) returned failure: nfs-common ... failed!`
2.
`bash: cannot set terminal process group (959): Inappropriate ioctl for device`
`bash: no job control in this shell`
this is normal status?
3.I can't use ssh.
I use command `ifconfig -a`, there have't `eth0`.
it only have `lo` and `sit0`.

And my reason is my team design a ip for ariane of pulp-platform.
So we hope use lowrisc-rootfs, but still use pulp-platform bitstream.

Thanks for your help.

jrrk2, 2019-12-18:
You probably managed to disable the Ethernet device driver amongst the various edits you have done. Needless to say if Ethernet is disabled, NFS and ssh will not work. Your best option right now would be to get the _unmodified_ LowRISC design up and running on your board, and then compare the boot logs between my working version and your version. For example I would expect to see a message such as this:

    [    3.007706] lowrisc-digilent-ethernet: Lowrisc ethernet platform (43000000-43007FFF) mapped to ffffffd004030000

early on during booting. If you don't get this either your device tree blob or kernel config (or Verilog) is wrong (the address may be different in pulp platform). You will appreciate it can be useful to combine outputs from different research groups in different countries, but it might not be the best starting point if you want to get something working quickly. Notwithstanding this, if you want to port your IP to the version of Ariane in lowRISC, I shall be happy to assist.

Right now I see a problem, you have made modifications to an unsupported configuration, and you don't have a reference baseline to compare against, and you are asking for advice whilst failing to provide adequate information to justify the deviation from the expected path of development. It's unlikely any other potential users can benefit from this discussion, except as an example of how not to do it.

This puts me in a difficult position, presuming to advice about a system of your own devising which I know next to nothing about. If I give you a bum steer, should I take the blame if your project fails? The best I can hope for is that at some point you will become sufficiently experienced to be able to cut your own swath.

monrick0303, 2019-12-19:
Sorry,
I only enable `CONFIG_ETHERNET` and `CONFIG_LOWRISC_DIGILENT_100MHZ`,
but disable `CONFIG_NET_VENDOR_LOWRISC` and `CONFIG_LOWRISC_MII_INIT`.
now i can enable ethernet through static ip.
(I modify /etc/network/interfaces)
And i only extra add  `ariane-sdk/configs/linux_defconfig `.

**it is boot logs:**
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 5.3.8-gc8da88dbc-dirty (yoyo@yoyo-All-Series) (gcc version 8.2.0 (GCC)) #13 Thu Dec 19 13:59:50 CST 2019
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] software IO TLB: mapped [mem 0xbeffb000-0xbeffb800] (0MB)
[    0.000000] elf_hwcap is 0x112d
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: swiotlb=noforce root=/dev/rda2 earlyprintk
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Sorting __ex_table...
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1019276K/1046528K available (4637K kernel code, 324K rwdata, 1432K rodata, 2524K init, 328K bss, 27252K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] plic: mapped 3 interrupts with 1 handlers for 2 contexts.
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x6eb3e4560, max_idle_ns: 881590406254 ns
[    0.000293] sched_clock: 64 bits at 15MHz, resolution 66ns, wraps every 4398046511080ns
[    0.005596] Console: colour dummy device 80x25
[    0.335285] printk: console [hvc0] enabled
[    0.343344] Calibrating delay loop (skipped), value calculated using timer frequency.. 30.00 BogoMIPS (lpj=150000)
[    0.360503] pid_max: default: 32768 minimum: 301
[    0.378085] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.390601] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.430799] *** VALIDATE proc ***
[    0.446169] *** VALIDATE cgroup1 ***
[    0.450483] *** VALIDATE cgroup2 ***
[    0.503932] devtmpfs: initialized
[    0.560584] random: get_random_u32 called from bucket_table_alloc.isra.11+0x4e/0x15e with crng_init=0
[    0.590022] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.606032] futex hash table entries: 256 (order: 0, 6144 bytes, linear)
[    0.656588] NET: Registered protocol family 16
[    1.514047] Bluetooth: Core ver 2.22
[    1.520789] NET: Registered protocol family 31
[    1.526375] Bluetooth: HCI device and connection manager initialized
[    1.538219] Bluetooth: HCI socket layer initialized
[    1.545818] Bluetooth: L2CAP socket layer initialized
[    1.555618] Bluetooth: SCO socket layer initialized
[    1.566325] clocksource: Switched to clocksource riscv_clocksource
[    2.021741] NET: Registered protocol family 2
[    2.053424] tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 8192 bytes, linear)
[    2.068382] TCP established hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    2.084699] TCP bind hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    2.099801] TCP: Hash tables configured (established 8192 bind 8192)
[    2.120389] UDP hash table entries: 512 (order: 2, 16384 bytes, linear)
[    2.131840] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    2.151034] NET: Registered protocol family 1
[    2.180667] RPC: Registered named UNIX socket transport module.
[    2.189777] RPC: Registered udp transport module.
[... 118 lines cut ...]
CONFIG_NFS_V4_1_IMPLEMENTATION_ID_DOMAIN="kernel.org"
CONFIG_NFS_V4_1_MIGRATION=y
CONFIG_NFS_USE_LEGACY_DNS=y
CONFIG_NFSD=y
CONFIG_NFSD_V2_ACL=y
CONFIG_NFSD_V3=y
CONFIG_NFSD_V3_ACL=y
CONFIG_NFSD_V4=y
CONFIG_NFSD_PNFS=y
CONFIG_NFSD_BLOCKLAYOUT=y
CONFIG_NFSD_SCSILAYOUT=y
CONFIG_GRACE_PERIOD=y
CONFIG_LOCKD=y
CONFIG_LOCKD_V4=y
CONFIG_NFS_ACL_SUPPORT=y
CONFIG_NFS_COMMON=y
CONFIG_SUNRPC=y
CONFIG_SUNRPC_GSS=y
CONFIG_SUNRPC_BACKCHANNEL=y
CONFIG_SUNRPC_SWAP=y

I so sorry about not provide adequate information,
and put you to a lot of trouble...

Thanks you

monrick0303, 2019-12-19:
**There is boot log of unmodified LowRISC linux:**
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Linux version 5.3.8-gdc77e5e11 (yoyo@yoyo-All-Series) (gcc version 8.2.0 (GCC)) #1 Thu Dec 19 16:00:06 CST 2019
[    0.000000] initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x00000000bfffffff]
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] software IO TLB: mapped [mem 0xbeffb000-0xbeffb800] (0MB)
[    0.000000] elf_hwcap is 0x112d
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: swiotlb=noforce root=/dev/rda2
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Sorting __ex_table...
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1019736K/1046528K available (4320K kernel code, 307K rwdata, 1314K rodata, 2532K init, 313K bss, 26792K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 0, nr_irqs: 0, preallocated irqs: 0
[    0.000000] plic: mapped 3 interrupts with 1 handlers for 2 contexts.
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x6eb3e4560, max_idle_ns: 881590406254 ns
[    0.000307] sched_clock: 64 bits at 15MHz, resolution 66ns, wraps every 4398046511080ns
[    0.005734] Console: colour dummy device 80x25
[    0.332632] printk: console [hvc0] enabled
[    0.340727] Calibrating delay loop (skipped), value calculated using timer frequency.. 30.00 BogoMIPS (lpj=150000)
[    0.357815] pid_max: default: 32768 minimum: 301
[    0.375152] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.387668] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.427847] *** VALIDATE proc ***
[    0.442942] *** VALIDATE cgroup1 ***
[    0.447228] *** VALIDATE cgroup2 ***
[    0.501204] devtmpfs: initialized
[    0.555934] random: get_random_u32 called from bucket_table_alloc.isra.11+0x4e/0x15e with crng_init=0
[    0.577883] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.593893] futex hash table entries: 256 (order: 0, 6144 bytes, linear)
[    0.627588] NET: Registered protocol family 16
[    1.427441] Bluetooth: Core ver 2.22
[    1.433698] NET: Registered protocol family 31
[    1.439987] Bluetooth: HCI device and connection manager initialized
[    1.450969] Bluetooth: HCI socket layer initialized
[    1.458721] Bluetooth: L2CAP socket layer initialized
[    1.468525] Bluetooth: SCO socket layer initialized
[    1.478415] clocksource: Switched to clocksource riscv_clocksource
[    1.936615] NET: Registered protocol family 2
[    1.968136] tcp_listen_portaddr_hash hash table entries: 512 (order: 1, 8192 bytes, linear)
[    1.983884] TCP established hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    2.000776] TCP bind hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    2.015400] TCP: Hash tables configured (established 8192 bind 8192)
[    2.035891] UDP hash table entries: 512 (order: 2, 16384 bytes, linear)
[    2.047322] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    2.066380] NET: Registered protocol family 1
[    2.095151] RPC: Registered named UNIX socket transport module.
[    2.104297] RPC: Registered udp transport module.
[    2.112027] RPC: Registered tcp transport module.
[    2.120010] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    8.080652] Initialise system trusted keyrings
[    8.094204] workingset: timestamp_bits=46 max_order=18 bucket_order=0
[    8.821432] NFS: Registering the id_resolver key type
[    8.829875] Key type id_resolver registered
[    8.835300] Key type id_legacy registered
[... 22 lines cut ...]
[   14.095363] Bluetooth: HCI UART protocol BCSP registered
[   14.109314] hidraw: raw HID events driver (C) Jiri Kosina
[   14.176892] NET: Registered protocol family 10
[   14.237947] Segment Routing with IPv6
[   14.246283] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[   14.286892] NET: Registered protocol family 17
[   14.294033] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   14.303704] Bluetooth: HIDP socket layer initialized
[   14.314554] start plist test
[   14.648715] end plist test
[   14.682219] Loading compiled-in X.509 certificates
[   14.697815] hctosys: unable to open rtc device (rtc0)
[   14.707200] Bluetooth: Starting self testing
[   14.713164] Bluetooth: Finished self testing
[   14.913899] Freeing unused kernel memory: 2532K
[   14.920278] This architecture does not have kernel memory protection.
[   14.931239] Run /init as init process
Mounting SD root
mount: mounting /dev/mmcblk0p2 on /mnt failed: No such device or address
mount -t ext4 /dev/mmcblk0p2 /mnt failed, dropping to ash

BusyBox v1.30.1 (Debian 1:1.30.1-4) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/bin/ash: can't access tty; job control turned off